notro/fbtft #398: /dev/fb1 not created, kernel panic

a7v3a6, 2016-06-19:
I would like to connect a Waveshare 4" RPI TFT to a Wandboard (imx6).
I use a kernel from Robert Nelson with a Debian 8:
`$ uname -a Linux arm 4.4.4-armv7-rt-x5 #1 SMP PREEMPT RT Sat Jun 18 19:06:47 CEST 2016 armv7l GNU/Linux` 
Framebuffer and SPI are enabled in the kernel. SPI is added to the device tree.
I initialize the display with:
`sudo modprobe flexfb  width=320  height=480  regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,1,0xC2,0x44,1,0xC5,0x00,0x00,0x00,0x00,1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3`
```
$ lsmod
Module                  Size  Used by
flexfb                 14217  0
fbtft                  38315  1 flexfb
caam_jr                17615  0
brcmfmac              257502  0
brcmutil                9380  1 brcmfmac
cfg80211              558305  1 brcmfmac
snd_soc_sgtl5000       17399  2
caam                    8596  0
snd_soc_fsl_ssi        15604  2
imx_pcm_fiq             5469  1 snd_soc_fsl_ssi
snd_soc_fsl_spdif      18884  2
snd_soc_imx_sgtl5000     4712  0
imx_pcm_dma             1675  2 snd_soc_fsl_ssi,snd_soc_fsl_spdif
snd_soc_imx_audmux      6721  1 snd_soc_imx_sgtl5000
snd_soc_imx_spdif       2551  0
dw_hdmi_ahb_audio       4960  0
snd_soc_core          192485  7 imx_pcm_dma,imx_pcm_fiq,snd_soc_sgtl5000,snd_soc_imx_sgtl5000,snd_soc_imx_spdif,snd_soc_fsl_ssi,snd_soc_fsl_spdif
snd_pcm_dmaengine       5894  1 snd_soc_core
snd_pcm               102545  6 dw_hdmi_ahb_audio,imx_pcm_fiq,snd_soc_core,snd_soc_sgtl5000,snd_soc_fsl_ssi,snd_pcm_dmaengine
snd_timer              23578  1 snd_pcm
snd                    72737  4 dw_hdmi_ahb_audio,snd_soc_core,snd_timer,snd_pcm
soundcore               8324  1 snd
uio_pdrv_genirq         4045  0
etnaviv                67118  0
uio                    10689  1 uio_pdrv_genirq
```
When I load the module the kernel crashes:
```
$ sudo modprobe fbtft_device debug=1 rotate=90 name=flexfb speed=16000000 gpios=reset:101,dc:90                                       
[ 1451.388376] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.                                      
[ 1451.405095] spidev spi0.0: spidev spi0.0 20000kHz 8 bits mode=0x00
[ 1451.411390] spidev spi0.1: spidev spi0.1 20000kHz 8 bits mode=0x00
[ 1451.417871] spidev spi0.0: Deleting spi0.0
[ 1451.427698] flexfb spi0.0: fbtft_request_gpios: 'reset' = GPIO101
[ 1451.434901] flexfb spi0.0: fbtft_request_gpios: 'dc' = GPIO90
[ 1451.812735] Kernel panic - not syncing: stack-protector: Kernel stack is corrupted in: bf406088                                          
[ 1451.812735]
[ 1451.812744] CPU: 0 PID: 594 Comm: modprobe Tainted: G         C      4.4.4-armv7-rt-x5 #1                                          
[ 1451.812747] Hardware name: Freescale i.MX6 Quad/DualLite (Device Tree)
[ 1451.812773] [<c001c214>] (unwind_backtrace) from [<c0015e44>] (show_stack+0x20/0x24)                                          
[ 1451.812790] [<c0015e44>] (show_stack) from [<c05a563c>] (dump_stack+0x80/0x94)                                          
[ 1451.812806] [<c05a563c>] (dump_stack) from [<c01731d8>] (panic+0xc4/0x24c)
[ 1451.812819] [<c01731d8>] (panic) from [<c004dcb4>] (print_tainted+0x0/0xb8)
[ 1451.812830] CPU1: stopping
[ 1451.812837] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G         C      4.4.4-armv7-rt-x5 #1                                          
[ 1451.812840] Hardware name: Freescale i.MX6 Quad/DualLite (Device Tree)
[ 1451.812857] [<c001c214>] (unwind_backtrace) from [<c0015e44>] (show_stack+0x20/0x24)                                          
[ 1451.812867] [<c0015e44>] (show_stack) from [<c05a563c>] (dump_stack+0x80/0x94)                                         
[ 1451.812879] [<c05a563c>] (dump_stack) from [<c0019874>] (handle_IPI+0x388/0x3e4)                                         
[ 1451.812890] [<c0019874>] (handle_IPI) from [<c0009728>] (gic_handle_irq+0x94/0x98)                                          
[ 1451.812901] [<c0009728>] (gic_handle_irq) from [<c0b4cc94>] (__irq_svc+0x54/0xa4)                                          
[ 1451.812905] Exception stack(0xee989f48 to 0xee989f90)
[ 1451.812914] 9f40: 00000001 00000000 00000000 c0028140 ee988000 c119bb04                                          
[ 1451.812921] 9f60: 00000001 c12b5148 c0b551b4 c118c674 00000000 ee989fa4 ee989fa8 ee989f98                                          
[ 1451.812927] 9f80: c0011cf8 c0011cfc 60070013 ffffffff
[ 1451.812942] [<c0b4cc94>] (__irq_svc) from [<c0011cfc>] (arch_cpu_idle+0x48/0x4c)                                          
[ 1451.812955] [<c0011cfc>] (arch_cpu_idle) from [<c009a8a0>] (default_idle_call+0x3c/0x48)                                          
[ 1451.812965] [<c009a8a0>] (default_idle_call) from [<c009ab7c>] (cpu_startup_entry+0x2d0/0x358)                                          
[ 1451.812974] [<c009ab7c>] (cpu_startup_entry) from [<c0019268>] (secondary_start_kernel+0x170/0x190)                                          
[ 1451.812984] [<c0019268>] (secondary_start_kernel) from [<100097cc>] (0x100097cc)                                                                                                                       
```
/dev/fb1 is not created:
```
:/sys/class/graphics$ ls
fb0  fbcon
```
The graphics fb1 entry in dmesg is missing:
```
[   54.878723] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[   54.895634] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[   54.907184] spidev spi0.0: spidev spi0.0 20000kHz 8 bits mode=0x00
[   54.913739] spidev spi0.1: spidev spi0.1 20000kHz 8 bits mode=0x00
[   54.920057] spidev spi0.0: Deleting spi0.0
[   54.927354] fbtft_device: GPIOS used by 'flexfb':
[   54.932770] fbtft_device: 'reset' = GPIO101
[   54.937154] fbtft_device: 'dc' = GPIO90
[   54.941019] spidev spi0.1: spidev spi0.1 20000kHz 8 bits mode=0x00
[   54.947229] spi spi0.0: flexfb spi0.0 16000kHz 8 bits mode=0x00
```
I'm stuck. What did I miss? Any ideas what I could try next?

notro, 2016-06-19:
Seems like you've hit a bug in (fbtft_init_display)[http://lxr.free-electrons.com/ident?i=fbtft_init_display].
-1 is a marker that signifies that a register number comes next, and the following values are written to that register. Since you obviously have forgotten the minus sign in front of many '1' in your init sequence, the ```msg[128]``` stack variable overflows. The array size should really have been 5*64 = 320.

Anyways, your problem will go away if you add a minus sign in front of your '1' numbers.

a7v3a6, 2016-06-21:
Yeap, that was it. Forgot the minus sign. Display is now up and running. Thank you!

closed 2016-06-21 by a7v3a6