notro/fbtft #399: Fail to get Adafruit 1.8" display to work after following all the steps

pranjalsingi, 2016-06-21:
@notro I have followed every step in your wonderful guide but still not able to work out my Adafruit 1.8" TFT to display.

When I run the command `sudo modprobe fbtft_device name=adafruit18` - my screen lights up with a white light but nothing happens. Below I would describe what steps I followed to get till here.

Steps:
1. `Changed the /dev/fb1 in /usr/share/X11/xorg.conf.d/99-fbturbo.conf`
2. Ran command for custom configurations.
3. Did an `sudo rpi-update` to install FBTFT drivers.
4. Uncomment the `dtparam=spi=on` from /boot/config.txt
5. Enabled driver for Adafruit 1.8" `sudo modprobe fbtft_device name=adafruit18`. Screen lights up.
6. dmesg shows the default pin assignments.
7. For testing did `sudo FRAMEBUFFER=/dev/fb1 startx` but nothing happens.
8. Moved the `99-fbturbo.conf` to ~ and rebooted but still nothing happens.

With all the above steps, I have LCD attach to the Raspberry Pi but it just shows a black screen with a blinking dash. When I comment the below in `/etc/X11/xorg.conf.d/fbtft.conf`and run the `startx`, the LCD shows me the desktop view.
#Section "ServerLayout"
    #Identifier "TFT"
    #Screen 0 "ScreenTFT"
#EndSection

I think my Pi recognize the screen is attached to it as when I load the drivers `sudo modprobe fbtft_device name=adafruit18` for adafruit18, it lights up but when if I load drivers of some different screen, it won't lights up. I have been working on it a while but its simply not working. 

Any help from you would be greatly appreciated Sir.

PS: I am following pin configuration from the website Raspberry official website pointed. Its [http://pinout.xyz/](url)  and the wiring I am following from [https://github.com/notro/fbtft/wiki/LCD-Modules#adafruit-18](url) .

Thanks

notro, 2016-06-22:
The easiest way to test is to use the console: ```con2fbmap 1 1```
A black backlit display indicates that it's working.
A white backlit display indicates that the driver has turned on backlight, but failed to initialize the controller. This is usually due to miswiring or a wrong init sequence.

pranjalsingi, 2016-06-22:
@notro Sir, I did try to test with `con2fbmap 1 1` but nothing happens. The screen just remains with a white backlit display.

This time I will attach some screen shots so that its more clear and you could help me better.
 Firstly I rebooted and did an ls -l . This shows that `99-turbo.conf` is in my ~  and the content is also printed.
Then I load the drivers with ` sudo modprobe fbtft_device name=adafruit18` so the screen lights up.

[image]

I have `dtparam=spi=on` in my /boot/config.txt . Printing the `dmesg` shown below
[image]
The DMA buffer memory is 4KiB for adafruit18. 

Now when I do `con2fbmap 1 1`, nothing happens. The screen is still white. 

Showing the contents of the file `/etc/X11/xorg.conf.d/fbtft.conf` . Nothing happens if the first Section is not commented. 
[image]
And when I do `sudo FRAMEBUFFER=/dev/fb1 startx` , there are no errors messages also.
[image]

But when I comment ScreenTFT section, the Monitor attached to Raspberry connected by HDMI shows the X Windows with command `sudo FRAMEBUFFER=/dev/fb1 startx`
[image]

Sir, I am novice in this field. Please tell me what can I do. As you mentioned that something can be wrong with wiring so let me tell you that I am using [http://pinout.xyz/](http://pinout.xyz/) for pinout and [https://github.com/notro/fbtft/wiki/LCD-Modules#adafruit-18](https://github.com/notro/fbtft/wiki/LCD-Modules#adafruit-18) for connection as I stated in my previous post. If something is wrong with it, please let me know.

Also I don't know what you meant by init sequence(I think initialization sequence). So could you tell what is it and where am I getting it wrong. I am stuck with it for many days and have no idea what is wrong and what to do.

Thanks

notro, 2016-06-23:
If you use the official Raspian image, it has had fbtft support for several months now.

Try lowering the speed: ```fbtft_device name=adafruit18 speed=4000000```

pranjalsingi, 2016-06-23:
@notro Sir it changes the speed down to 4MHz from 32 MHz when I run the above command but still neither `sudo FRAMEBUFFER=/dev/fb1 startx` nor `sudo con2fbmap 1 1` works to display anything on TFT screen.

notro, 2016-06-23:
Well then I don't know. A working setup should give a black backlit screen when the driver is loaded.

pranjalsingi, 2016-06-24:
@notro Thank you for all the help. I greatly appreciate that. Still trying to make that screen work.
Thanks

closed 2017-02-22 by notro