notro/fbtft #400: Support for Adafruit 2.8" TFT LCD with ILI9341?

pdp7, 2016-06-22:
I'd like to get this display working with fbtft and have not had any success so far:

2.8" TFT LCD with Touchscreen Breakout Board w/MicroSD Socket - ILI9341
PRODUCT ID: 1770
https://www.adafruit.com/product/1770

This is the startup sequence that the Python library uses:
https://github.com/adafruit/Adafruit_Python_ILI9341/blob/master/Adafruit_ILI9341/ILI9341.py#L179

I thought I would post here in the hopes that someone might have a suggestion as to what steps try next.

thanks!

notro, 2016-06-22:
I discovered recently that the PiTFT 2.8 (#1601) has got a MI0283QT-11 panel, the same as rpi-display. And I see that 1770 uses the same panel. So in theory it should work.
I assume you have closed the [IM{1,2,3} jumpers](https://learn.adafruit.com/adafruit-2-dot-8-color-tft-touchscreen-breakout-v2?view=all#spi-mode-jumpers) if you're using SPI.

ArduinoKid, 2016-06-25:
@pdp7 I have replied to your comment on my old question [here](https://github.com/notro/fbtft/issues/324), and it should hopefully help you out.

pdp7, 2016-08-20:
@notro curious if you might have any thoughts.

This is an Adafruit display with ILI9341:
* [2.8" TFT LCD with Cap Touch Breakout Board w/MicroSD Socket (PRODUCT ID: 2009)](https://www.adafruit.com/product/2090)

It works **OK** with your bone-debian-7.5-2014-05-14-2gb image and the [pre-compiled fbtft drivers for Linux 3.8.13-bone50](https://www.element14.com/community/community/designcenter/single-board-computers/next-gen_beaglebone/blog/2015/11/18/beaglebone-black-lcds-with-prebuilt-fbtft-drivers).
  * 

However, I can't seem to get it to work on newer images and/or newer kernels.  For example, [Debian 7.11 (2016-06-15) with 4.1.29-bone22](https://gist.github.com/pdp7/25ea49ce87bfa1fc95cd02a5618ce34d):
```
fb_ili9340 spi1.0: fbtft_request_gpios: gpio_request_one('dc'=48) failed with -16
fb_ili9340: probe of spi1.0 failed with error -16
```
Any idea what the cause of that error could be?

thanks!

notro, 2016-08-20:
-16 is -EBUSY so the gpio is claimed/requested already.

gpio_request_one() -> gpiod_request() -> __gpiod_request():
```c
        if (test_and_set_bit(FLAG_REQUESTED, &desc->flags) == 0) {
                desc_set_label(desc, label ? : "?");
                status = 0;
        } else {
                status = -EBUSY;
                goto done;
        }
```
http://lxr.free-electrons.com/ident?i=gpio_request_one

```c
#define EBUSY           16      /* Device or resource busy */
```
http://lxr.free-electrons.com/source/include/uapi/asm-generic/errno-base.h

notro, 2016-08-20:
If debugfs is enabled, have a look at /sys/kernel/debug/gpio. It should list all requested gpios.

pdp7, 2016-08-20:
@notro thank you very much for the insights!

pdp7, 2016-08-21:
I'll close this issue as I don't think the issue has do to do with fbtft.

I'll add a comment here once I've found a solution, in case anyone else runs into the issue, too.

closed 2016-08-21 by pdp7