notro/fbtft #401: double framebuffer

flexiti, 2016-06-29:
fbtft - can I use double framebuffer using virtual resolution ?

eg.
```
 vinfo.yoffset = vinfo.yres;
 ioctl(fbfd, FBIOPAN_DISPLAY, &vinfo)
```

notro, 2016-06-30:
No, it's not supported.

flexiti, 2016-07-01:
It is a pity. 
Is there some other way to avoid interference ? (Unfortunately less FPS, the more visible)

notro, 2016-07-01:
I don't know. I don't actually make use of these displays myself, I just make drivers for them :-)
SPI is usually a better interface choice: https://github.com/notro/fbtft/wiki/Performance

flexiti, 2016-07-01:
The problem framebuffer does not depend on whether the display is parallel or SPI
It is rather about the fbtft

eg. drawing a picture , you're in the middle of the drawing, while fbtft send it to the display and shows.
If we have eg. 5fps - for 200 milisecond will be displayed not finished picture (although eg. painting the entire screen takes us eg.2 ms, but we were unlucky in mid of our work your driver has send it to display and displays an unfinished drawing by 1/5 seconds - and this is seen as interference and flickers.

If we had a double framebuffer, we could decide when the drawing is finished and can be sent to the display. (just by set new pointer)
If we have less fps non finished picture is diplayed for longer, if we have more fps shorter.
but interference and flickers will be always,sometimes less, sometimes more visible.
Double framebuffer would eliminate this.

notro, 2016-07-02:
I don't see how that can help with tearing since we can't flip the buffer in the display controller which is the one that's being scanned out. So we will always write to the display RAM while it's being scanned out, regardless of which buffer we send.

If the controller had twice the amount of RAM necessary, we might be able to do page flipping using hw scrolling, but that would increase the price so I doubt it is available for cheap.

MIPI compliant controllers has a Tearing Effect Line output that could have been used, but I haven't seen any displays that make this signal available on the pinout.

flexiti, 2016-07-04:
I do not know if we understand each other well :)
It's not about a larger display buffer in display controller , only bigger fbtft buffor.
Normally fbtft transfer **buffer** to the display ram, and user write **to the same buffer** his new data.

Fbtft can have two buffers (or **one 2 times bigger buffer**). Then fbtft from one buffer sends to the memory of the display, and the user enters data into the second buffer.
Then, the user swap the buffers (not copy, just change pointers)
Fbtft only should pay attention to the virtual resolution(vinfo.yres) and offset to display(vinfo.yoffset)

http://betteros.org/tut/graphics1.php
https://www.raspberrypi.org/forums/viewtopic.php?f=67&t=19073

eg.
```
uint8_t *fbp,	//Front buffer base pointer
		*bbp;	//back buffer base pointer

void init_fbdev()
{
....
     ioctl(fd, FBIOGET_VSCREENINFO, &vinfo); // get defaults from kernel 
     vinfo.yres_virtual = vinfo.yres * 2; // request a virtual framebuffer twice as tall
     ioctl(fbfd, FBIOPUT_VSCREENINFO, &vinfo); // store the changes back into kernel config	
...
	fbp = mmap(0, screensize*2, PROT_READ | PROT_WRITE, MAP_SHARED, fb_fd, (off_t)0);
	bbp = fbp + screensize;
}

inline void swap_buffers()
{
	if (vinfo.yoffset==0)
		vinfo.yoffset = screensize;
	else
		vinfo.yoffset=0;

	//"Pan" to the back buffer
	ioctl(fb_fd, FBIOPAN_DISPLAY, &vinfo);

	//Update the pointer to the back buffer so we don't draw on the front buffer
	long tmp;
	tmp=fbp;
	fbp=bbp;
	bbp=tmp;
}

// example clear function
void clear()
{
	for (x=0;x<vinfo.xres;x++)
		for (y=0;y<vinfo.yres;y++)
		{
			long location = (x+vinfo.xoffset) * (vinfo.bits_per_pixel/8) + (y+vinfo.yoffset) *                 finfo.line_length;
			*((uint16_t*)(bbp + location)) = Red;
		}
    swap_buffers();
}

```

notro, 2016-07-05:
I understand that panning can be used for buffer flipping as you show here, but what I don't understand is how does this helps with tearing on fbtft?
On a "normal" display controller it's possible to change the buffer being scanned out at vblank by using panning resulting in no tearing.
But with fbtft displays we can't change/pan the buffer being scanned out, because it's on the controller and usually the same size as what's being displayed. So we will always have to write to display memory as it's being scanned out.

flexiti, 2016-07-05:
for a better example set fps=1:
We draw a line (it takes us eg. 10 ms), and when we're in the middle of the line, linux start the process of rewriting the buffer content to TFT screen - we see half a line on the TFT screen).
Although we have finish painting line after next 5ms(the line is all now) but on screen we see only half of it around the second.

On the other hand (eg. we "draw" circle and then "delete" it):
- Single buffer
   We will see: nothing, nothing, **part of circle**, the whole circle, **part of circle**, nothing, nothing

(and part of circle will be completely random)
- Double buffer
   We will see: nothing, nothing, the whole circle, the whole circle, nothing, nothing

notro, 2016-07-05:
A typical frame frequency on a 320x240 ILI9341 controller is 70hz, or 70 frames per second.
So the controller continously scans out the onboard Graphics RAM at this rate.
The maximum throughput I've seen with SPI on Raspberry Pi (125MHz) is ~12MB/s for a 480x320 display.
If we use theoretical numbers that would give a 320x240 display a maximum of 35-40fps.
So the display is scanning out at twice the speed I'm able to update it.
The only way we could avoid tearing is if we had access to the vblank signal.
That way we could start updating GRAM after vblank+delay and update half the GRAM, stop as the "beam" passes, then continue with the rest.
In other words we update half the GRAM per vblank. This signal is available as Tearing Effect Line on some controllers.
But it's impossible to achieve this kind of precision with ordinary Linux, so I guess the only way to avoid tearing is to either use a small display that it's possible to update in time, or use the RGB interface on these controllers coupled with a DPI display controller on the computer.

So in short: The framebuffer on the computer isn't the one that the display controller is scanning out. Which means that panning on the computer framebuffer doesn't help. It isn't the computer that scans out the buffer to the panel. We would need to do panning on the onboard display controller GRAM do achieve that.

flexiti, 2016-07-06:
Sending from the internal LCD memory to the physical screen is fast (eg. 70Hz), this is no problem, it is fast enough.

The problem is lower frequency of sending from fbtft to the LCD memory.
(As you describe eg.12 fps with the tinylcd35 display, and may be as low as 1 fps when parallel LCD is used) 
[image]

I suggest such a thing (using virtual resolution).
Until now, I suppose resolution virtual always equals real in FBTFT  ??
 if you think it's unnecessary it's ok.,quite a long time already talked about this :)

notro, 2016-07-06:
> I suppose resolution virtual always equals real in FBTFT ??

Yes.

There won't be any core changes to fbtft since it will be deprecated. I'm working on a replacement: https://github.com/notro/tinydrm/wiki

flexiti, 2016-07-07:
Thank you for the information :)

closed 2017-02-22 by notro