notro/fbtft #403: Dual displays

davelawr, 2016-07-19:
I have a project which requires a pair of small TFTs. My client is looking at the Adafruit ST7735R. Is it possible to connect two to the same RPi?

Thanks,

notro, 2016-07-21:
Yes: https://www.youtube.com/watch?v=0vfDifIU7gw

davelawr, 2016-07-21:
Thanks! Is this documented in a HOWTO?

notro, 2016-07-23:
I haven't seen a Device Tree overlay for that display, but you can have a look at this: https://github.com/raspberrypi/linux/blob/rpi-4.4.y/arch/arm/boot/dts/overlays/pitft22-overlay.dts
compatible should be "sitronix,st7735r"
Then add a node for your second display.

The default gamma values for that driver is probably off for the adafruit display: http://lxr.free-electrons.com/source/drivers/staging/fbtft/fbtft_device.c#L276
You can probably use the 'gamma' DT property (string), but I have never actually tried it. See http://lxr.free-electrons.com/ident?i=fbtft_probe_dt

Info about the various properties can be found here: https://github.com/notro/fbtft/wiki/fbtft_device
fbtft_device was made when Raspberry Pi didn't have DT support.

Refs:
https://github.com/notro/fbtft/wiki/FBTFT-RPI-overlays
https://www.kernel.org/doc/Documentation/devicetree/bindings/spi/spi-bus.txt
https://github.com/notro/fbtft/wiki/Debug

davelawr, 2016-08-27:
Hi,

after a little break I am back on this project. I now have two Sainsmart 
1.8" displays and using your wiring,

"dtparam=spi=on" in /boot/config.txt

and "modprobe fbtft_device name=adafruit18 rotate=90"

I have a working display using current debian jessie on a RPi 3B.

Please can talk me through what is necessary to fire up the second 
display? How do I wire it? what extra config is needed to make [I guess] 
/dev/fb2?

Many thanks,

notro, 2016-08-27:
You have to at least add one of the display using Device Tree to make it work. fbtft_device can only add one device.
Maybe this can be of help: https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=109065

davelawr, 2016-08-27:
Hi,

Thanks for the pointer, I've made the overlay work {including the 
"status="ok"" fix described} for a single display. For two displays, do 
I wire up SCL/SDA/CS/5V/GNC in parallel? and then separate DC and RES 
GPIO's?

Your patience is much appreciated; this is all new ground for me. I've 
never worked with overlays or device trees before.

notro, 2016-08-30:
You need separate CS, Chip Select, as well.

davelawr, 2016-08-30:
Hi,

Thank you, I connected up to SE1 and set the reset and DC GPIOs to 23 
and 18 and it's working!

At least on loading static images with 'fbi', I've not tried anything 
more advanced (X or QT as descrived by Hydrael)

I'm sure a HOWTO and FAQ on your your wiki would be well received.

Best regards,

>
> <https://github.com/notro/fbtft/issues/403#issuecomment-243389938>, or 
> mute the thread

notro, 2016-08-30:
> I'm sure a HOWTO and FAQ on your your wiki would be well received.

Anyone with a github account can edit the wiki ;-)

closed 2017-02-22 by notro