notro/fbtft #404: intel Edison failed when probe device mod

0x000000FF, 2016-07-26:
I have an intel edison ,and a TFT(hx8340bn) which the fbtft already surported.however,I use GPIOs instead of SPI,and I add a new platform in to fbtft_device.c,as below:
`.name = "mylcd",
		.pdev = &(struct platform_device) {
			.name = "fb_hx8340bn",
			.id = 0,
			.dev = {
			.release = fbtft_device_pdev_release,
			.platform_data = &(struct fbtft_platform_data) {
				.display = {
					.buswidth = 8,
					.backlight = 21,
				},
				.bgr = false,
				.gpios = (const struct fbtft_gpio []) {
					{ "reset", 9 },
					{ "dc", 47 },
					{ "wr", 33 },
					{ "cs", 23 },
					{ "rd", 24 },
					{ "db00", 37 },
					{ "db01", 51 },
					{ "db02", 50 },
					{ "db03", 38 },
					{ "db04", 31 },
					{ "db05", 45 },
					{ "db06", 32 },
					{ "db07", 46 },
					{},
				},
			},
			}`
at last ,I created the flash file with yocto.
When I test the function ,
modprobe fbtft_device name=mylcd
the console did not put any error,but dmesg give me this:

[  281.289764] fbtft_device:  SPI devices registered:
[  281.289798] fbtft_device:      ads7955 spi5.0 25000kHz 16 bits mode=0x00
[  281.289817] fbtft_device:      spidev spi5.1 25000kHz 8 bits mode=0x00
[  281.289830] fbtft_device:  'fb' Platform devices registered:
[  281.290136] fbtft_device:  GPIOS used by 'mostfunlcd':
[  281.290153] fbtft_device:    'reset' = GPIO9
[  281.290167] fbtft_device:    'dc' = GPIO47
[  281.290180] fbtft_device:    'wr' = GPIO33
[  281.290193] fbtft_device:    'cs' = GPIO23
[  281.290206] fbtft_device:    'rd' = GPIO24
[  281.290219] fbtft_device:    'db00' = GPIO37
[  281.290232] fbtft_device:    'db01' = GPIO51
[  281.290244] fbtft_device:    'db02' = GPIO50
[  281.290257] fbtft_device:    'db03' = GPIO38
[  281.290270] fbtft_device:    'db04' = GPIO31
[  281.290282] fbtft_device:    'db05' = GPIO45
[  281.290295] fbtft_device:    'db06' = GPIO32
[  281.290308] fbtft_device:    'db07' = GPIO46
[  281.290319] fbtft_device:  'fb' Platform devices registered:
[  281.290386] fbtft_device:      fb_hx8340bn id=0 pdata? yes
[  281.305230] fb_hx8340bn: probe of fb_hx8340bn.0 failed with error -12

and there is no fb device in /dev/.did I miss something?how could solve this?thanks

notro, 2016-07-27:
-12 is -ENOMEM which is probably returned because [fbtft_framebuffer_alloc](http://lxr.free-electrons.com/ident?i=fbtft_framebuffer_alloc) fails.
Try adding printk's to drivers/staging/fbtft/fbtft-core.c to track down which of the allocations that fail.
Using the fbtft_device parameter ```debug=7``` might get you some more logging output.

0x000000FF, 2016-07-28:
thanks for your replay,I found the error,the "txbuf = dmam_alloc_coherent(dev, txbuflen, &par->txbuf.dma, GFP_DMA);" in file fbtft-core.c ,line 917,dmam_alloc_coherent() return failed,because of edsion's damned DMA ,I changed to "txbuf = devm_kzalloc(par->info->device, txbuflen, GFP_DMA | GFP_ATOMIC);",so far so good.Thanks verymuch.

closed 2016-07-28 by 0x000000FF