notro/fbtft #405: flexfb init=-1.. -2...-3  ;   -1 , -2, -3 is mean what?

DRAgon734465502, 2016-07-28:
flexfb  width=320  height=480  regwidth=16 init=
-1,0xb0,0x0,
-1,0x11,
-2,250,
-1,0x3A,0x55,
-1,0xC2,0x44,
-1,0xC5,0x00,0x00,0x00,0x00,
-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,
-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,
-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,
-1,0x36,0x28,-1,0x11,-1,0x29,-3

what is the mean of " -1 , -2, -3 "

notro, 2016-07-28:
See the init section: https://github.com/notro/fbtft/wiki/fbtft_device

yafengabc, 2016-07-29:
-1,0xb0,0x0, means COMMND 0xb0 DATA 0x0
-2,250, means delay 250ms
-3 means finish

closed 2017-02-22 by notro