notro/fbtft #406: the screen no gpios “dc”

yafengabc, 2016-07-29:
I have an 3.5inch IPS LCD，and the controler is ili9481
the screen only 6 pins
3.3V
GND
SCL
SDA
RST
CS

and not dc（or RS）pin
i connected SDA to MOSI （GPIO10）
SCL to CLK（GPIO11） 
CS to CE0 （GPIO08）
RST to GPIO25
and use：
sudo modprobe fbtft_device debug=3 rotate=90 name=ili9481 speed=12000000 gpios=reset:25,dc:24,cs:8

but the screen not view any things
the dmesg info is:
[  172.199885] spidev spi0.0: spidev spi0.0 500kHz 8 bits mode=0x00
[  172.209993] spidev spi0.1: spidev spi0.1 500kHz 8 bits mode=0x00
[  172.219739] bcm2708_fb soc:fb: soc:fb id=-1 pdata? no
[  172.228507] spidev spi0.0: Deleting spi0.0
[  172.237128] fbtft_device: GPIOS used by 'ili9481':
[  172.245955] fbtft_device: 'reset' = GPIO25
[  172.252247] fb_ili9481: module is from the staging directory, the quality is unknown, you have been warned.
[  172.253723] fb_ili9481 spi0.0: fbtft_request_gpios: 'reset' = GPIO25
[  172.253752] fb_ili9481 spi0.0: fbtft_request_gpios: 'dc' = GPIO24
[  172.253778] fb_ili9481 spi0.0: fbtft_request_gpios: 'cs' = GPIO8
[  172.253798] fb_ili9481 spi0.0: fbtft_verify_gpios()
[  172.253810] fb_ili9481 spi0.0: fbtft_reset()
[  172.317034] fbtft_device: 'dc' = GPIO24
[  172.324503] fbtft_device: 'cs' = GPIO8
[  172.331923] spidev spi0.1: spidev spi0.1 500kHz 8 bits mode=0x00
[  172.341635] fb_ili9481 spi0.0: fb_ili9481 spi0.0 12000kHz 8 bits mode=0x00
[  172.373878] fb_ili9481 spi0.0: init: write(0x11)
[  172.382167] fb_ili9481 spi0.0: init: mdelay(50)
[  172.440062] fb_ili9481 spi0.0: init: write(0xD0) 0x07 0x42 0x18
[  172.449605] fb_ili9481 spi0.0: init: write(0xD1) 0x00 0x07 0x10
[  172.458892] fb_ili9481 spi0.0: init: write(0xD2) 0x01 0x02
[  172.467945] fb_ili9481 spi0.0: init: write(0xC0) 0x10 0x3B 0x00 0x02 0x11
[  172.477918] fb_ili9481 spi0.0: init: write(0xC5) 0x03
[  172.486056] fb_ili9481 spi0.0: init: write(0x3A) 0x55
[  172.493939] fb_ili9481 spi0.0: init: write(0xC8) 0x00 0x32 0x36 0x45 0x06 0x16 0x37 0x75 0x77 0x54 0x0C 0x00
[  172.509725] fb_ili9481 spi0.0: init: write(0x29)
[  172.738607] fb_ili9481 spi0.0: Display update: 1356 kB/s, fps=0
[  172.747847] fb_ili9481 spi0.0: fbtft_register_backlight(): led pin not set, exiting.
[  172.759395] graphics fb1: fb_ili9481 frame buffer, 480x320, 300 KiB video memory, 4 KiB DMA buffer memory, fps=20, spi0.0 at 12 MHz

there has an example C51 program only use 4pins to control it：
#include <intrins.h>
#include <reg51.h>
#include <absacc.h>

sbit	CSB	=	P3^3;
sbit	RESETB	=	P3^4;
sbit	SDA	=	P3^5;
sbit	SCL	=	P3^6;

sbit	KEY =	P3^2;

void LCD_INIT_ILI9481(void);
void LCD_WRITE_DATA_dis(char colorH,char colorL);
void LCD_TEST_SingleColor_16BPP(char colorH,char colorL);
void LCD_TEST_SingleColor(char colorH,char colorz,char colorL);
void LCD_WRITE_CMD(char cmd1);
void LCD_WRITE_DATA(char dataa);
void LCD_TEST_barColor();
void HD_reset(void);
void Keypress(void);

void delayms(int value);
void delayus(int value1);
unsigned char code barcolor[];

unsigned char code barcolor[]={
0Xff,0X00,0X00,0X00,0Xff,0X00,0X00,0X00,0Xff,0Xff,0Xff,0Xff,0X00,0X00,0X00,0Xff,
0Xff,0X00,0X00,0Xff,0Xff,0Xff,0X00,0Xff,
};									  
void main(){

        HD_reset();
        delayms(50);
		LCD_INIT_ILI9481();
	while(1)
	{
/*       LCD_TEST_SingleColor(0xff,0x00,0x00); //display red

       LCD_TEST_SingleColor(0x00,0xff,0x00);  //display blue

	   LCD_TEST_SingleColor(0x00,0x00,0xff);  //display blue

       LCD_TEST_SingleColor(0xff,0xff,0x00);  //display yellow

	   LCD_TEST_SingleColor(0x00,0xff,0xff);  //display 
*/
//	   LCD_TEST_SingleColor_16BPP(0xf8,0x00);
// Keypress();
//	   LCD_TEST_SingleColor_16BPP(0x07,0xE0);
//Keypress();
//	   LCD_TEST_SingleColor_16BPP(0x00,0x1F);
//Keypress();
  LCD_TEST_barColor();
//Keypress();
	}
}

void HD_reset(){
	RESETB=1;
	delayms(1);	
	RESETB=0;
	delayms(10);
	RESETB=1;
	delayms(100);
	}

   void LCD_INIT_ILI9481(void)
	{

	 LCD_WRITE_CMD(0x11);
	 	delayms(20);

	 LCD_WRITE_CMD(0xD0);
	 LCD_WRITE_DATA(0x07);
	 LCD_WRITE_DATA(0x41);
	 LCD_WRITE_DATA(0x1d);

	 LCD_WRITE_CMD(0xD1);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x2b);
	 LCD_WRITE_DATA(0x1f);

	 LCD_WRITE_CMD(0xD2);
	 LCD_WRITE_DATA(0x01);
	 LCD_WRITE_DATA(0x11);

	 LCD_WRITE_CMD(0xC0);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x3B);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x02);
	 LCD_WRITE_DATA(0x11);
	 LCD_WRITE_DATA(0x00);

	 LCD_WRITE_CMD(0xC5);
	 LCD_WRITE_DATA(0x03);

	 LCD_WRITE_CMD(0xC6);  //interface control
	 LCD_WRITE_DATA(0x80);

	 LCD_WRITE_CMD(0xC8);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x14);
	 LCD_WRITE_DATA(0x33);
	 LCD_WRITE_DATA(0x10);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x16);
	 LCD_WRITE_DATA(0x44);
	 LCD_WRITE_DATA(0x36);
	 LCD_WRITE_DATA(0x77);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x0f);
	 LCD_WRITE_DATA(0x00);

	 	 	 
	 LCD_WRITE_CMD(0xF3);
	 LCD_WRITE_DATA(0x40);
	 LCD_WRITE_DATA(0x0A);

	 LCD_WRITE_CMD(0xF0);
	 LCD_WRITE_DATA(0x08);

	 LCD_WRITE_CMD(0xF6);
	 LCD_WRITE_DATA(0x84);

	 LCD_WRITE_CMD(0xF7);
	 LCD_WRITE_DATA(0x80);

	 LCD_WRITE_CMD(0x0C);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x55);

	 LCD_WRITE_CMD(0xB4);
	 LCD_WRITE_DATA(0x00);

	 LCD_WRITE_CMD(0xB3);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x01);
	 LCD_WRITE_DATA(0x06);
	 LCD_WRITE_DATA(0x01);

	 LCD_WRITE_CMD(0x36);
	 LCD_WRITE_DATA(0x48);

	 LCD_WRITE_CMD(0x3a);
	 LCD_WRITE_DATA(0x66);

	 LCD_WRITE_CMD(0x20);
//	 LCD_WRITE_CMD(0x21);

	 LCD_WRITE_CMD(0x2A);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x01);
	 LCD_WRITE_DATA(0x3F);

	 LCD_WRITE_CMD(0x2B);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x01);
	 LCD_WRITE_DATA(0xdf);

	 delayms(120);

	 LCD_WRITE_CMD(0x29);
	 delayms(120);

	 LCD_WRITE_CMD(0x2C);

	} 

//-------------------------------9 bit chansfer------------
void LCD_WRITE_CMD(char cmd1)
{
  	unsigned char nub;

	CSB=0;

	SCL=0;				 
	SDA=0;
	SCL=1;
	for(nub=0;nub<8;nub++)
	{
		SCL=0;			
			if((cmd1&0x80)==0x80)
			{
				SDA=1;
			}
			else
			{
				SDA=0;
			}
			cmd1=(cmd1<<1);
		SCL=1;

	}
	CSB=1;

}

void LCD_WRITE_DATA(char dataa)
{
  	unsigned char nub;

	CSB=0;	

	SCL=0;		 
	SDA=1;
	SCL=1;
	for(nub=0;nub<8;nub++)
	{
		SCL=0;			
			if((dataa&0x80)==0x80)
			{
				SDA=1;
			}
			else
			{
				SDA=0;
			}
			dataa=(dataa<<1);
		SCL=1;
	}
	CSB=1;

}

//--------------------16bp218bp--------------
void LCD_WRITE_DATA_dis(char colorH,char colorL)
{
	LCD_WRITE_DATA(colorH&0xf8);
	LCD_WRITE_DATA(((colorH&0x07)<<5)+((colorL&0xe0)>>3));
	LCD_WRITE_DATA((colorL&0x1f)<<3);
}

void LCD_TEST_SingleColor(char colorH,char colorz,char colorL){
     int i,j;

	 LCD_WRITE_CMD(0x2A);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x01);
	 LCD_WRITE_DATA(0x3F);

	 LCD_WRITE_CMD(0x2B);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x01);
	 LCD_WRITE_DATA(0xe0);

	 LCD_WRITE_CMD(0x2C);

	 for (i=0;i<480;i++){
		for (j=0;j<320;j++){
		LCD_WRITE_DATA(colorH);
		LCD_WRITE_DATA(colorz);
		LCD_WRITE_DATA(colorL);

			}
		}
	}
 //##########################红，绿，蓝，黄，橙，黑，白。紫############
void LCD_TEST_barColor(){
     int i,j,k=0,p;

	 LCD_WRITE_CMD(0x2A);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x01);
	 LCD_WRITE_DATA(0x3F);

	 LCD_WRITE_CMD(0x2B);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x01);
	 LCD_WRITE_DATA(0xe0);

	 LCD_WRITE_CMD(0x2C);

	 for (p=0;p<8;p++)
	 {
	 	for (i=0;i<60;i++){
			for (j=0;j<320;j++){
			LCD_WRITE_DATA(barcolor[k]);
			LCD_WRITE_DATA(barcolor[k+1]);
			LCD_WRITE_DATA(barcolor[k+2]);
			}
		}
		k++;
		k++;
		k++;
	  }
	}

void LCD_TEST_SingleColor_16BPP(char colorH,char colorL){
     int i,j;

	 LCD_WRITE_CMD(0x2A);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x01);
	 LCD_WRITE_DATA(0x3F);

	 LCD_WRITE_CMD(0x2B);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x00);
	 LCD_WRITE_DATA(0x01);
	 LCD_WRITE_DATA(0xe0);

	 LCD_WRITE_CMD(0x2C);

	 for (i=0;i<480;i++){
		for (j=0;j<320;j++){
		LCD_WRITE_DATA_dis(colorH,colorL);

			}
		}
	}

void Keypress()
	{
	while(KEY)
	delayms(10);
	}

void delayms(int value)
	{
	while (value){
		delayus(99);
		value--;
		}
	}

void delayus(int value1)
	{
	while (value1)

	value1--;
	}

yafengabc, 2016-07-29:
i read the ILI9481 s datasheet
i found this chip need use IM0 IM1 IM2 to select the interface type
if this three pins is 1 0 1,the data need 9bit,and the first bit means D(data)/CX(command)
if this three pins is 1 1 1 the data is 8bit and use single DC pin to select command/data
is there any way to use 9bit data type?

notro, 2016-07-29:
From the top of my head: fbtft_device custom width= height= buswidth=9 ...

Source: http://lxr.free-electrons.com/source/drivers/staging/fbtft/fbtft_device.c

closed 2017-02-22 by notro