notro/fbtft #407: Sainsmart 18 tft not working on RPi 3

11ishan, 2016-08-01:
Hey notro,
I'm facing a similar problem as this guy. 
http://raspberrypi.stackexchange.com/questions/37611/tft-kernel-module-not-loading-on-startup

Please could you help me out with it?
Thank you

11ishan, 2016-08-02:
The screen becomes black after i load the drivers using modprobe, but doesn't display anything

notro, 2016-08-02:
This might help: https://github.com/notro/fbtft/wiki#make-it-permanent-debian-jessie-arch-linux-systemd

11ishan, 2016-08-02:
I tried the steps you suggested,  but no luck.
Dmesg is showing me an error " Failed to start load kernel modules.

11ishan, 2016-08-02:
The driver is loading properly, but the display isn't displaying anything, just a black screen as soon as i use modprobe

11ishan, 2016-08-02:
And there is no fbtft.conf file in both locations

notro, 2016-08-03:
> The driver is loading properly, but the display isn't displaying anything, just a black screen as soon as i use modprobe

That's expected. This will put console 1 on /dev/fb1:
con2fbmap 1 1

For X windows see: https://github.com/notro/fbtft/wiki/FBTFT-on-Raspian

11ishan, 2016-08-04:
If i edit the 99-fbturbo.conf file, i get an error saying failed to load kernel modules and later i get failed to start light display manager.  During boot.

11ishan, 2016-08-04:
It seems fbtft drivers aren't compatible with RPi 3.   The drivers start loading bcm2708_fb* various stuff like width, height etc.

11ishan, 2016-08-04:
Using journalctl -xn,  i get the following error 

Host name not found:  0.debian.pool
Host name not found:  1.debian.pool
Host name not found:  2.debian.pool
Host name not found:  3.debian.pool

11ishan, 2016-08-04:
And, 
Host name not found:  0.debian.pool.ntp.org
Host name not found: 1.debian.pool.ntp.org
Host name not found:  2.debian.pool.ntp.org
Host name not found:  3.debian.pool.ntp.org

notro, 2016-08-07:
I think you're better off asking in the Pi forums: https://www.raspberrypi.org/forums/viewforum.php?f=44

11ishan, 2016-08-08:
Ok. Thanks for helping me out.

11ishan, 2016-08-08:
Got the tft working!!!  Thanks a lot for your help!!!

11ishan, 2016-08-08:
As it turns out the fbturbo config file was causing problems.

closed 2017-02-22 by notro