notro/fbtft #409: Capactive touch support for Adafruit 2.8" breakout (PID 2090)?

pdp7, 2016-08-22:
@notro I'm trying to get cap touch support working on the BeagleBone for [Adafruit 2.8" breakout (PID 2090)](https://www.adafruit.com/product/2090) which uses `FT6206` cap touch controller.  It appears that the Raspberry Pi image from Adafruit uses fbtft patches to add `ft6x06_ts` driver.

Do you know if there has been any work to upstream ft6206 support?

_Related Adafruit forum post:
[2,8 " display on beagleBone black](https://forums.adafruit.com/viewtopic.php?f=49&t=95309&start=15)_

notro, 2016-08-22:
I cleaned up Adafruits version and sent it upstream as mentioned here: https://github.com/raspberrypi/linux/issues/1050

However it turned out that another driver had support for that chip already, so it's going away: https://patchwork.kernel.org/patch/9264845/

Here's the binding doc patch: https://patchwork.kernel.org/patch/9269027/

pdp7, 2016-08-23:
@notro  Thanks very much for the detailed response

closed 2016-08-23 by pdp7