notro/fbtft #410: AdaFruit PiTFT Plus 480x320 3.5" Display

XYEFA, 2016-08-22:
Hello,

I am using an AdaFruit PiTFT Plus 480x320 3.5" Display (https://www.adafruit.com/products/2441), and I noticed the HX8357 drivers were in staging but I have been unable to get these drivers to work on my ARM based board (not a Raspberry Pi).

I wanted to find out the state of the driver for the HX8357 and who has had success getting this display to work.

Regards!

DRAgon734465502, 2016-08-24:
are you sure the LCD is  HX8357 drivers chip

XYEFA, 2016-08-24:
Hello and thank you for the reply. This is what I am referencing: https://learn.adafruit.com/adafruit-pitft-3-dot-5-touch-screen-for-raspberry-pi/downloads

DRAgon734465502, 2016-08-25:
http://www.spotpear.com/learn/HX8357-D.rar

This is the LCD from china as the same HX8357d

Hope will help you

XYEFA, 2016-08-25:
I will look at that.

I am getting a lot debug messages written to the screen ... 

fb_hx8357d spi2.0: fbtft_write_reg8_bus8: 2a 00 00 01 df
fb_hx8357d spi2.0: fbtft_write_spi(len=1): 2a
fb_hx8357d spi2.0: fbtft_write_spi(len=4): 00 00 01 df
fb_hx8357d spi2.0: fbtft_write_reg8_bus8: 2b 00 18 00 1f
fb_hx8357d spi2.0: fbtft_write_spi(len=1): 2b
fb_hx8357d spi2.0: fbtft_write_spi(len=4): 00 18 00 1f
fb_hx8357d spi2.0: fbtft_write_reg8_bus8: 2c
fb_hx8357d spi2.0: fbtft_write_spi(len=1): 2c
fb_hx8357d spi2.0: fbtft_write_vmem16_bus8(offset=23040, len=7680)
fb_hx8357d spi2.0: fbtft_write_spi(len=4096): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
fb_hx8357d spi2.0: fbtft_write_spi(len=3584): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
fb_hx8357d spi2.0: Display update: 86 kB/s, fps=0
fb_hx8357d spi2.0: fbtft_update_display(start_line=24, end_line=31)
fb_hx8357d spi2.0: fbtft_write_reg8_bus8: 2a 00 00 01 df
fb_hx8357d spi2.0: fbtft_write_spi(len=1): 2a
fb_hx8357d spi2.0: fbtft_write_spi(len=4): 00 00 01 df
fb_hx8357d spi2.0: fbtft_write_reg8_bus8: 2b 00 18 00 1f
fb_hx8357d spi2.0: fbtft_write_spi(len=1): 2b
fb_hx8357d spi2.0: fbtft_write_spi(len=4): 00 18 00 1f
fb_hx8357d spi2.0: fbtft_write_reg8_bus8: 2c
fb_hx8357d spi2.0: fbtft_write_spi(len=1): 2c
fb_hx8357d spi2.0: fbtft_write_vmem16_bus8(offset=23040, len=7680)
fb_hx8357d spi2.0: fbtft_write_spi(len=4096): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
fb_hx8357d spi2.0: fbtft_write_spi(len=3584): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
fb_hx8357d spi2.0: Display update: 95 kB/s, fps=0

<<

XYEFA, 2016-08-25:
Things are looking good.

closed 2016-08-25 by XYEFA

DRAgon734465502, 2016-08-30:
hello
do you solve your problem ?