notro/fbtft #412: How do you silence the console debug output on tty0?

XYEFA, 2016-08-26:
It looks like I have the AdaFruit Display working on my ARM board, but it is blasting out a lot of debug messages on my serial tty0 , which I use for my console login (backdoor/frontdoor/configuration). How do I silence these messages from the driver?

fb_hx8357d spi2.0: fbtft_write_vmem16_bus8(offset=23040, len=7680)
fb_hx8357d spi2.0: fbtft_write_spi(len=4096): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
fb_hx8357d spi2.0: fbtft_write_spi(len=3584): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
fb_hx8357d spi2.0: Display update: 86 kB/s, fps=0
fb_hx8357d spi2.0: fbtft_update_display(start_line=24, end_line=31)
fb_hx8357d spi2.0: fbtft_write_reg8_bus8: 2a 00 00 01 df
fb_hx8357d spi2.0: fbtft_write_spi(len=1): 2a
fb_hx8357d spi2.0: fbtft_write_spi(len=4): 00 00 01 df
fb_hx8357d spi2.0: fbtft_write_reg8_bus8: 2b 00 18 00 1f
fb_hx8357d spi2.0: fbtft_write_spi(len=1): 2b
fb_hx8357d spi2.0: fbtft_write_spi(len=4): 00 18 00 1f
fb_hx8357d spi2.0: fbtft_write_reg8_bus8: 2c
fb_hx8357d spi2.0: fbtft_write_spi(len=1): 2c
fb_hx8357d spi2.0: fbtft_write_vmem16_bus8(offset=23040, len=7680)
fb_hx8357d spi2.0: fbtft_write_spi(len=4096): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
fb_hx8357d spi2.0: fbtft_write_spi(len=3584): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
fb_hx8357d spi2.0: Display update: 95 kB/s, fps=0

notro, 2016-08-27:
The debug property controls this.

XYEFA, 2016-08-27:
Thank you for the reply. I read the following: https://github.com/notro/fbtft/wiki/Debug

When set from the **Device Tree**, are the debug levels are the debug values numbered from (integer) 1 to 7, or is it a bit field?

<<

notro, 2016-08-30:
It's a combination, bits 0,1,2 are used for the shorthand levels.

http://lxr.free-electrons.com/source/drivers/staging/fbtft/fbtft-sysfs.c#L152

closed 2017-02-22 by notro