notro/fbtft #416: 3.5 Inch RPI LCD with XPT2046 Controller

Tes3awy, 2016-09-13:
I have bought a brand new RPI LCD. The controller of this LCD is XPT2046. meanwhile, I can't find this LCD controller listed in the tables provided. Does anyone have any idea which one is this LCD screen in ftbt_device.c file ?

Texy, 2016-09-13:
That chip is the touch panel controller, not the display controller. 
Texy

Tes3awy, 2016-09-13:
So which one should I choose or what info do you want from me to add?

xJoyride, 2016-11-04:
Hey @Tes3awy, I bought the same clone board as you. It took me a while, but eventually I found a post that helped me get it up and running.

https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=139917&start=25

Check it out! Good luck man!

notro, 2018-05-10:
Closing issue since there has been no activity for more than 2 months. Reopen if needed.

closed 2018-05-10 by notro

cnlohr, 2018-11-01:
I know this is a dead thread and off topic, but has anyone figured out what the actual display controller is?  I need to use a display like this but not with a pi.  I really need the SPI spec for the display.

cnlohr, 2018-11-01:
Looks like it's probably the ILI9486 see https://github.com/juj/fbcp-ili9341