notro/fbtft #422: T15P00 - 1.5 inch 480x240 LCD with OTA5182A driver

tr1p1ea, 2016-10-26:
Due to the recent wave of smart watches out of China there is increased ability to access small footprint displays with resolutions greater than what is typically available currently.

I have purchased a handful of T15P00 displays which are 1.5 inch and have a resolution of 480x240 - which is quite nice. They also run the OTA5182A driver which is something that I haven't really heard of. It does claim to have a kind of 3-wire SPI setup which could prove useful.

These displays are [hopefully] great for small RPI applications and I would like to see about getting support in fbtft if possible.

I have datasheets for the display and controller along with basic circuitry and some example init code below:
Display datasheet: http://tr1p1ea.net/files/downloads/hardware/lcd/t15p00/T15P00_SPEC.pdf
Display minimal power circuit: http://tr1p1ea.net/files/downloads/hardware/lcd/t15p00/T15P00%20peripheral%20circuit.pdf
Driver datasheet: http://tr1p1ea.net/files/downloads/hardware/lcd/t15p00/OTA5182A-C2_V071.pdf
Example init sequence: http://tr1p1ea.net/files/downloads/hardware/lcd/t15p00/OTA5182A_initial%20code(Vert0.1).txt

Any help on getting this display functioning would be appreciated.

notro, 2016-10-28:
fbtft only works with displays that has onboard RAM which the controller scans out to the display.
The SPI interface on this controller is only used for configuration and not for pixel data which fbtft requires.

Maybe DPI can be used: https://www.raspberrypi.org/documentation/hardware/raspberrypi/dpi/README.md

tr1p1ea, 2016-10-31:
OK I see, makes sense. I have a couple more mini-displays that have more common driver IC's (which should have GRAM iirc) - but are not SPI.

http://tr1p1ea.net/files/downloads/hardware/lcd/TF-SPEC15411A-V0.pdf
http://tr1p1ea.net/files/downloads/hardware/lcd/TF-SPEC15412A-V0.pdf

Is it still possible to utilise these with fbtft?

Thanks heaps for the feedback, and of course the fantastic work on fbtft.

notro, 2016-10-31:
ILI9341 is supported. Your display has a parallel bus, which will be driven by the gpio subsystem and it will be slow compared to SPI.

Something like this:
```
modprobe fbtft_device custom name=fb_ili9341 width=xx height=xx init=-1,....,-3 gpios=reset:17,dc:1,wr:0,cs:21,db00:9,db01:11,db02:18,db03:23,db04:24,db05:25,db06:8,db07:7,led:4
```
See https://github.com/notro/fbtft/wiki/fbtft_device

When you have it working you can make it load it automatically in 2 ways:
- https://github.com/notro/fbtft/wiki#make-it-permanent-debian-jessie-arch-linux-systemd
- https://github.com/notro/fbtft/wiki/FBTFT-RPI-overlays

tr1p1ea, 2016-11-03:
Awesome, the perm stuff really helps out.

It seems like this is a bit slow so I have ordered some SPI mini LCD's to test out.

One has the ST7789V driver which iirc is supported to some extent. The other runs ST7796H which I haven't read much about though I assume it would be somewhat similar but may differ from an initialisation perspective.

The datasheets for the displays are here:

http://tr1p1ea.net/files/downloads/hardware/lcd/F122US01A-spec.pdf
http://tr1p1ea.net/files/downloads/hardware/lcd/F61TS02A-SPEC.pdf

Thanks heaps for your help.

closed 2016-11-15 by tr1p1ea

epikao, 2017-07-02:
Hello, do you have the ST7789V fbtft, flexfb configuration code?