notro/fbtft #423: Really struggling to follow instructions for installing.

alidaf, 2016-11-02:
Hi notro.
Thank you for sharing your fbtft code. I have some issues with the instructions:
In places you have examples, such as 'example SPI config'. What do I do with this information?
Further on there is an instruction to ...
cd /usr/src/linux
zcat /proc/config.gz > .config
but /proc/config.gz doesn't exist.
The entire section regarding Device Trees is really hard to understand too.
I'm trying to compile for the SSD1322 by using the presslab-us fork but they have just linked to your instructions so I am absolutely stuck.

notro, 2016-11-06:
If Raspberry Pi, this is how to build a kernel: https://www.raspberrypi.org/documentation/linux/kernel/building.md

This is the latest fbtft driver added in 4.6: https://github.com/raspberrypi/linux/commit/12e2a34165d514bdbcab3e3e46f12b44cb318361
That will give you an idea of how to add a driver to the kernel mentioned in the build doc.

You can add a device like this: ```modprobe fbtft_device custom name=fb_ssd1322 width=xx height=xx speed =16000000 gpios=dc:xx,reset:xx```
https://github.com/notro/fbtft/wiki/fbtft_device

Try Device Tree when you get that working.

alidaf, 2016-11-07:
Thanks for that. I'll give it another try.

alidaf, 2016-11-07:
The kernel is compiling fine and booting now but there is no sign of the new driver. I have copies of the files (Kconfig, Makefile & fb_ssd1322.c) in linux/drivers/staging/fbtft/ as well as linux/drivers/video/fbtft/ but neither seem to be captured in the compilation. Is there another Makefile to edit?

notro, 2016-11-09:
Have you enabled it?
```make menuconfig```
Device Drivers  ---> [*] Staging drivers  --->  <M>   Support for small TFT LCD display modules  --->

```
$ grep FB_TFT_SSD .config
CONFIG_FB_TFT_SSD1289=m
# CONFIG_FB_TFT_SSD1305 is not set
CONFIG_FB_TFT_SSD1306=m
# CONFIG_FB_TFT_SSD1325 is not set
CONFIG_FB_TFT_SSD1331=m
CONFIG_FB_TFT_SSD1351=m

```

alidaf, 2016-11-10:
I was following the instructions for Linux 3.15+, which do not have the menuconfig step. Nonetheless I tried it and they are still not being compiled despite being checked in the menu. Using make menuconfig also stopped earlier steps from being repeated after it didn't work. I've had to delete everything and start again but I'm back to the same step now.

I have cloned into /linux/drivers/video and modified
/linux/drivers/video/Kconfig
 added 'source "drivers/video/fbtft/Kconfig"' below the similar fbdev directive.
/linux/drivers/video/Makefile
 added 'obj-y += fbtft/' below the similar fbdev directive.
/linux/drivers/video/fbtft/Kconfig
 added a config directive for the SSD1322, similar to SSD1306.
/linux/drivers/video/fbtft/Makefile
added 'obj-$(CONFIG_FB_TFT_SSD1322) += fb_ssd1322.o' below the SSD1306 line.

As far as I can tell, the latter 2 are purely for using 'make menuconfig', but this appears to be a deprecated step for Linux 3.15+.

Previously I tried cloning into /linux/drivers/video/fbdev as explicit in your instructions but the Kconfig and Makefile files in that folder don't seem to have the expected content, i.e. the 'source' and 'obj-y' directives. It still didn't work.

notro, 2016-11-10:
Which Linux version are you using?

alidaf, 2016-11-10:
4.4.26-v7+

notro, 2016-11-10:
In that case you should put your driver in drivers/staging/fbtft, not drivers/video/fbtft.
Like in the commit I pointed to: https://github.com/raspberrypi/linux/commit/12e2a34165d514bdbcab3e3e46f12b44cb318361

alidaf, 2016-11-10:
That's the first thing I tried after your first response but that didn't work, which is why I've tried to piece together all of the different and conflicting bits of information all over the wiki. The thing is, all of the other drivers that are in /linux/drivers/video/fbtft still get compiled, but mine doesn't, and nothing in 'staging/fbtft' does either, despite the Kconfig in 'staging' pointing to staging/fbtft/.

notro, 2016-11-10:
Does CONFIG_FB_TFT_SSD1322 show up in ```.config````?

If you have git cloned the repo, can you do a *git diff* of your work and post the result somewhere so I can look at it? The important bit is the changes to drivers/staging/fbtft/{Kconfig,Makefile}.

alidaf, 2016-11-10:
I've managed to compile everything and modprobe the driver. It didn't compile anything in staging. I think it depends on where 'make menuconfig' is actually run. There was a problem with the ssd1322 driver file due to the change to the FBTFT_REGISTER_DRIVER call requiring an additional parameter but I fixed that, ran 'make' and 'make modules_install' and it worked. 'make menuconfig' seems to be an absolute requirement but to make sure, I'm going to clear everything out again, run through what I think the correct procedure is, document it and send you the result.

alidaf, 2016-11-10:
No /dev/fb1 device has been created so I can't test it. There is no 'graphics fb1' result in dmesg.

notro, 2016-11-10:
The first step is to provide a driver which you have done.
```
# just to illustrate my point
$ sudo modprobe fb_ssd1331

$ ls -l /sys/bus/spi/drivers/
total 0
drwxr-xr-x 2 root root 0 Nov 10 19:47 fb_ssd1331
drwxr-xr-x 2 root root 0 Nov 10 19:46 spidev
drwxr-xr-x 2 root root 0 Nov 10 19:46 stmpe-spi
```

The next step is to add a device that will make use of the driver.
```
# spi is enabled
$ grep dtparam /boot/config.txt
dtparam=spi=on

# spi bus
$ ls -l /sys/bus/spi/devices/
total 0
lrwxrwxrwx 1 root root 0 Nov 10 19:55 spi0.0 -> ../../../devices/platform/soc/20204000.spi/spi_master/spi0/spi0.0
lrwxrwxrwx 1 root root 0 Nov 10 19:53 spi0.1 -> ../../../devices/platform/soc/20204000.spi/spi_master/spi0/spi0.1

# add fbtft device
$ sudo modprobe fbtft_device custom name=fb_ssd1322 width=100 height=100 speed=16000000 gpios=dc:25

$ dmesg | tail
[  164.290735] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[  164.303145] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[  164.306791] spidev spi0.0: spidev spi0.0 500kHz 8 bits mode=0x00
[  164.306817] spidev spi0.1: spidev spi0.1 500kHz 8 bits mode=0x00
[  164.306880] bcm2708_fb soc:fb: soc:fb id=-1 pdata? no
[  164.306930] spidev spi0.0: Deleting spi0.0
[  164.308044] fbtft_device: GPIOS used by 'fb_ssd1322':
[  164.308058] fbtft_device: 'dc' = GPIO25
[  164.308081] spidev spi0.1: spidev spi0.1 500kHz 8 bits mode=0x00
[  164.308097] spi spi0.0: fb_ssd1322 spi0.0 16000kHz 8 bits mode=0x00

$ cat /sys/bus/spi/devices/spi0.0/modalias
spi:fb_ssd1322

```
There's no driver that supports fb_ssd1322 in my system, so no binding/probing happens. Thus no ```graphics: ``` message.

alidaf, 2016-11-11:
Hi notro,
I have managed to compile, load and test now. Thanks for your patience and help. I have written a wiki guide (for myself, but it may be useful for others) on how to compile for the SSD1322 (and possibly for other additions).

You can find it here...https://github.com/alidaf/raspberryPi/wiki/SSD1322(SPI)---notro's-fbtft-framebuffer-driver.

I'll close this issue and then try to start using the driver in my own project.

Many thanks.

alidaf, 2016-11-11:
That link needs the full-stop at the end to work!

closed 2016-11-11 by alidaf