notro/fbtft #425: ST7789V based display

tr1p1ea, 2016-11-15:
I have a small display that is based on the ST7789V driver.

I haven't seen support in this wiki but I have seen it here:

https://github.com/torvalds/linux/tree/master/drivers/staging/fbtft

The datasheets for the display is here:

http://tr1p1ea.net/files/downloads/hardware/lcd/F122US01A-spec.pdf

Is there any way to get this working in Raspbian Jessie (with PIXEL) - kernel 4.4?

Thanks heaps for your help.

notro, 2016-11-15:
That driver is not enabled in the official Pi kernel:
```
$ sudo modprobe configs
$ zgrep FB_TFT_ST7789V /proc/config.gz
# CONFIG_FB_TFT_ST7789V is not set
```
You can create an issue and ask for it to be enabled: https://github.com/raspberrypi/linux/issues

Or you can use the flexfb driver which requires you to construct an init= sequence from the driver source code:
```
modprobe fbtft_device name=flexfb speed=16000000 gpios=reset:xx,dc:xx
modprobe flexfb width=240 height=320 init=-1,...,-3
```
https://github.com/notro/fbtft/wiki/flexfb
https://github.com/notro/fbtft/wiki/fbtft_device

tr1p1ea, 2016-11-15:
OK The flexfb option might work since the command set between the ST7735R and the ST7789V seem to be equivalent, so its setaddrwin should work as expected. I have been provided with the init code for the display:

`Void ST7789V_GP012_initial(void)
{
//-----------------------------------ST7789S reset sequence------------------------------------//
LCD_RESET=1;
Delayms(1); //Delay 1ms
LCD_RESET=0;
Delayms(10); //Delay 10ms
LCD_RESET=1;
Delayms(120); //Delay 120ms
//---------------------------------------------------------------------------------------------------//
WriteCommand(0x11);
Delayms(120); //Delay 120ms
//-------------------------------display and color format setting-----------------------------//
WriteCommand(0x36); 
Write_Data (0x00);

WriteCommand(0x3A);  
Write_Data (0x06);//  #06
//--------------------------------ST7789S Frame rate setting----------------------------------//
WriteCommand(0xB2);  
Write_Data (0x0C);  
Write_Data (0x0C);  
Write_Data (0x00);  
Write_Data (0x33);  
Write_Data (0x33);

WriteCommand(0xB7);  
Write_Data (0x75);
//---------------------------------ST7789S Power setting--------------------------------------//

WriteCommand(0xBB);  
Write_Data (0x21);

WriteCommand(0xC0);  
Write_Data (0x2C);

WriteCommand(0xC2);  
Write_Data (0x01);

WriteCommand(0xC3); 
Write_Data (0x17);

WriteCommand(0xC4);  
Write_Data (0x20);

WriteCommand(0xC6);  
Write_Data (0x0A);

WriteCommand(0xD0);  
Write_Data (0xA4);  
Write_Data (0xA1);
//--------------------------------ST7789S gamma setting---------------------------------------//
WriteCommand(0xE0);  
Write_Data (0xD0);  
Write_Data (0x0F);  
Write_Data (0x17);  
Write_Data (0x0D);  
Write_Data (0x0E);  
Write_Data (0x02);  
Write_Data (0x3D);  
Write_Data (0x44);  
Write_Data (0x4E);  
Write_Data (0x0C);  
Write_Data (0x1B);  
Write_Data (0x1B);  
Write_Data (0x1C);  
Write_Data (0x1F);

WriteCommand(0xE1);  
Write_Data (0xD0);  
Write_Data (0x0F);  
Write_Data (0x17);  
Write_Data (0x0D);  
Write_Data (0x0E);  
Write_Data (0x02);  
Write_Data (0x3D);  
Write_Data (0x44);  
Write_Data (0x4E);  
Write_Data (0x0C);  
Write_Data (0x1B);  
Write_Data (0x1B);  
Write_Data (0x1C);  
Write_Data (0x1F);

WriteCommand(0x29);
}
`
Which I have translated below, the only issue is what to do with the LCD_RESET at the beginning? Do you think this is a flag to toggle the reset on the module? Otherwise I have the below:

`modprobe fbtft_device name=flexfb speed=16000000 gpios=reset:24,dc:25
modprobe flexfb width=240 height=240 init=-1,0x11,-2,120,-1,0x36,0x00,-1,0x3A,0x06,-1,0xB2,0x0C,0x0C,0x00,0x33,0x33,-1,0xB7,0x75,-1,0xBB,0x21,-1,0xC0,0x2C,-1,0xC2,0x01,-1,0xC3,0x17,-1,0xC4,0x20,-1,0xC6,0x0A,-1,0xD0,0xA4,0xA1,-1,0xE0,0xD0,0x0F,0x17,0x0D,0x0E,0x02,0x3D,0x44,0x4E,0x0C,0x1B,0x1B,0x1C,0x1F,-1,0xE1,0xD0,0x0F,0x17,0x0D,0x0E,0x02,0x3D,0x44,0x4E,0x0C,0x1B,0x1B,0x1C,0x1F,-1,0x29,-3`

notro, 2016-11-15:
LCD_RESET seems to be controlling the reset pin. Like what fbtft does with the 'reset' gpio: http://lxr.free-electrons.com/ident?i=fbtft_reset

tr1p1ea, 2016-11-15:
OK so it looks like fbtft has it's own device reset/enable sequence? It should be suitable with any luck.

I'll give the felxfb option a try today and report back.

Thanks heaps for your help on this!

notro, 2018-05-10:
Closing issue since there has been no activity for more than 2 months. Reopen if needed.

closed 2018-05-10 by notro

tr1p1ea, 2018-05-10:
Hi sorry all, yes I did manage to get the display working with flexfb as per the below:

This is the init code that I use with flexfb:

modprobe fbtft_device name=flexfb speed=16000000 gpios=reset:25,dc:24
modprobe flexfb setaddrwin=0 width=240 height=240 init=-1,0x11,-2,120,-1,0x36,0x00,-1,0x3A,0x05,-1,0xB2,0x0C,0x0C,0x00,0x33,0x33,-1,0xB7,0x35,-1,0xBB,0x1A,-1,0xC0,0x2C,-1,0xC2,0x01,-1,0xC3,0x0B,-1,0xC4,0x20,-1,0xC6,0x0F,-1,0xD0,0xA4,0xA1,-1,0x21,-1,0xE0,0x00,0x19,0x1E,0x0A,0x09,0x15,0x3D,0x44,0x51,0x12,0x03,0x00,0x3F,0x3F,-1,0xE1,0x00,0x18,0x1E,0x0A,0x09,0x25,0x3F,0x43,0x52,0x33,0x03,0x00,0x3F,0x3F,-1,0x29,-3

DRAgon734465502, 2018-06-26:
I get the LCD also , but how can I to rotate the display ? 
I set :
-1,0x36,0x00,-1,   (0 degree)
and 
-1,0x36,0x60,-1,  (90 degree)

But I have tried anything , I can not get  180 degree display and  270 degree display

notro, 2018-06-27:
Do you swap width/height when trying 180/270?

buzztiaan, 2018-07-28:
its a square display @notro , 240x240 pixels

MaikB85, 2018-09-02:
Hi,

This works great for my Adafruit 1,54 240x240 Display (I think it ist the same Display which tr1p1ea uses):
> modprobe fbtft_device name=flexfb speed=32000000 gpios=reset:25,dc:24
modprobe flexfb setaddrwin=0 width=240 height=240 init=-1,0x11,-2,120,-1,0x36,0x00,-1,0x3A,0x05,-1,0xB2,0x0C,0x0C,0x00,0x33,0x33,-1,0xB7,0x35,-1,0xBB,0x1A,-1,0xC0,0x2C,-1,0xC2,0x01,-1,0xC3,0x0B,-1,0xC4,0x20,-1,0xC6,0x0F,-1,0xD0,0xA4,0xA1,-1,0x21,-1,0xE0,0x00,0x19,0x1E,0x0A,0x09,0x15,0x3D,0x44,0x51,0x12,0x03,0x00,0x3F,0x3F,-1,0xE1,0x00,0x18,0x1E,0x0A,0x09,0x25,0x3F,0x43,0x52,0x33,0x03,0x00,0x3F,0x3F,-1,0x29,-3

How can I use this to start automatically on boot?

aewatt95, 2020-02-11:
Thanks a lot for this. Just one addition, as this took a few hours to get it right:
I experimented with this display module with NO CS pin on an orange pi zero with Armbian Linux 4.19.62. If you encounter a blank screen, try this:

`sudo modprobe fbtft_device name=flexfb speed=32000000 gpios=reset:3,dc:0 busnum=1 mode=3`

The important part is to set the **spi mode to 3**. 
If you are on an orange pi or want to use another spibus, set the busnum accordingly.