notro/fbtft #426: SSD1331 display is flipped horizontally.

grndkntrl, 2016-11-20:
Hi,

I just managed to get a SSD1331 based 0.95" OLED up and running on my Raspberry Pi Zero (Raspbian Jessie) using the Framebuffer mirroring method (https://github.com/notro/fbtft/wiki/Framebuffer-use) and everything is great so far, apart from the fact that the display is flipped about the horizontal axis.

It's not rotated, but just flipped like you would see if you looked at it in a actual mirror held to one side (left/right).

Have I missed a step somewhere?

here's the relevant dmesg output section:

`[    4.131185] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[    4.141097] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[    4.142933] spidev spi0.0: spidev spi0.0 500kHz 8 bits mode=0x00
[    4.142974] spidev spi0.1: spidev spi0.1 500kHz 8 bits mode=0x00
[    4.143037] bcm2708_fb soc:fb: soc:fb id=-1 pdata? no
[    4.143111] spidev spi0.0: Deleting spi0.0
[    4.144246] fbtft_device: GPIOS used by 'ssd1331':
[    4.144274] fbtft_device: 'reset' = GPIO24
[    4.144290] fbtft_device: 'dc' = GPIO25
[    4.144318] spidev spi0.1: spidev spi0.1 500kHz 8 bits mode=0x00
[    4.144345] spi spi0.0: fb_ssd1331 spi0.0 20000kHz 8 bits mode=0x03
[    6.990711] fb_ssd1331: module is from the staging directory, the quality is unknown, you have been warned.
[    9.157676] graphics fb1: fb_ssd1331 frame buffer, 96x64, 12 KiB video memory, 4 KiB DMA buffer memory, fps=20, spi0.0 at 20 MHz`

bar1792, 2016-11-21:
By horizontally flipped do you mean such as the image in this link?:
https://docs.gimp.org/en/gimp-layer-flip-horizontal.html

Ironically, this is exactly what I am looking to do for my current display.

grndkntrl, 2016-11-21:
Yep, exactly that. I looked at the SSD1331 datasheet and there is a option there for flipping the display about an axis but I couldn't find any way of telling the baked in release of FBTFT to use the correct orientation, unless I've missed something.

bar1792, 2016-11-21:
That is one thing I was hoping was an easily modifiable setting in the FBTFT, in a similar vein in how the raspberry pi has a /boot/config.txt where one could simply write:
display_rotate=1,2,3, 0x20000, or 0x10000

bar1792, 2016-11-21:
I found this that may be what you're looking for, I haven't had any time to read through it thoroughly but it may work:

https://github.com/notro/fbtft/wiki/Touchpanel#watterott-mi0283qt-9a

notro, 2016-11-22:
This seems to be a similar case #300

grndkntrl, 2016-11-22:
Hmmm, so given that I'm using the baked-into-the-kernel FBTFT, how do I set the mirror flag? There's no dts file that I can modify, so if I create one won't that conflict with the built-in fb_ssd1331.ko kernel module or just get ignored?

notro, 2016-11-22:
You can use the fbtft_device init= argument.
https://github.com/notro/fbtft/wiki/fbtft_device#parameters

closed 2018-05-10 by notro