notro/fbtft #428: How to use fbtft for TFT01-2.2S LCD?

rezaee, 2016-12-02:
Hi
I have an arm-based board(nanopi m1)that has 40 pins GPIO similar to raspberry pi and Debian Jessie OS. also I have this display: http://www.elecfreaks.com/wiki/index.php?title=2.2S%22_TFT_LCD:_TFT01-2.2S and want to connect it directly to my board. 
May you help me how can I do that?

notro, 2016-12-02:
That display is mentioned in the wiki: https://github.com/notro/fbtft/wiki/More-LCD-Modules#tft01-22sp
Check that your kernel supports fbtft:
```
$ modinfo fbtft
filename:       /lib/modules/4.9.0-rc5-v7+/kernel/drivers/staging/fbtft/fbtft.ko
license:        GPL
srcversion:     3EFDADD511D35BFEE3B20A7
depends:
staging:        Y
intree:         Y
vermagic:       4.9.0-rc5-v7+ SMP mod_unload modversions ARMv7 p2v8
parm:           debug:override device debug level (ulong)
parm:           dma:Use DMA buffer (bool)
```
fbtft was added to linux in 4.10.

rezaee, 2016-12-03:
Hi notro and very thank you for reply.
But I am newbie in linux and don't know how to check fbtft support?
Are armbian or raspbian OS in they last version support that? If yes, I want to download and trying them on my board if it's possible!

rezaee, 2016-12-03:
this is my OS information and I think there is no support for fbtft:
`Linux FriendlyARM 3.4.39-h3 #76 SMP PREEMPT Wed Sep 7 13:46:05 CST 2016 armv7l GNU/Linux`

notro, 2016-12-03:
This command will tell if you have fbtft support: modinfo fbtft_device
The fbtft module can be builtin, but fbtft_device is most likely not.
This will show which drivers are enabled if any:
```
$ zgrep CONFIG_FB_TFT /proc/config.gz
CONFIG_FB_TFT=m
<snip>
```

AFAICT this is the tree they build the kernel from, and it does have fbtft:
https://github.com/friendlyarm/h3_lichee/tree/master/linux-3.4/drivers/staging/fbtft

I don't know which defconfig is used, but this one for instance have only 2 drivers enabled:
https://github.com/friendlyarm/h3_lichee/blob/97e1a057000511abe8ed85adee6c6474398a427c/linux-3.4/arch/arm/configs/sun8iw7p1smp_defconfig#L2870

rezaee, 2016-12-04:
This is the resault of `modinfo fbtft`:
`modinfo: ERROR: Module fbtft not found.`

And this is the resaults of `modinfo fbtft_device`
```
filename:       /lib/modules/3.4.39-h3/kernel/drivers/staging/fbtft/fbtft_device.ko
license:        GPL
author:         Noralf Tronnes
description:    Add a FBTFT device.
depends:        
staging:        Y
intree:         Y
vermagic:       3.4.39-h3 SMP preempt mod_unload ARMv7 p2v8 
parm:           name:Devicename (required). name=list => list all supported devices. (charp)
parm:           rotate:Angle to rotate display counter clockwise: 0, 90, 180, 270 (uint)
parm:           busnum:SPI bus number (default=0) (uint)
parm:           cs:SPI chip select (override device default) (uint)
parm:           speed:SPI speed (override device default) (uint)
parm:           mode:SPI mode (override device default) (int)
parm:           gpios:List of gpios. Comma separated with the form: reset:23,dc:24 (when overriding the default, all gpios must be specified) (charp)
parm:           fps:Frames per second (override driver default) (uint)
parm:           gamma:String representation of Gamma Curve(s). Driver specific. (charp)
parm:           txbuflen:txbuflen (override driver default) (int)
parm:           bgr:BGR bit (supported by some drivers). (int)
parm:           startbyte:Sets the Start byte used by some SPI displays. (uint)
parm:           custom:Add a custom display device. Use speed= argument to make it a SPI device, else platform_device (bool)
parm:           width:Display width, used with the custom argument (uint)
parm:           height:Display height, used with the custom argument (uint)
parm:           buswidth:Display bus width, used with the custom argument (uint)
parm:           init:Init sequence, used with the custom argument (array of int)
parm:           debug:level: 0-7 (the remaining 29 bits is for advanced usage) (ulong)
parm:           verbose:0 silent, >0 show gpios, >1 show devices, >2 show devices before (default=3) (uint)

```
And unfortunately I have no 'config.gz' :` zgrep CONFIG_FB_TFT /proc/config.gz`
`gzip: /proc/config.gz: No such file or directory`

I only have 'config-3.4.39-h3.new' in /boot directory that it's an empty file I think!

rezaee, 2016-12-04:
What's the meaning of these lines?
```
    Add to drivers/video/fbdev/Kconfig:   source "drivers/video/fbdev/fbtft/Kconfig"
    Add to drivers/video/fbdev/Makefile:  obj-y += fbtft/
```

I have no `Kconfig` and `Makefile` in that location!

alidaf, 2016-12-04:
@rezaee Have a look at my wiki for using the SSD1322.

https://github.com/alidaf/raspberryPi/wiki/SSD1322(SPI)---notro's-fbtft-framebuffer-driver

It's a different display but I've tried to consolidate all of Notro's guides into a single page that should work for any display by changing the references to the actual module. I had to work through it all empirically and this is what worked for me in the end. There is a section that deals with those two lines but it depends on whether you use the staging location or not. I chose not to because I determined that the Kconfig and Makefile were different in different locations and this is how I ended up understanding it all.

notro, 2016-12-04:
> And unfortunately I have no 'config.gz'

Maybe the config is available as a module: sudo modprobe configs
Or just look in the folder to see which drivers are available: /lib/modules/3.4.39-h3/kernel/drivers/staging/fbtft/

If some fbtft drivers are built into the kernel image/binary, they would show up here: /sys/bus/spi/drivers and here: /sys/module/ (a module registers a driver).

> I have no Kconfig and Makefile in that location!

They were used before fbtft was added to staging, so some kernel trees have fbtft in that place.

rezaee, 2016-12-04:
No the result of `sudo modprobe configs` is `modprobe: FATAL: Module configs not found.`
In this path:  /lib/modules/3.4.39-h3/kernel/drivers/staging/fbtft/  there is only one file named `fbtft_device.ko`.

In this path: /sys/bus/spi/drivers  there are three folders: fb_st7735s ,  fb_st7789s , spidev .
And inside this path: /sys/module/  there are many folders that "spidev" is one of them!

rezaee, 2016-12-04:
There are two folders in: /lib/modules : `3.4.39-h3 and  3.18.1+`
In this path: /lib/modules/3.18.1+/kernel/drivers/video/fbdev/fbtft/fbtft I have `fb_ili9341.c`

notro, 2016-12-04:
I guess your options are either to build your own kwernel with the driver enabled, or lobby to get the drivers enabled in the defconfig used to build your kernel.
I looked at fb_st7735s and fb_st7789s, and they can't be made to work with your display. fbtft_device is quite flexible so it can override the init sequence, but the set_var() function sets a different value to the 0x36 register which controls mirror/flip/rotation.

rezaee, 2016-12-04:
So, how can I make working my display? 
I am newbie in linux and don't know which solution is better! 
May you tell me what do I should do step by step to make it working?

notro, 2016-12-04:
As I said, you need to build a new kernel with the driver enabled.
If adafruit22a is the one that works with this display, you need the fb_ili9340 driver.
http://lxr.free-electrons.com/source/drivers/staging/fbtft/fbtft_device.c#L338

I have no experience with nanopi. http://nanopi.io/support.html

rezaee, 2016-12-04:
If I chnge my OS to armbian, will the problem be solved?
As I asked in their forum, it has fbtft support in all the kernels!

rezaee, 2016-12-05:
I have changed my OS to Armbian and now I have the kernel support of fbtft. So what's the next step?

```
root@nanopim1:/home/m# modinfo fbtft
filename:       /lib/modules/3.4.112-sun8i/kernel/drivers/video/fbtft/fbtft.ko
license:        GPL
depends:        fb_sys_fops,syscopyarea,sysfillrect,sysimgblt
intree:         Y
vermagic:       3.4.112-sun8i SMP preempt mod_unload modversions ARMv7 p2v8 
parm:           debug:override device debug level (ulong)
parm:           dma:Use DMA buffer (bool)

```

rezaee, 2016-12-05:
I did this command: 
`modprobe fbtft_device custom name=fb_ili9341 fbtft_device.gpios=reset:25,dc:24,led:18 fbtft_device.speed=16000000 fbtft_device.rotate=90 fbtft_device.bgr=1 fbtft_device.custom=1`

And the result of dmesg after that was:
```
[  361.544784] ---[ end trace 4484803e1c8239d6 ]---
[  361.544891] fb_ili9341: probe of spi0.0 failed with error -22
[  361.618986] fbtft_device:    'dc' = GPIO24
[  361.623624] fbtft_device:    'led' = GPIO18
[  361.628266] fbtft_device:  SPI devices registered:
[  361.633616] fbtft_device:      fb_ili9341 spi0.0 16000kHz 8 bits mode=0x00

```
I have checked all the pins and they're connected true!

notro, 2016-12-05:
The info before ```end trace``` is important.
Add this option to (maybe) get some more info: ```debug=7```

rezaee, 2016-12-05:
Where should I add `debug=7` ?

This is the` dmesg` result:
```

[  216.512005] fbtft_device:  SPI devices registered:
[  216.512027] fbtft_device:      spidev spi0.0 33000kHz 8 bits mode=0x00
[  216.512038] fbtft_device:  'fb' Platform devices registered:
[  216.512132] fbtft_device: Deleting spi0.0
[  216.512566] fbtft_device:  GPIOS used by 'fb_ili9341':
[  216.512579] fbtft_device:    'reset' = GPIO25
[  216.512587] fbtft_device:    'dc' = GPIO24
[  216.512595] fbtft_device:    'led' = GPIO18
[  216.512603] fbtft_device:  SPI devices registered:
[  216.512613] fbtft_device:      fb_ili9341 spi0.0 16000kHz 8 bits mode=0x00
[  216.519793] fb_ili9341 spi0.0: fbtft_request_gpios: gpio_request_one('reset'=25) failed with -22
[  216.519816] ------------[ cut here ]------------
[  216.519841] WARNING: at drivers/gpio/gpiolib.c:104 gpio_ensure_requested+0x54/0xb4()
[  216.519852] autorequest GPIO-25
[  216.519858] Modules linked in: fb_ili9341(+) fbtft_device fbtft syscopyarea sysfillrect sysimgblt fb_sys_fops rfcomm bnep bluetooth mali_drm drm sunxi_cir mali ump gpio_sunxi [last unloaded: scsi_wait_scan]
[  216.519946] [<c0016a20>] (unwind_backtrace+0x0/0xe8) from [<c0687a10>] (dump_stack+0x20/0x24)
[  216.519972] [<c0687a10>] (dump_stack+0x20/0x24) from [<c0029720>] (warn_slowpath_common+0x5c/0x74)
[  216.519994] [<c0029720>] (warn_slowpath_common+0x5c/0x74) from [<c0029778>] (warn_slowpath_fmt+0x40/0x48)
[  216.520017] [<c0029778>] (warn_slowpath_fmt+0x40/0x48) from [<c03a34e4>] (gpio_ensure_requested+0x54/0xb4)
[  216.520042] [<c03a34e4>] (gpio_ensure_requested+0x54/0xb4) from [<c03a3a10>] (gpio_direction_input+0x84/0x188)
[  216.520077] [<c03a3a10>] (gpio_direction_input+0x84/0x188) from [<bf0ee3f0>] (fbtft_free_gpios+0xb4/0xd4 [fbtft])
[  216.520120] [<bf0ee3f0>] (fbtft_free_gpios+0xb4/0xd4 [fbtft]) from [<bf0ef064>] (fbtft_register_framebuffer+0x2bc/0x2e8 [fbtft])
[  216.520156] [<bf0ef064>] (fbtft_register_framebuffer+0x2bc/0x2e8 [fbtft]) from [<bf0f0348>] (fbtft_probe_common+0x240/0x2b0 [fbtft])
[  216.520190] [<bf0f0348>] (fbtft_probe_common+0x240/0x2b0 [fbtft]) from [<bf1060ac>] (fbtft_driver_probe_spi+0x24/0x2c [fb_ili9341])
[  216.520221] [<bf1060ac>] (fbtft_driver_probe_spi+0x24/0x2c [fb_ili9341]) from [<c045a0fc>] (spi_drv_probe+0x28/0x2c)
[  216.520248] [<c045a0fc>] (spi_drv_probe+0x28/0x2c) from [<c042cf7c>] (driver_probe_device+0xd0/0x20c)
[  216.520271] [<c042cf7c>] (driver_probe_device+0xd0/0x20c) from [<c042d184>] (__driver_attach+0x78/0x9c)
[  216.520293] [<c042d184>] (__driver_attach+0x78/0x9c) from [<c042b500>] (bus_for_each_dev+0x7c/0xa0)
[  216.520315] [<c042b500>] (bus_for_each_dev+0x7c/0xa0) from [<c042ca3c>] (driver_attach+0x28/0x30)
[  216.520336] [<c042ca3c>] (driver_attach+0x28/0x30) from [<c042c5e0>] (bus_add_driver+0xf4/0x230)
[  216.520358] [<c042c5e0>] (bus_add_driver+0xf4/0x230) from [<c042d784>] (driver_register+0xac/0x124)
[  216.520380] [<c042d784>] (driver_register+0xac/0x124) from [<c045a554>] (spi_register_driver+0x54/0x68)
[  216.520408] [<c045a554>] (spi_register_driver+0x54/0x68) from [<bf108014>] (fbtft_driver_module_init+0x14/0x30 [fb_ili9341])
[  216.520436] [<bf108014>] (fbtft_driver_module_init+0x14/0x30 [fb_ili9341]) from [<c000867c>] (do_one_initcall+0xa8/0x164)
[  216.520463] [<c000867c>] (do_one_initcall+0xa8/0x164) from [<c0080740>] (sys_init_module+0xe40/0x1918)
[  216.520487] [<c0080740>] (sys_init_module+0xe40/0x1918) from [<c000df60>] (ret_fast_syscall+0x0/0x30)
[  216.520502] ---[ end trace e995b112b87c0787 ]---
[  216.520697] ------------[ cut here ]------------
[  216.520720] WARNING: at drivers/gpio/gpiolib.c:104 gpio_ensure_requested+0x54/0xb4()
[  216.520731] autorequest GPIO-24
[  216.520739] Modules linked in: fb_ili9341(+) fbtft_device fbtft syscopyarea sysfillrect sysimgblt fb_sys_fops rfcomm bnep bluetooth mali_drm drm sunxi_cir mali ump gpio_sunxi [last unloaded: scsi_wait_scan]
[  216.520824] [<c0016a20>] (unwind_backtrace+0x0/0xe8) from [<c0687a10>] (dump_stack+0x20/0x24)
[  216.520847] [<c0687a10>] (dump_stack+0x20/0x24) from [<c0029720>] (warn_slowpath_common+0x5c/0x74)
[  216.520868] [<c0029720>] (warn_slowpath_common+0x5c/0x74) from [<c0029778>] (warn_slowpath_fmt+0x40/0x48)
[  216.520891] [<c0029778>] (warn_slowpath_fmt+0x40/0x48) from [<c03a34e4>] (gpio_ensure_requested+0x54/0xb4)
[  216.520915] [<c03a34e4>] (gpio_ensure_requested+0x54/0xb4) from [<c03a3a10>] (gpio_direction_input+0x84/0x188)
[  216.520946] [<c03a3a10>] (gpio_direction_input+0x84/0x188) from [<bf0ee3f0>] (fbtft_free_gpios+0xb4/0xd4 [fbtft])
[  216.520982] [<bf0ee3f0>] (fbtft_free_gpios+0xb4/0xd4 [fbtft]) from [<bf0ef064>] (fbtft_register_framebuffer+0x2bc/0x2e8 [fbtft])
[  216.521019] [<bf0ef064>] (fbtft_register_framebuffer+0x2bc/0x2e8 [fbtft]) from [<bf0f0348>] (fbtft_probe_common+0x240/0x2b0 [fbtft])
[  216.521053] [<bf0f0348>] (fbtft_probe_common+0x240/0x2b0 [fbtft]) from [<bf1060ac>] (fbtft_driver_probe_spi+0x24/0x2c [fb_ili9341])
[  216.521080] [<bf1060ac>] (fbtft_driver_probe_spi+0x24/0x2c [fb_ili9341]) from [<c045a0fc>] (spi_drv_probe+0x28/0x2c)
[  216.521104] [<c045a0fc>] (spi_drv_probe+0x28/0x2c) from [<c042cf7c>] (driver_probe_device+0xd0/0x20c)
[  216.521127] [<c042cf7c>] (driver_probe_device+0xd0/0x20c) from [<c042d184>] (__driver_attach+0x78/0x9c)
[  216.521149] [<c042d184>] (__driver_attach+0x78/0x9c) from [<c042b500>] (bus_for_each_dev+0x7c/0xa0)
[... cut ...]
[  216.521908] [<c045a554>] (spi_register_driver+0x54/0x68) from [<bf108014>] (fbtft_driver_module_init+0x14/0x30 [fb_ili9341])
[  216.521934] [<bf108014>] (fbtft_driver_module_init+0x14/0x30 [fb_ili9341]) from [<c000867c>] (do_one_initcall+0xa8/0x164)
[  216.521958] [<c000867c>] (do_one_initcall+0xa8/0x164) from [<c0080740>] (sys_init_module+0xe40/0x1918)
[  216.521981] [<c0080740>] (sys_init_module+0xe40/0x1918) from [<c000df60>] (ret_fast_syscall+0x0/0x30)
[  216.521995] ---[ end trace e995b112b87c0789 ]---
[  216.522093] fb_ili9341: probe of spi0.0 failed with error -22

```

rezaee, 2016-12-05:
This is the full text of `dmesg` after this command: 

`sudo modprobe fbtft_device custom name=fb_ili9341  gpios=reset:25,dc:24,led:18 speed=16000000 rotate=90 bgr=1 debug=7
`

```
[    0.000000] Booting Linux on physical CPU 0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Linux version 3.4.112-sun8i (root@xenial) (gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.1) ) #14 SMP PREEMPT Wed Sep 14 20:29:31 CEST 2016
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine: sun8i
[    0.000000] cma: CMA: reserved 256 MiB at 50000000
[    0.000000] Memory policy: ECC disabled, Data cache writealloc
[    0.000000] On node 0 totalpages: 262144
[    0.000000] free_area_init_node: node 0, pgdat c0a24900, node_mem_map c0b5d000
[    0.000000]   Normal zone: 1710 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 192850 pages, LIFO batch:31
[    0.000000]   HighMem zone: 594 pages used for memmap
[    0.000000]   HighMem zone: 66990 pages, LIFO batch:15
[    0.000000] script_init enter!
[    0.000000] script_init exit!
[    0.000000] PERCPU: Embedded 8 pages/cpu @c1473000 s11968 r8192 d12608 u32768
[    0.000000] pcpu-alloc: s11968 r8192 d12608 u32768 alloc=8*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 259840
[    0.000000] Kernel command line: console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p1 rootwait rootfstype=ext4 cgroup_enable=memory swapaccount=1 panic=10 consoleblank=0 enforcing=0 loglevel=1
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] allocated 2097152 bytes of page_cgroup
[    0.000000] please try 'cgroup_disable=memory' option if you don't want memory cgroups
[    0.000000] Memory: 1024MB = 1024MB total
[    0.000000] Memory: 758708k/758708k available, 289868k reserved, 270336K highmem
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     vmalloc : 0xf0000000 - 0xff000000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xef800000   ( 760 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc095bce8   (9552 kB)
[    0.000000]       .init : 0xc095c000 - 0xc09aeec0   ( 332 kB)
[    0.000000]       .data : 0xc09b0000 - 0xc0a29750   ( 486 kB)
[    0.000000]        .bss : 0xc0a29f04 - 0xc0b5cf90   (1229 kB)
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Additional per-CPU info printed with stalls.
[    0.000000] NR_IRQS:544
[    0.000000] Architected local timer running at 24.00MHz.
[    0.000000] Switching to timer-based delay loop
[    0.000000] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 178956ms
[    0.000000] Console: colour dummy device 80x30
[    0.000000] console [tty1] enabled
[    0.000260] Calibrating delay loop (skipped), value calculated using timer frequency.. 4800.00 BogoMIPS (lpj=24000000)
[    0.000283] pid_max: default: 32768 minimum: 301
[    0.000623] Mount-cache hash table entries: 512
[    0.001589] Initializing cgroup subsys cpuacct
[    0.001604] Initializing cgroup subsys memory
[    0.001652] Initializing cgroup subsys devices
[    0.001663] Initializing cgroup subsys freezer
[    0.001674] Initializing cgroup subsys blkio
[    0.001698] Initializing cgroup subsys perf_event
[    0.001778] CPU: Testing write buffer coherency: ok
[    0.001823] ftrace: allocating 25740 entries in 76 pages
[    0.030294] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.030312] [sunxi_smp_prepare_cpus] enter
[    0.030349] Setting up static identity map for 0x40691938 - 0x40691990
[    0.031292] CPU1: Booted secondary processor
[... 528 lines cut ...]
[   76.218442] [<c0016a20>] (unwind_backtrace+0x0/0xe8) from [<c0687a10>] (dump_stack+0x20/0x24)
[   76.218464] [<c0687a10>] (dump_stack+0x20/0x24) from [<c0029720>] (warn_slowpath_common+0x5c/0x74)
[   76.218484] [<c0029720>] (warn_slowpath_common+0x5c/0x74) from [<c0029778>] (warn_slowpath_fmt+0x40/0x48)
[   76.218506] [<c0029778>] (warn_slowpath_fmt+0x40/0x48) from [<c03a34e4>] (gpio_ensure_requested+0x54/0xb4)
[   76.218528] [<c03a34e4>] (gpio_ensure_requested+0x54/0xb4) from [<c03a3a10>] (gpio_direction_input+0x84/0x188)
[   76.218556] [<c03a3a10>] (gpio_direction_input+0x84/0x188) from [<bf0ef3f0>] (fbtft_free_gpios+0xb4/0xd4 [fbtft])
[   76.218591] [<bf0ef3f0>] (fbtft_free_gpios+0xb4/0xd4 [fbtft]) from [<bf0f0064>] (fbtft_register_framebuffer+0x2bc/0x2e8 [fbtft])
[   76.218627] [<bf0f0064>] (fbtft_register_framebuffer+0x2bc/0x2e8 [fbtft]) from [<bf0f1348>] (fbtft_probe_common+0x240/0x2b0 [fbtft])
[   76.218660] [<bf0f1348>] (fbtft_probe_common+0x240/0x2b0 [fbtft]) from [<bf1070ac>] (fbtft_driver_probe_spi+0x24/0x2c [fb_ili9341])
[   76.218687] [<bf1070ac>] (fbtft_driver_probe_spi+0x24/0x2c [fb_ili9341]) from [<c045a0fc>] (spi_drv_probe+0x28/0x2c)
[... cut ...]
[   76.218862] [<c045a554>] (spi_register_driver+0x54/0x68) from [<bf109014>] (fbtft_driver_module_init+0x14/0x30 [fb_ili9341])
[   76.218888] [<bf109014>] (fbtft_driver_module_init+0x14/0x30 [fb_ili9341]) from [<c000867c>] (do_one_initcall+0xa8/0x164)
[   76.218911] [<c000867c>] (do_one_initcall+0xa8/0x164) from [<c0080740>] (sys_init_module+0xe40/0x1918)
[   76.218933] [<c0080740>] (sys_init_module+0xe40/0x1918) from [<c000df60>] (ret_fast_syscall+0x0/0x30)
[   76.218946] ---[ end trace 53050e35f955d0bd ]---
[   76.219036] fb_ili9341: probe of spi0.0 failed with error -22

```

notro, 2016-12-05:
Ok, [gpio_request_one()](http://lxr.free-electrons.com/source/drivers/gpio/gpiolib.c?v=3.4#L1284) fails with -EINVAL (-22) somewhere down the line.

Let's try and control the gpio from userspace. This should turn on backlight, run as root:
```
echo 18 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio18/direction
echo 1 > /sys/class/gpio/gpio18/value
```

rezaee, 2016-12-05:
No, nothing happend!

rezaee, 2016-12-05:
This is contents of `/etc/modules `: 
```
#gpio_sunxi
#w1-sunxi
#w1-gpio
#w1-therm
#sunxi-cir
fbtft_device custom name=fb_ili9341  gpios=reset:25,dc:24,led:18 speed=16000000 rotate=90 bgr=1
```

Maybe I must change something here?

notro, 2016-12-05:
> No, nothing happend!

Nothing in the kernel log?

Try removing the comment in front of gpio_sunxi.
See: http://linux-sunxi.org/GPIO#Accessing_the_GPIO_pins_through_sysfs_on_sunxi-3.4

rezaee, 2016-12-05:
Nothing in terminal and LCD happend. it made a folder named gpio18 in /sys/class/gpio.
I did remove the front comment of gpio_sunxi and reboot my system amd try the commands again but nothing happend!

notro, 2016-12-05:
> it made a folder named gpio18 in /sys/class/gpio

If you can't control the gpio using this interface, then fbtft won't work. They use the same functions underneath.

rezaee, 2016-12-05:
I asked about the problem in armbian forum and hope to find the solution!
Very thank you for your help and so sorry I took your time!

rezaee, 2016-12-06:
I think I found the problem! they are not same GPIO pins as they are in RaspberryPi3 in Nanopi-M1 (although they say this in they're advertise!!). you can see GPIO pins of nanopi-m1 here: http://wiki.friendlyarm.com/wiki/index.php/NanoPi_M1

I changed the command to this(without changing wiring):
`sudo modprobe fbtft_device custom name=fb_ili9341 gpios=reset:1,dc:201,led:6 speed=16000000 rotate=90 bgr=1`

And the problem changed to new Error: 
```
[ 1626.941908] fbtft_device:  SPI devices registered:
[ 1626.941929] fbtft_device:      spidev spi0.0 33000kHz 8 bits mode=0x00
[ 1626.941940] fbtft_device:  'fb' Platform devices registered:
[ 1626.942033] fbtft_device: Deleting spi0.0
[ 1626.942462] fbtft_device:  GPIOS used by 'fb_ili9341':
[ 1626.942473] fbtft_device:    'reset' = GPIO1
[ 1626.942482] fbtft_device:    'dc' = GPIO201
[ 1626.942489] fbtft_device:    'led' = GPIO6
[ 1626.942497] fbtft_device:  SPI devices registered:
[ 1626.942508] fbtft_device:      fb_ili9341 spi0.0 16000kHz 8 bits mode=0x00
[ 1627.047324] fb_ili9341 spi0.0: fbtft_request_gpios: gpio_request_one('reset'=1) failed with -16
[ 1627.047442] fb_ili9341: probe of spi0.0 failed with error -16

```
I am trying to find the problem, and then trying to run the LCD with your help!
Thank you again in advanced.

rezaee, 2016-12-06:
I checked again these codes with equal pin number in my board(6) and it worked! and turned on my LCD backlight!

echo 6 > /sys/class/gpio/export
echo out > /sys/class/gpio/gpio6/direction
echo 1 > /sys/class/gpio/gpio6/value

rezaee, 2016-12-07:
I think there is a problem with fbtft code!
because I can export and unexport every GPIO pins but when I try to use LCD I get this error:
```
fb_ili9341 spi0.0: fbtft_request_gpios: gpio_request_one('reset'=200) failed with -16
[  101.058042] fb_ili9341: probe of spi0.0 failed with error -16
```
I changed gpio pins but the error is the same! but testing gpio pins with export command works!

rezaee, 2016-12-07:
I commented again `#gpio_sunxi` in `/etc/modules` and the problems solved. I have no errors at this time:

```
[   95.158413] fbtft_device:  SPI devices registered:
[   95.163798] fbtft_device:      spidev spi0.0 33000kHz 8 bits mode=0x00
[   95.171075] fbtft_device:  'fb' Platform devices registered:
[   95.177424] fbtft_device: Deleting spi0.0
[   95.182402] fbtft_device:  GPIOS used by 'fb_ili9341':
[   95.188130] fbtft_device:    'reset' = GPIO1
[   95.192970] fbtft_device:    'dc' = GPIO201
[   95.197616] fbtft_device:    'led' = GPIO6
[   95.202216] fbtft_device:  SPI devices registered:
[   95.207541] fbtft_device:      fb_ili9341 spi0.0 16000kHz 8 bits mode=0x00
[   95.513259] graphics fb8: fb_ili9341 frame buffer, 320x240, 150 KiB video memory, 16 KiB buffer memory, fps=20, spi0.0 at 16 MHz

```

But the new problem is my LCD only shows a blanck white sceern and nothing else! it seems only led backlight command works truely!

rezaee, 2016-12-07:
This is the result of:  `dmesg | grep graphics`

`[   78.466083] graphics fb8: fb_ili9341 frame buffer, 320x240, 150 KiB video memory, 16 KiB buffer memory, fps=20, spi0.0 at 16 MHz`

rezaee, 2016-12-07:
I tried this commands:
```
sudo mkdir -p /etc/X11/xorg.conf.d/
sudo wget https://gist.github.com/notro/63fc8f563b5f85010d30/raw/5e68d6c6df9471f973b3da5305e73a5017590a69/fbtft.conf -O /etc/X11/xorg.conf.d/fbtft.conf
```

because` FRAMEBUFFER=/dev/fb1 startx  `and `con2fbmap 1 1 `didn't work. but now the main display(HDMI) changed and went to blank screen and also TFT screen doesn't work!

And now I have no screen and I think I must reinstall my OS again! :(

notro, 2016-12-07:
> But the new problem is my LCD only shows a blanck white sceern and nothing else!

This means that for some reason the controller has not received the initialization sequence, or the sequence is not correct.
You should get a black backlit display.

> [ 78.466083] graphics fb8:

Your framebuffer is /dev/fb8

I find fbcon to be the best and least intrusive way to test if a display is working.
In your case: ```con2fbmap 1 8```

rezaee, 2016-12-07:
`con2fbmap 1 8` doesn't work! I tested it wit con2fbmap 1 (0to8) and non of them worked!

notro, 2016-12-07:
No, I understand that. As long as the display is white it won't work. I just pointed to an easy way to test the display, instead of messing around with xorg.

rezaee, 2016-12-07:
Very very thank you for your patient and helps!
I shame to say but the problem was about incorrect wiring SLK and MISO! because my focus was about GPIO pins I didn't check them.

Now the display works very nice!

I have some questions:
when I change the desktop with `FRAMEBUFFER=/dev/fb8 startx` and then want to go back to HDMI display, I use `FRAMEBUFFER=/dev/fb0 startx` or` FRAMEBUFFER=/dev/fb1 startx`but TFT LCD freezes when display changes. why it happens and can I fix it?

Other question is How can I use touchpanel? does my display have touchpanel support? if yes, how can I activate ir?

notro, 2016-12-08:
I can't help you with X, I only use it occasionally to test displays.
No touchpanel support on this display.

closed 2018-05-10 by notro