notro/fbtft #429: Can't boot after install fbtft!

michaelgino, 2016-12-02:
Hi,

I bought the 128×128 OLED screen from you, and I want to use with Raspbian on Pi 3 B model, like here, http://www.freetronics.com.au/pages/oled128-quickstart-guide-raspberry-pi#.WEBAqfkrL1c

I just start to try the setup, but after the first step,

sudo wget https://raw.github.com/Hexxeh/rpi-update/master/rpi-update -O /usr/bin/rpi-update
sudo chmod +x /usr/bin/rpi-update
sudo REPO_URI=https://github.com/notro/rpi-firmware rpi-update

The Pi can’t boot anymore, after the logo everything was black :/
Can anyone help me?

Thank you very much!

notro, 2016-12-02:
The rpi-firmware kernel is deprecated and predates Pi3, so this quickstart is clearly outdated.
The official Pi kernel has support for fbtft now, so there's no need to update it.

There's some info here: https://github.com/notro/fbtft/wiki#raspberry-pi-spi-dma

If you can't make it work, this is from the quickstart:
> Please direct any support inquiries regarding the OLED128 to us at Freetronics (forums or email), not to notro.

michaelgino, 2016-12-04:
Hi, 

Thank you very much for the quick answer! With your help I succsessfully sloved this part of the problem, and I tried with on Pi2 too.
Now, everithing seems work fine, but when I login from SHH, and write the `FRAMEBUFFER=/dev/fb1` `startx` command, I get this back;

`pi@raspberrypi:~` $ FRAMEBUFFER=/dev/fb1 startx

X.Org X Server 1.17.2
Release Date: 2015-06-16
X Protocol Version 11, Revision 0
Build Operating System: Linux 3.8.11-2-ARCH armv7l Raspbian
Current Operating System: Linux raspberrypi 4.0.7-v7+ #1 SMP PREEMPT Sat Jul 11 20:44:05 CEST 2015 armv7l
Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2709.boardrev=0xa01041 bcm2709.serial=0xb664b8da smsc95xx.macaddr=B8:27:EB:64:B8:DA bcm2708_fb.fbswap=1 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
Build Date: 11 January 2016  01:32:44PM
xorg-server 2:1.17.2-1+rpi1 (http://www.debian.org/support)
Current version of pixman: 0.33.3
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.1.log", Time: Sun Dec  4 16:32:38 2016
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found.
The XKEYBOARD keymap compiler (xkbcomp) reports:
> Warning:          Type "ONE_LEVEL" has 1 levels, but <RALT> has 2 symbols
>                   Ignoring extra symbols
Errors from xkbcomp are not fatal to the X server

X.Org X Server 1.17.2
Release Date: 2015-06-16
X Protocol Version 11, Revision 0
Build Operating System: Linux 3.8.11-2-ARCH armv7l Raspbian
Current Operating System: Linux raspberrypi 4.0.7-v7+ #1 SMP PREEMPT Sat Jul 11 20:44:05 CEST 2015 armv7l
Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2709.boardrev=0xa01041 bcm2709.serial=0xb664b8da smsc95xx.macaddr=B8:27:EB:64:B8:DA bcm2708_fb.fbswap=1 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
Build Date: 11 January 2016  01:32:44PM
xorg-server 2:1.17.2-1+rpi1 (http://www.debian.org/support)
Current version of pixman: 0.33.3
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.1.log", Time: Sun Dec  4 16:32:38 2016
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found.
The XKEYBOARD keymap compiler (xkbcomp) reports:
> Warning:          Type "ONE_LEVEL" has 1 levels, but <RALT> has 2 symbols
>                   Ignoring extra symbols
Errors from xkbcomp are not fatal to the X server
^Cxinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.

xinit: unexpected signal 2
Couldn't get a file descriptor referring to the `console`

What can be the problem?

Thank you very much!

notro, 2016-12-04:
Have you changed or deleted /usr/share/X11/xorg.conf.d/99-fbturbo.conf ?
See https://github.com/notro/fbtft/wiki/FBTFT-on-Raspian#where-to-display-x-windows

michaelgino, 2016-12-04:
Hi,

Thanks for your help, now I trie to delete, change to "fb1" and with the "Install custom configuration", but the problem is the same with this measage;

```
X.Org X Server 1.17.2
Release Date: 2015-06-16
X Protocol Version 11, Revision 0
Build Operating System: Linux 3.8.11-2-ARCH armv7l Raspbian
Current Operating System: Linux raspberrypi 4.0.7-v7+ #1 SMP PREEMPT Sat Jul 11 20:44:05 CEST 2015 armv7l
Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2709.boardrev=0xa01041 bcm2709.serial=0xb664b8da smsc95xx.macaddr=B8:27:EB:64:B8:DA bcm2708_fb.fbswap=1 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
Build Date: 11 January 2016  01:32:44PM
xorg-server 2:1.17.2-1+rpi1 (http://www.debian.org/support)
Current version of pixman: 0.33.3
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.1.log", Time: Sun Dec  4 23:30:13 2016
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found.
The XKEYBOARD keymap compiler (xkbcomp) reports:
> Warning:          Type "ONE_LEVEL" has 1 levels, but <RALT> has 2 symbols
>                   Ignoring extra symbols
Errors from xkbcomp are not fatal to the X server

X.Org X Server 1.17.2
Release Date: 2015-06-16
X Protocol Version 11, Revision 0
Build Operating System: Linux 3.8.11-2-ARCH armv7l Raspbian
Current Operating System: Linux raspberrypi 4.0.7-v7+ #1 SMP PREEMPT Sat Jul 11 20:44:05 CEST 2015 armv7l
Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2709.boardrev=0xa01041 bcm2709.serial=0xb664b8da smsc95xx.macaddr=B8:27:EB:64:B8:DA bcm2708_fb.fbswap=1 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
Build Date: 11 January 2016  01:32:44PM
xorg-server 2:1.17.2-1+rpi1 (http://www.debian.org/support)
Current version of pixman: 0.33.3
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.1.log", Time: Sun Dec  4 23:30:13 2016
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found.
The XKEYBOARD keymap compiler (xkbcomp) reports:
> Warning:          Type "ONE_LEVEL" has 1 levels, but <RALT> has 2 symbols
>                   Ignoring extra symbols
Errors from xkbcomp are not fatal to the X server
```
Thank you very much!

notro, 2016-12-05:
I don't know. I think /var/log/Xorg.1.log will show if it tries to use /dev/fb1.

michaelgino, 2016-12-05:
Thanks, here is the Xorg.1.log file, is it show something?
```

pi@raspberrypi:~ $ cat /home/pi/Documents/Xorg.1.log
[   224.231]
X.Org X Server 1.17.2
Release Date: 2015-06-16
[   224.232] X Protocol Version 11, Revision 0
[   224.232] Build Operating System: Linux 3.8.11-2-ARCH armv7l Raspbian
[   224.232] Current Operating System: Linux raspberrypi 4.0.7-v7+ #1 SMP PREEMP                                                                                                   T Sat Jul 11 20:44:05 CEST 2015 armv7l
[   224.232] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bc                                                                                                   m2708_fb.fbheight=984 bcm2709.boardrev=0xa01041 bcm2709.serial=0xb664b8da smsc95                                                                                                   xx.macaddr=B8:27:EB:64:B8:DA bcm2708_fb.fbswap=1 bcm2709.disk_led_gpio=47 bcm270                                                                                                   9.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=                                                                                                   0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=serial0,1152                                                                                                   00 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repai                                                                                                   r=yes rootwait quiet splash plymouth.ignore-serial-consoles
[   224.233] Build Date: 11 January 2016  01:32:44PM
[   224.233] xorg-server 2:1.17.2-1+rpi1 (http://www.debian.org/support)
[   224.233] Current version of pixman: 0.33.3
[   224.234]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[   224.234] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   224.235] (==) Log file: "/var/log/Xorg.1.log", Time: Mon Dec  5 16:17:00 201                                                                                                   6
[   224.236] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[   224.237] (==) No Layout section.  Using the first Screen section.
[   224.238] (==) No screen section available. Using defaults.
[   224.238] (**) |-->Screen "Default Screen Section" (0)
[   224.238] (**) |   |-->Monitor "<default monitor>"
[   224.239] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[   224.239] (**) |   |-->Device "Allwinner A10/A13 FBDEV"
[   224.239] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[   224.239] (==) Automatically adding devices
[   224.240] (==) Automatically enabling devices
[   224.240] (==) Automatically adding GPU devices
[   224.240] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
[   224.240]    Entry deleted from font path.
[   224.240] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[   224.240]    Entry deleted from font path.
[   224.240] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[   224.240]    Entry deleted from font path.
[   224.241] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[   224.241]    Entry deleted from font path.
[   224.241] (==) FontPath set to:
        /usr/share/fonts/X11/100dpi/:unscaled,
        /usr/share/fonts/X11/Type1,
        /usr/share/fonts/X11/100dpi,
        built-ins
[   224.241] (==) ModulePath set to "/usr/lib/xorg/modules"
[   224.241] (II) The server relies on udev to provide the list of input devices                                                                                                   .
        If no devices become available, reconfigure udev or disable AutoAddDevic                                                                                                   es.
[   224.241] (II) Loader magic: 0x76f4cf00
[   224.241] (II) Module ABI versions:
[   224.241]    X.Org ANSI C Emulation: 0.4
[   224.241]    X.Org Video Driver: 19.0
[   224.241]    X.Org XInput driver : 21.0
[   224.241]    X.Org Server Extension : 9.0
[   224.242] (II) no primary bus or device found
[   224.242] (II) LoadModule: "glx"
[   224.243] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[   224.264] (II) Module glx: vendor="X.Org Foundation"
[   224.264]    compiled for 1.17.2, module version = 1.0.0
[   224.264]    ABI class: X.Org Server Extension, version 9.0
[   224.264] (==) AIGLX enabled
[   224.265] (II) LoadModule: "fbturbo"
[   224.265] (II) Loading /usr/lib/xorg/modules/drivers/fbturbo_drv.so
[   224.266] (II) Module fbturbo: vendor="X.Org Foundation"
[   224.266]    compiled for 1.17.2, module version = 0.5.1
[   224.266]    Module class: X.Org Video Driver
[   224.266]    ABI class: X.Org Video Driver, version 19.0
[   224.267] (II) FBTURBO: driver for framebuffer: fbturbo
[   224.267] (--) using VT number 2
[... 100 lines cut ...]
[   224.907] (--) evdev: USB-compliant keyboard: Found absolute axes
[... cut ...]
[   224.911] (**) USB-compliant keyboard: (accel) acceleration factor: 2.000
[   224.911] (**) USB-compliant keyboard: (accel) acceleration threshold: 4
[   224.913] (II) config/udev: Adding input device USB-compliant keyboard (/dev/                                                                                                   input/mouse1)
[   224.913] (II) No input driver specified, ignoring this device.
[   224.913] (II) This device may have been added with another device file.
```

Thank you very much!

notro, 2016-12-05:
It is using /dev/fb0 (hdmi):
```
[   224.402] (II) FBTURBO(0): hardware: BCM2708 FB (video memory: 3505kB)

[   224.404] (--) FBTURBO(0): Virtual size is 1824x984 (pitch 1824)
```

michaelgino, 2016-12-05:
Hi,

Yes, because I first use the method from the Freetronic itiner, like;

> "The easiest way to verify the display is working is by loading a graphical "X Windows" session on it. On recent versions of Raspbian, you'll need to first edit the configuration file /usr/share/X11/xorg.conf.d/99-fbturbo.conf on your Pi. Look for this line:
> 
> Option          "fbdev" "/dev/fb0"
> Comment this line out by placing a # at the beginning, so it looks like this:
> 
> #Option          "fbdev" "/dev/fb0"

But now I trie with this,

> Change to /dev/fb1 in /usr/share/X11/xorg.conf.d/99-fbturbo.conf

to see what is in the Xorg.1.log file.

 Many thanks!

michaelgino, 2016-12-06:
Hi, 

I test with change to "fbdev" "/dev/fb1" in the config file, now this is the Xorg.1.log;

```
pi@raspberrypi:~ $ sudo cat /var/log/Xorg.1.log
[   292.732]
X.Org X Server 1.17.2
Release Date: 2015-06-16
[   292.732] X Protocol Version 11, Revision 0
[   292.733] Build Operating System: Linux 3.8.11-2-ARCH armv7l Raspbian
[   292.733] Current Operating System: Linux raspberrypi 4.0.7-v7+ #1 SMP PREEMP                                                                                                   T Sat Jul 11 20:44:05 CEST 2015 armv7l
[   292.733] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bc                                                                                                   m2708_fb.fbheight=984 bcm2709.boardrev=0xa01041 bcm2709.serial=0xb664b8da smsc95                                                                                                   xx.macaddr=B8:27:EB:64:B8:DA bcm2708_fb.fbswap=1 bcm2709.disk_led_gpio=47 bcm270                                                                                                   9.disk_led_active_low=0 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=                                                                                                   0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=serial0,1152                                                                                                   00 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repai                                                                                                   r=yes rootwait quiet splash plymouth.ignore-serial-consoles
[   292.733] Build Date: 11 January 2016  01:32:44PM
[   292.734] xorg-server 2:1.17.2-1+rpi1 (http://www.debian.org/support)
[   292.734] Current version of pixman: 0.33.3
[   292.734]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[   292.734] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   292.735] (==) Log file: "/var/log/Xorg.1.log", Time: Tue Dec  6 07:20:16 201                                                                                                   6
[   292.735] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[   292.736] (==) No Layout section.  Using the first Screen section.
[   292.737] (==) No screen section available. Using defaults.
[   292.737] (**) |-->Screen "Default Screen Section" (0)
[   292.737] (**) |   |-->Monitor "<default monitor>"
[   292.738] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[   292.738] (**) |   |-->Device "Allwinner A10/A13 FBDEV"
[   292.738] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[   292.738] (==) Automatically adding devices
[   292.738] (==) Automatically enabling devices
[   292.738] (==) Automatically adding GPU devices
[   292.738] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
[   292.738]    Entry deleted from font path.
[   292.738] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[   292.738]    Entry deleted from font path.
[   292.738] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[   292.738]    Entry deleted from font path.
[   292.738] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[   292.739]    Entry deleted from font path.
[   292.739] (==) FontPath set to:
        /usr/share/fonts/X11/100dpi/:unscaled,
        /usr/share/fonts/X11/Type1,
        /usr/share/fonts/X11/100dpi,
        built-ins
[   292.739] (==) ModulePath set to "/usr/lib/xorg/modules"
[   292.739] (II) The server relies on udev to provide the list of input devices                                                                                                   .
        If no devices become available, reconfigure udev or disable AutoAddDevic                                                                                                   es.
[   292.739] (II) Loader magic: 0x76fb9f00
[   292.739] (II) Module ABI versions:
[   292.739]    X.Org ANSI C Emulation: 0.4
[   292.739]    X.Org Video Driver: 19.0
[   292.739]    X.Org XInput driver : 21.0
[   292.739]    X.Org Server Extension : 9.0
[   292.739] (II) no primary bus or device found
[   292.740] (II) LoadModule: "glx"
[   292.741] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[   292.754] (II) Module glx: vendor="X.Org Foundation"
[   292.754]    compiled for 1.17.2, module version = 1.0.0
[   292.754]    ABI class: X.Org Server Extension, version 9.0
[   292.754] (==) AIGLX enabled
[   292.754] (II) LoadModule: "fbturbo"
[   292.755] (II) Loading /usr/lib/xorg/modules/drivers/fbturbo_drv.so
[   292.755] (II) Module fbturbo: vendor="X.Org Foundation"
[   292.756]    compiled for 1.17.2, module version = 0.5.1
[   292.756]    Module class: X.Org Video Driver
[   292.756]    ABI class: X.Org Video Driver, version 19.0
[... 323 lines cut ...]
[   716.464] (II) XINPUT: Adding extended input device "USB-compliant keyboard" (type: KEYBOARD, id 8)
[   716.464] (**) Option "xkb_rules" "evdev"
[   716.465] (**) Option "xkb_model" "pc105"
[   716.465] (**) Option "xkb_layout" "hu"
[... cut ...]
[   716.472] (II) No input driver specified, ignoring this device.
[   716.472] (II) This device may have been added with another device file.
[   787.325] (II) evdev: USB-compliant keyboard: Close
[   787.326] (II) UnloadModule: "evdev"
[   787.326] (II) evdev: USB-compliant keyboard: Close
[   787.326] (II) UnloadModule: "evdev"
[   787.326] (II) evdev: USB Optical Mouse: Close
[   787.327] (II) UnloadModule: "evdev"
[   787.365] (II) Server terminated successfully (0). Closing log file.
```

I don't know what can be the problem :/

Thank you!!!

notro, 2016-12-07:
I don't know much about xorg, but at least now it's using the right framebuffer:
```
[   292.909] (II) FBTURBO(0): using /dev/fb1

[   292.910] (II) FBTURBO(0): hardware: fb_ssd1351 (video memory: 32kB)
[   292.910] (**) FBTURBO(0): Option "fbdev" "/dev/fb1"
```

Have you verified by other means that the display is working?
I use the framebuffer console for testing: ```con2fbmap 1 1```

michaelgino, 2016-12-07:
Yes, the display is working with Arduino. But the unfortunately do nothing. Maybe the `spi-bcm2708` can be a problem, but I running out of ideas :/

notro, 2016-12-08:
Have you contacted Freetronics?

michaelgino, 2016-12-08:
Yes, I waiting for the answer, thanks!

michaelgino, 2016-12-09:
Hi,

Nothing works, but then I downgrade to Raspbian "Wheezy", and everything was fine!

Is it possible with this display, to make an extended desktop with to displays, like in this video?
https://www.youtube.com/watch?v=wxJpWRPKe-g

How can I do that?

Thanks!

closed 2018-05-10 by notro