notro/fbtft #431: Device tree wiki

alidaf, 2016-12-14:
I have followed the instructions for compiling a device tree file but it doesn't seem to work, i.e. no there is no fb1 device or any spi devices on reboot.

Using  "sudo vcdbg log msg" seems to indictate that the driver has loaded but lsmod doesn't show anything.

Do you have any other methods to test?

Due to the kernel > 4.4, the compiled device tree is foo.dtbo instead of foo-overlay.dtb in /boot/overlays. Apart from that I don't think there are any other differences.

notro, 2016-12-14:
Device Tree adds devices, kernel modules provide drivers.
I have enabled spidev and it shows up like this in DT:
```
$ ls -l /proc/device-tree/soc/spi@7e204000/
drwxr-xr-x 2 root root  0 Dec 14 22:22 spidev@0
drwxr-xr-x 2 root root  0 Dec 14 22:22 spidev@1
```
And like this as the "real" spi device:
```
$ ls -l /sys/bus/spi/devices/
lrwxrwxrwx 1 root root 0 Dec 14 20:59 spi0.0 -> ../../../devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.0
lrwxrwxrwx 1 root root 0 Dec 14 22:23 spi0.1 -> ../../../devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1
```
The spidev driver is built into the kernel, so it doesn't show up, but the spi master/controller driver module shows up:
```
$ lsmod
Module                  Size  Used by
spi_bcm2835             6932  0
```
You need both the device and the driver. fbtft kernel modules which provides drivers, should load automatically when the device is present (except flexfb).

I've fixed the dtbo extension in the wiki.

alidaf, 2016-12-15:
I don't understand that in the context of my problem. After the device tree is loaded and allocated memory, should the modules be loaded automatically and how is this achieved? There is nothing in the device tree code that identifies which driver to use. I thought that device trees superseded the need to edit files in /etc/modules-load.d and /etc/modprobe.d, otherwise what's the point?

I have both spi devices in /proc but only one device in /sys. The drivers don't show up with lsmod until I modprobe them.

Thanks

notro, 2016-12-15:
When a driver is loaded or a device is added,  a "scan" is done to see if there's an unbound device that matches a driver.
In the case of DT, the drivers of_device_id table is compared to the device compatible string.
If they match, probe is called with the device as an argument.

For this to work, the driver has to be loaded/registered.
The fbtft drivers use the MODULE_ALIAS macro to add an alias to help with that.
(MODULE_DEVICE_TABLE should have handled this, but doesn't for some reason).

The alias show up here:
```
$ modinfo fb_ili9341
filename:       /lib/modules/4.8.8+/kernel/drivers/staging/fbtft/fb_ili9341.ko
license:        GPL
author:         Christian Vogelgsang
description:    FB driver for the ILI9341 LCD display controller
alias:          platform:ili9341
alias:          spi:ili9341
alias:          platform:fb_ili9341
alias:          spi:fb_ili9341
srcversion:     039E612314516D8772E72E6
alias:          of:N*T*Cilitek,ili9341C*
alias:          of:N*T*Cilitek,ili9341
depends:        fbtft
staging:        Y
intree:         Y
vermagic:       4.8.8+ mod_unload modversions ARMv6 p2v8
```
I don't remember the details, but when a device is added, udev now has the info it needs to autoload the **module**, which contains the **driver** that the **device** needs:
```
$ ls -l /sys/bus/spi/devices/spi0.0/uevent
-rw-r--r-- 1 root root 4096 Dec 14 20:59 /sys/bus/spi/devices/spi0.0/uevent
tinydrm:~$
tinydrm:~$
tinydrm:~$ cat /sys/bus/spi/devices/spi0.0/uevent
DRIVER=spidev
OF_NAME=spidev
OF_FULLNAME=/soc/spi@7e204000/spidev@0
OF_COMPATIBLE_0=spidev
OF_COMPATIBLE_N=1
MODALIAS=spi:spidev
```

> I don't understand that in the context of my problem.

If this doesn't help either, then you need to provide the DT overlay. It's very difficult to help, when you don't provide any details.

alidaf, 2016-12-15:
Thank you. I didn't want to be presumptuous by supplying my overlay and I tried to give the information that I thought was immediately relevant. I'm sure the problem is trivial to solve but I've never tried to do anything like this before and therefore nothing is immediately obvious.

I have added a couple of aliases to the driver file and now get modules loaded but there is no fb1 device. The modules don't seem quite right as the fb_ssd1322 module isn't used!

 Module               Size  Used by
fb_ssd1322        4701  0 
fbtft                   35310  1 fb_ssd1322
syscopyarea      3161  1 fbtft
sysfillrect           3716  1 fbtft
sysimgblt           2422  1 fbtft
fb_sys_fops       1677  1 fbtft

The driver and dts file are both here... https://github.com/alidaf/raspberryPi/tree/master/displayPi/ssd1322-spi/fbtft

notro, 2016-12-15:
```
FBTFT_REGISTER_DRIVER(DRVNAME, "SSD1322", &display);
```
SSD1322 is added to of_device_id and used to match against the compatible string: ssd1322

This should work:
```
FBTFT_REGISTER_DRIVER(DRVNAME, "ssd1322", &display);
```

alidaf, 2016-12-15:
I see. It probably would have been better if I'd started off with one of your Solomon drivers and modified it rather than taking the presslab-us version and adding what I thought was missing. I did originally have "Solomon, ssd1322" but didn't appreciate that field was used to match. It makes more sense now.
I've changed it back to "Solomon, ssd1322" and even though lsmod looks the same, fb1 now exists and fbtest works as expected.
Many thanks.

closed 2016-12-15 by alidaf