notro/fbtft #433: Waveshare 3.5" no touch interrupts

Levantinyx, 2017-01-03:
Hi Notro,

first off a big thanks for all the work you put into this. After a long search through outdated posts I followed this post: https://www.raspberrypi.org/forums/viewtopic.php?p=977889#p977889 to get my Waveshare 3.5" (A) running on a Pi 2 B. The screen is working fine, however I do not get any touch capabilities and from research I seem not to generate the interrupts.
A few logs/entries:
cat /proc/bus/input/devices
`I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="ADS7846 Touchscreen"
P: Phys=spi0.1/input0
S: Sysfs=/devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input2
U: Uniq=
H: Handlers=mouse1 event2 
B: PROP=0
B: EV=b
B: KEY=400 0 0 0 0 0 0 0 0 0 0
B: ABS=1000003`
dmesg | grep -i ADS7846
`[    8.505545] spi_master spi0: spi_device register error /soc/spi@7e204000/ads7846@1
[    8.505569] spi_master spi0: Failed to create SPI device for /soc/spi@7e204000/ads7846@1
[    8.950482] ads7846 spi0.1: touchscreen, irq 183
[    8.951436] input: ADS7846 Touchscreen as /devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input2`
cat /proc/interrupts grep ads7846
`183:          0          0          0          0  pinctrl-bcm2835  17 Edge      ads7846`
evtest
`No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:   PixArt HP USB Optical Mouse
/dev/input/event1:   HID 046a:0011
/dev/input/event2:   ADS7846 Touchscreen
Select the device event number [0-2]: 2
Input driver version is 1.0.1
Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
Input device name: "ADS7846 Touchscreen"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 330 (BTN_TOUCH)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
      Value      0
      Min        0
      Max     4095
    Event code 1 (ABS_Y)
      Value      0
      Min        0
      Max     4095
    Event code 24 (ABS_PRESSURE)
      Value      0
      Min        0
      Max      255
Properties:
Testing ... (interrupt to exit)`
Nothing happens when I click the touchscreen in evtest.
It looks very similiar to what CoderFX is having trouble with here: https://github.com/notro/fbtft/issues/215 but sadly there was no resolution posted.
I would be glad if you could help me find out what is going wrong as I am still extremely new to Linux and Raspberry Pi in particular

notro, 2017-01-08:
The forum post uses gpio 25:
```
dtoverlay=ads7846,penirq=25,penirq_pull=2,xohms=150,swapxy=1,xmin=300,ymin=700,xmax=3800,ymax=3400,pmax=255
```
But you seem to use 17:
```
183: 0 0 0 0 pinctrl-bcm2835 17 Edge ads7846
```

And this doesn't look right:
```
[ 8.505545] spi_master spi0: spi_device register error /soc/spi@7e204000/ads7846@1 
[ 8.505569] spi_master spi0: Failed to create SPI device for /soc/spi@7e204000/ads7846@1 
[ 8.950482] ads7846 spi0.1: touchscreen, irq 183 
[ 8.951436] input: ADS7846 Touchscreen as /devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input2
```

How do you add the ads7846 device?

Levantinyx, 2017-01-09:
Hi Notro,
thanks for your response. I was indeed using penirq=17 as I tried through a few other forum posts some of which suggested using number 17 was the way to go. 
Resetting to the configurations used in the linked post (and rebooting) leaves devices and interrupts as is but dmesg now only shows:
```
[    9.310549] ads7846 spi0.1: touchscreen, irq 183
[    9.311769] input: ADS7846 Touchscreen as 
```
/devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input2
my /boot/config.txt used to end with: 
`dtparam=audio=on
dtoverlay=ads7843,speed=50000,penirq=17,swapxy=1
dtoverlay=waveshare35a,rotate=180`
after resetting it is now as described in the forum:
`dtparam=audio=on
dtoverlay=ads7843,penirq=25,penirq_pull=2,xohms=150,,swapxy=1,xmin=300,ymin=700,xmax=3800,ymax=3400,pmax=255
dtoverlay=waveshare35a`
in both cases dtparam=spi=on was specified farther up, added using rasbi-config.
I hope the config.txt was what you wanted to see, as I said, I'm still new to linux and rasberry pi, so honestly I have no idea whether I should use pinrq=25 or 17 (and whether to write the Section "InputDevice" into /usr/share/X11/xorg.conf.d/99-calibration.txt or /etc/X11/xorg.conf.d/99-calibration.txt - currently in both)
Looking forward to an answer and agian thanks for your help,

Levantin

notro, 2017-01-09:
The overlay already has an entry for the touch controller:
https://github.com/swkim01/waveshare-dtoverlays/blob/master/waveshare35a-overlay.dts

You don't need the ads7846 overlay, btw there's no ads7843 overlay as you have in your last post.

As to why it doesn't work, I really don't know. These waveshare displays generate a lot of calls for help. I don't know if that's because they sell an insanly amount of displays, or if they just suck at supporting their product.

Levantinyx, 2017-01-09:
Mmh seems I overlooked another attempt (someone mentioned it would be 7843 not 7846)... anyways I commented that line out as you said it isn't needed and nothing changed. Thanks for trying to help me, gues I'll eihter let it be or try again with another screen (not waveshare then XD)

amrithmmh, 2018-03-26:
inside /etc/initramfs-tools/modules
spi_bcm2708
fbtft
fb_s6d02a1

and
/etc/initramfs-tools/scripts/init-top/spi script executable
#!/bin/sh
modprobe spi_bcm2708

not working for me
tried changing spi_bcm2708 to 283

what to use for waveshare 3.5 LCD (B)...just need the plymouth to work in lcd in fb1