notro/fbtft #437: tft can not work

muzixiaosanshi, 2017-01-08:
Hi notro:
     I use kernel 4.4 to compile driver module,I see the module already made.:  fb_st7789.ko and other module.so I copy it to pi and depmod.  when i use command:  modprobe admatec_c-berry28 and modprobe fb_st7789 , It all display :FATAL module fb_st7789 not found, but when I type: modprobe fbtft_device  name=  admatec_c-berry28, dmesg: display:
module is from the staging directory,the quality is unknown,you have been warned.
 fbtft_device:      spidev spi0.0 500kHz 8 bits mode=0x00   
 fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00     
 fbtft_device:      bcm2708_fb id=-1 pdata? no   
 fbtft_device: Deleting spi0.0  
 fbtft_device:  GPIOS used by 'admatec_c-berry28':    
fbtft_device:    'reset' = GPIO25
 fbtft_device:    'dc' = GPIO22
fbtft_device:    'led' = GPIO18   
 fbtft_device:  SPI devices registered:   
 fbtft_device:      spidev spi0.1 500kHz 8 bits mode=0x00   
 fbtft_device:      fb_st7789v spi0.0 32000kHz 8 bits mode=0x00
fb_st7789v  : disagrees about version of symbol fbtft_probe_common
fb_st7789v  : unknown symbol fbtft_probe_common(err  -22)

DO you know how to solve it ,thanks!

notro, 2017-01-08:
This is the culprit
```
fb_st7789v : disagrees about version of symbol fbtft_probe_common
```
You need to build the module with the Module.symvers that matches your kernel.
You can look at [rpi-source](https://github.com/notro/rpi-source) and see how it fishes that out.

You can also try ```modprobe -f fb_st7789v```

muzixiaosanshi, 2017-01-08:
modprobe -f fb_st7789v  I have already try it ,but it does not work ,the error still exist.

muzixiaosanshi, 2017-01-09:
@notro    rpi-source is done at rpi or the PC(cross-compile env) ? I do it in the PC,it display

--2017-01-09 10:13:03--  https://raw.githubusercontent.com/notro/rpi-source/master/rpi-source
Connecting to 10.10.5.18:8080... connected.
Proxy request sent, awaiting response... 200 OK
Length: 12670 (12K) [text/plain]
Saving to: ‘/usr/bin/rpi-source’

/usr/bin/rpi-source                                100%[================================================================================================================>]  12.37K  18.4KB/s    in 0.7s    

2017-01-09 10:13:06 (18.4 KB/s) - ‘/usr/bin/rpi-source’ saved [12670/12670]

ERROR:
Unrecognised CPU x86_64

Help: https://github.com/notro/rpi-source/wiki

Do you know what is the reason?

muzixiaosanshi, 2017-01-09:
Above is use this command:
sudo wget https://raw.githubusercontent.com/notro/rpi-source/master/rpi-source -O /usr/bin/rpi-source && sudo chmod +x /usr/bin/rpi-source && /usr/bin/rpi-source -q --tag-update

notro, 2017-01-09:
rpi-source should be used on the Raspberry Pi. It exists because there's no kernel header package for the Pi.

If you're cross compiling, why not use the kernel you're building?
https://www.raspberrypi.org/documentation/linux/kernel/building.md

muzixiaosanshi, 2017-01-12:
@notro Thanks for your answer,I can see file fb1 in my   /dev/,But I want to know How can I let the rpi3 display on my tft(st7789v)?

notro, 2017-01-12:
Console: con2fbmap 1 1
X windows, see: https://github.com/notro/fbtft/wiki/FBTFT-on-Raspian

muzixiaosanshi, 2017-01-12:
@notro Thanks for your help, Ok，But when I type "Con2fbmap 1 1",The screen display blank.So I use dmesg command: It display  :
spidev spi0.0 500khz 8bits mode=0x00
But my screen is 9bit ,I want to know  if I modified in the fb_st7789v.c  :
static struct fbtft_display display = {
	# **.regwidth = 8, change 9**
	.width = 240,
	.height = 320,
	.gamma_num = 2,
	.gamma_len = 14,
	.gamma = DEFAULT_GAMMA,
	.fbtftops = {
		.init_display = init_display,
		.set_var = set_var,
		.set_gamma = set_gamma,
		.blank = blank,
	},
};
It will display well? My solution is right?

muzixiaosanshi, 2017-01-15:
@notro Hi notro,recently,I see ST7789V datasheet,it  have some control mode.For example :8bit  parallel databus,  FBTFT support this mode?

notro, 2017-01-15:
fbtft supports parallel bus, but the next version [tinydrm](https://github.com/notro/tinydrm) will probably not.
It is slow because each bus pin/gpio is toogled individually.

Performance numbers for a 8-bit parallel connected display: https://github.com/notro/fbtft/wiki/Performance#itdb28
The fps parameter is really a delay in jiffies: HZ/fps

muzixiaosanshi, 2017-01-16:
@notro Thanks for your detailed answer,It help me so much.But how can I realize the 8bit-parallel display?Now, st7789v 4-line serial mode is realize on my pi.

notro, 2017-01-18:
I assume your display exposes the parallel bus, so you have to set the IM[3:0] pins to 8080 parallel 8-bit mode. 
If you want to edit fbtft_device.c, here's one example: http://lxr.free-electrons.com/source/drivers/staging/fbtft/fbtft_device.c#L667
Or you can use the custom parameter to fbtft_device: https://github.com/notro/fbtft/wiki/fbtft_device#custom-display

closed 2018-05-10 by notro