notro/fbtft #439: Problem using itdb24 driver.

recner, 2017-02-03:
Hi Notro
I have trouble setting up fbtft.
My display is using S6D1121 driver.
I am trying to load it using "sudo modprobe fbtft_device name=itdb24 gpios=db0:7,db1:5,db2:3,db3:1,db4:6,db5:4,db6:2,db7:0,cs:14,dc:12,wr:10,reset:11,rd:8,led:9"
Backlight is not lighting up. Screen also doesn't turn on.

I think my problem is here `[   55.093561] fb_s6d1121 fb_s6d1121.0: fbtft_request_gpios: gpio_request_one('db7'=0) failed with -22` but i'm unable to find any solution.

I can manually set GPIO pins by:
```
echo 9 > /sys/class/gpio/export
echo "out" > /sys/class/gpio/gpio9/direction
echo 1 > /sys/class/gpio/gpio9/value
```

Thank you in advance.

device: Cubieboard A10 with armbian installed

uname -a:
`Linux cubieboard 3.4.113-sun4i #1 PREEMPT Thu Feb 2 21:26:29 CET 2017 armv7l armv7l armv7l GNU/Linux`

dmesg:
```
[   55.053329] fbtft_device:  SPI devices registered:
[   55.067966] fbtft_device:  'fb' Platform devices registered:
[   55.073161] fbtft_device:  GPIOS used by 'itdb24':
[   55.080736] fbtft_device:    'db0' = GPIO7
[   55.084244] fbtft_device:    'db1' = GPIO5
[   55.093561] fb_s6d1121 fb_s6d1121.0: fbtft_request_gpios: gpio_request_one('db7'=0) failed with -22
[   55.108444] fbtft_device:    'db2' = GPIO3
[   55.117506] fbtft_device:    'db3' = GPIO1
[   55.120562] fbtft_device:    'db4' = GPIO6
[   55.124200] fbtft_device:    'db5' = GPIO4
[   55.127254] fbtft_device:    'db6' = GPIO2
[   55.130303] fbtft_device:    'db7' = GPIO0
[   55.134217] ------------[ cut here ]------------
[   55.140931] WARNING: at drivers/gpio/gpiolib.c:104 gpio_ensure_requested+0x58/0xb4()
[   55.142769] autorequest GPIO-14
[   55.144603] Modules linked in: fb_s6d1121(+) fbtft_device(+) fbtft syscopyarea sysfillrect sysimgblt fb_sys_fops bnep disp_ump mali_drm drm cpufreq_userspace mali ump sunxi_ir hidp rfcomm hci_uart bluetooth
[   55.168446] [<c00143b8>] (unwind_backtrace+0x0/0x12c) from [<c0032708>] (warn_slowpath_common+0x54/0x64)
[   55.176707] [<c0032708>] (warn_slowpath_common+0x54/0x64) from [<c0032748>] (warn_slowpath_fmt+0x30/0x40)
[   55.185055] [<c0032748>] (warn_slowpath_fmt+0x30/0x40) from [<c040bbf8>] (gpio_ensure_requested+0x58/0xb4)
[   55.193751] [<c040bbf8>] (gpio_ensure_requested+0x58/0xb4) from [<c040bd18>] (gpio_direction_input+0xc4/0x1c4)
[   55.202733] [<c040bd18>] (gpio_direction_input+0xc4/0x1c4) from [<bf13c260>] (fbtft_free_gpios+0x70/0xe4 [fbtft])
[   55.213013] [<bf13c260>] (fbtft_free_gpios+0x70/0xe4 [fbtft]) from [<bf13cf50>] (fbtft_register_framebuffer+0x174/0x2d0 [fbtft])
[   55.223630] [<bf13cf50>] (fbtft_register_framebuffer+0x174/0x2d0 [fbtft]) from [<bf13e29c>] (fbtft_probe_common+0x12c/0x2cc [fbtft])
[   55.232861] [<bf13e29c>] (fbtft_probe_common+0x12c/0x2cc [fbtft]) from [<c047776c>] (driver_probe_device+0xb8/0x378)
[   55.240951] [<c047776c>] (driver_probe_device+0xb8/0x378) from [<c0477afc>] (__driver_attach+0x8c/0x90)
[   55.248692] [<c0477afc>] (__driver_attach+0x8c/0x90) from [<c04759e4>] (bus_for_each_dev+0x68/0x9c)
[   55.256433] [<c04759e4>] (bus_for_each_dev+0x68/0x9c) from [<c0476cf0>] (bus_add_driver+0xd0/0x28c)
[   55.264173] [<c0476cf0>] (bus_add_driver+0xd0/0x28c) from [<c047811c>] (driver_register+0x78/0x134)
[   55.272085] [<c047811c>] (driver_register+0x78/0x134) from [<c0008640>] (do_one_initcall+0x104/0x15c)
[   55.280176] [<c0008640>] (do_one_initcall+0x104/0x15c) from [<c00739ec>] (sys_init_module+0x8d8/0x1d64)
[   55.288093] [<c00739ec>] (sys_init_module+0x8d8/0x1d64) from [<c000eca0>] (ret_fast_syscall+0x0/0x30)
[   55.291667] ---[ end trace e54264fa7926eac7 ]---
[   55.306226] fbtft_device:    'cs' = GPIO14
[   55.309290] fbtft_device:    'dc' = GPIO12
[   55.312340] fbtft_device:    'wr' = GPIO10
[   55.318738] fbtft_device:    'reset' = GPIO11
[   55.321704] fbtft_device:    'rd' = GPIO8
[   55.325013] fbtft_device:    'led' = GPIO9
[   55.329624] fbtft_device:  'fb' Platform devices registered:
[   55.334028] fbtft_device:      fb_s6d1121 id=0 pdata? yes
[   55.344771] ------------[ cut here ]------------
[   55.351490] WARNING: at drivers/gpio/gpiolib.c:104 gpio_ensure_requested+0x58/0xb4()
[   55.353328] autorequest GPIO-12
[   55.355163] Modules linked in: fb_s6d1121(+) fbtft_device fbtft syscopyarea sysfillrect sysimgblt fb_sys_fops bnep disp_ump mali_drm drm cpufreq_userspace mali ump sunxi_ir hidp rfcomm hci_uart bluetooth
[   55.378735] [<c00143b8>] (unwind_backtrace+0x0/0x12c) from [<c0032708>] (warn_slowpath_common+0x54/0x64)
[   55.386997] [<c0032708>] (warn_slowpath_common+0x54/0x64) from [<c0032748>] (warn_slowpath_fmt+0x30/0x40)
[   55.395345] [<c0032748>] (warn_slowpath_fmt+0x30/0x40) from [<c040bbf8>] (gpio_ensure_requested+0x58/0xb4)
[   55.404041] [<c040bbf8>] (gpio_ensure_requested+0x58/0xb4) from [<c040bd18>] (gpio_direction_input+0xc4/0x1c4)
[   55.413022] [<c040bd18>] (gpio_direction_input+0xc4/0x1c4) from [<bf13c260>] (fbtft_free_gpios+0x70/0xe4 [fbtft])
[   55.423294] [<bf13c260>] (fbtft_free_gpios+0x70/0xe4 [fbtft]) from [<bf13cf50>] (fbtft_register_framebuffer+0x174/0x2d0 [fbtft])
[   55.433912] [<bf13cf50>] (fbtft_register_framebuffer+0x174/0x2d0 [fbtft]) from [<bf13e29c>] (fbtft_probe_common+0x12c/0x2cc [fbtft])
[   55.443143] [<bf13e29c>] (fbtft_probe_common+0x12c/0x2cc [fbtft]) from [<c047776c>] (driver_probe_device+0xb8/0x378)
[   55.451233] [<c047776c>] (driver_probe_device+0xb8/0x378) from [<c0477afc>] (__driver_attach+0x8c/0x90)
[   55.458975] [<c0477afc>] (__driver_attach+0x8c/0x90) from [<c04759e4>] (bus_for_each_dev+0x68/0x9c)
[   55.466716] [<c04759e4>] (bus_for_each_dev+0x68/0x9c) from [<c0476cf0>] (bus_add_driver+0xd0/0x28c)
[   55.474455] [<c0476cf0>] (bus_add_driver+0xd0/0x28c) from [<c047811c>] (driver_register+0x78/0x134)
[   55.482367] [<c047811c>] (driver_register+0x78/0x134) from [<c0008640>] (do_one_initcall+0x104/0x15c)
[   55.490458] [<c0008640>] (do_one_initcall+0x104/0x15c) from [<c00739ec>] (sys_init_module+0x8d8/0x1d64)
[   55.498376] [<c00739ec>] (sys_init_module+0x8d8/0x1d64) from [<c000eca0>] (ret_fast_syscall+0x0/0x30)
[   55.501949] ---[ end trace e54264fa7926eac8 ]---
[   55.507349] ------------[ cut here ]------------
[   55.514060] WARNING: at drivers/gpio/gpiolib.c:104 gpio_ensure_requested+0x58/0xb4()
[   55.515897] autorequest GPIO-10
```

recner, 2017-02-04:
Oh, I haven't noticed that gpios start at 1.
Now I'm using this command  `sudo modprobe fbtft_device name=itdb24 gpios=db0:8,db1:6,db2:4,db3:2,db4:7,db5:5,db6:3,db7:1,cs:15,dc:13,wr:11,reset:12`
LED backlight is wired directly to GND, so backlight is always on. RD is wired to 3.3v.
All pins are wired correctly, tested with python script from [troubleshooting](https://github.com/notro/fbtft/wiki/Troubleshooting) 
Screen however stays white. Is it correct driver for this screen?

screen: [http://www.ebay.com/itm/282274565558](http://www.ebay.com/itm/282274565558)
dmesg:
```
[   34.721559] dma0: IRQ with no loaded buffer?
[   45.935222] fbtft_device:  SPI devices registered:
[   45.939842] fbtft_device:  'fb' Platform devices registered:
[   45.943808] fbtft_device:  GPIOS used by 'itdb24':
[   45.962076] fbtft_device:    'db0' = GPIO8
[   45.983445] fbtft_device:    'db1' = GPIO6
[   45.986501] fbtft_device:    'db2' = GPIO4
[   45.989550] fbtft_device:    'db3' = GPIO2
[   46.003354] fbtft_device:    'db4' = GPIO7
[   46.006612] fbtft_device:    'db5' = GPIO5
[   46.009673] fbtft_device:    'db6' = GPIO3
[   46.012723] fbtft_device:    'db7' = GPIO1
[   46.023345] fbtft_device:    'cs' = GPIO15
[   46.026403] fbtft_device:    'dc' = GPIO13
[   46.029452] fbtft_device:    'wr' = GPIO11
[   46.032762] fbtft_device:    'reset' = GPIO12
[   46.054890] fbtft_device:  'fb' Platform devices registered:
[   46.059302] fbtft_device:      fb_s6d1121 id=0 pdata? yes
[   46.343368] graphics fb2: fb_s6d1121 frame buffer, 240x320, 150 KiB video memory, 4 KiB buffer memory, fps=20
```

notro, 2017-02-04:
> Is it correct driver for this screen?

Looks that way, but the default init sequence in the driver isn't the same as one Arduino init I found, some register settings was missing.

I couldn't find any reference on how to setup the controller on the ebay page. The init sequence is panel specific.

fbtft init: https://github.com/notro/fbtft/blob/master/fb_s6d1121.c#L41
Arduino init: https://github.com/iot-playground/Arduino/blob/master/external_libraries/UTFT/tft_drivers/s6d1121/initlcd.h

init= parameter: https://github.com/notro/fbtft/wiki/fbtft_device#parameters

recner, 2017-02-04:
I tried to pass init sequence from arduino library via init parameter and nothing happened (white screen).
I connected screen to an arduino and uploaded UTFT library example. It also doesn't work (white screen).
I assume that I have broken lcd.  
I'm closing this, because it's definitely not an issue of fbtft.
**Thank you for helping :)**

closed 2017-02-04 by recner