notro/fbtft #440: 3.5 tft display laggy in pi1B with retropie

hamlet01, 2017-02-21:
Hello.
First sorry for my bad english. I will try to explain my self as good I can
I have installed a 3.5 Zoll Display from alliexpress on my Raspberry pi 1B with RetroPie. Becouse the connector was not aligned I have soldering 4cm wires from each tft pin to the pi.
I have installed de LCD-show driver and configured it with sudo ./LCD35-show
I have forced 640x480 resolution into HDMI (supossed scaled to 320x240 into tft display) editing /boot/config.txt with
sudo nano config.txt and adding:
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=4
I have installed fbcp and added to the boot
The result is the tft mirrored from the HDMI (THATS OK) but... the display is SO slow (laggy)
I have tryed to change speed settings in /etc/modules but I think that I dont do somenthing well
Can you help me please??

Mybe this can help...

pi@retropie:~ $ dmesg | grep spi
[   10.998928] spi-bcm2835 20204000.spi: chipselect 1 already in use
[   10.998968] spi_master spi0: spi_device register error /soc/spi@7e204000/waveshare35a-ts@1
[   10.998998] spi_master spi0: Failed to create SPI device for /soc/spi@7e204000/waveshare35a-ts@1
[   14.478463] ads7846 spi0.1: f(sample) 384 KHz?
[   14.478544] ads7846: probe of spi0.1 failed with error -22
[   15.650350] graphics fb1: fb_ili9486 frame buffer, 480x320, 300 KiB video memory, 32 KiB DMA buffer memory, fps=33, spi0.0 at 16 MHz

pi@retropie:/boot $ lsmod
Module                  Size  Used by
snd_hrtimer             2009  1
snd_seq                58506  1
snd_seq_device          5266  1 snd_seq
cfg80211              499234  0
rfkill                 21397  1 cfg80211
fb_ili9486              3213  2
fbtft                  32615  1 fb_ili9486
syscopyarea             3161  1 fbtft
sysfillrect             3716  1 fbtft
ads7846                12525  0
sysimgblt               2422  1 fbtft
fb_sys_fops             1677  1 fbtft
hwmon                   4552  1 ads7846
joydev                  9194  0
evdev                  11650  0
snd_bcm2835            23131  0
snd_pcm                95473  1 snd_bcm2835
snd_timer              22556  3 snd_hrtimer,snd_pcm,snd_seq
snd                    68368  6 snd_bcm2835,snd_timer,snd_pcm,snd_seq,snd_seq_device
spi_bcm2835             7424  0
i2c_bcm2708             5740  0
bcm2835_gpiomem         3823  0
bcm2835_wdt             4133  0
uio_pdrv_genirq         3718  0
uio                    10230  1 uio_pdrv_genirq
uinput                  7735  0
fuse                   90216  1
ipv6                  367607  28

pi@retropie:~ $ dmesg | grep graphics
[   15.650350] graphics fb1: fb_ili9486 frame buffer, 480x320, 300 KiB video memory, 32 KiB DMA buffer memory, fps=33, spi0.0 at 16 MHz

/etc/modules:

# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.
uinput
fbtft_device speed=16000000

/boot/config.txt:

dtparam=i2c_arm=on
dtparam=i2c1=on
dtparam=spi=on
enable_uart=1
hdmi_force_hotplug=1
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=4
overscan_left=30

THANKS

notro, 2017-02-22:
A 480x320 display at 16MHz will be slow.
The tontec display which runs at 125MHz is probably the only 480x320 SPI display with good fbcp performance: https://github.com/notro/fbtft/wiki/Performance#mz61581-pi-ext
fbcp triggers a full diplay update every time, as opposed to X windows which only triggers update of the changed region.

LiqwdE, 2017-03-02:
Can you paste you /boot/config.txt

This was where my problem was on my 3.5 waveshare.

closed 2018-05-10 by notro