notro/fbtft #441: Assistance with Waveshare 3.2 inch v4 screen

ravenousld3341, 2017-02-28:
1. Amazing work here. Thank you!
2. I don't know where else to go with this. Hopefully you can assist. All information I have follows.

> Setup a Waveshare 3.2 with Raspberry Pi 2 B
> 
> I have a raspberry Pi 2 B,
> a monitor Wasveshare Spotpear 3.2 inch LCD v4
> OS is Raspberry, Wheezy with NOOBS_v1_4.1
> 
> USE PASTE AND COPY
> 
> 1) sudo nano /usr/share/X11/xorg.conf.d/99-fbturbo.conf
> 	Option “fbdev” “/dev/fb0″ and change the fb0 to fb1
> 
> 2) Download the driver for the LCD
> 	sudo REPO_URI=https://github.com/notro/rpi-firmware rpi-update
> 	sudo reboot
> 
> 3) Enable SPI via raspi-config
> 	sudo raspi-config
> 	find the “Advanced Options” line
> 	find the line that says “SPI    ENABLE/DISABLE AUTOMATIC LOADING“
> 	Click enter here
> 	Enter yes and reboot sudo reboot
> 
> 4) Edit the /etc/modules 
> 	sudo nano /etc/modules
> 
> 	Add the following code below the snd-bcm2835 line
> 
> spi-bcm2708
> ads7846
> 
> fbtft_device rotate=270 name=waveshare32b gpios=dc:22,reset:27 speed=48000000
> 	
> fb_ili9340 width=320 height=240
> buswidth=8init=-1,0xCB,0x39,0x2C,0x00,0x34,0x02,-1,0xCF,0x00,0XC1,0X30,
> -1,0xE8,0x85,0x00,0x78,-1,0xEA,0x00,0x00,-1,0xED,0x64,0x03,0X12,0X81,
> -1,0xF7,0x20,-1,0xC0,0x23,-1,0xC1,0x10,-1,0xC5,0x3e,0x28,-1,0xC7,0x86,
> -1,0x36,0x28,-1,0x3A,0x55,-1,0xB1,0x00,0x18,-1,0xB6,0x08,0x82,0x27,
> -1,0xF2,0x00,-1,0x26,0x01,-1,0xE0,0x0F,0x31,0x2B,0x0C,0x0E,0x08,0x4E,0xF1,
> 0x37,0x07,0x10,0x03,0x0E,0x09,0x00,-1,0XE1,0x00,0x0E,0x14,0x03,0x11,0x07,
> 0x31,0xC1,0x48,0x08,0x0F,0x0C,0x31,0x36,0x0F,-1,0x11,-2,120,-1,0x29,-1,0x2c,-3
> 	"FROM fb_ili9340 UNTIL -3 ALL IN ONE LINE !!!"
> 	Remark: rotate in the line fbtft_device rotates your desktop, 
> 	you can change this (90, 180, 270)
> 
> 5) Edit /boot/cmdline.txt: sudo nano /boot/cmdline.txt
> 	ENTER ALL BELOW IN ONE LINE!
> 
> 	dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p6 rootfstype=ext4 elevator=deadline rootwait fbtft_device.custom fbtft_device.name=waveshare32b fbtft_device.gpios=dc:22,reset:27 fbtft_device.bgr=1 fbtft_device.speed=48000000 fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo dma.dmachans=0x7f35 console=tty1 consoleblank=0 fbtft_device.fps=50 fbtft_device.rotate=0
> 
> 6) sudo reboot
> 
> 7) Activating the touchscreen
> 	sudo nano /boot/config.txt
> 	"Insert this line on the end of config.txt"
> 	dtoverlay=ads7846,speed=500000,penirq=17,swapxy=1
> 
> 8) Calibation Entry, keeps the monitor steady or install xinput_calibrator
> 
> 	sudo nano /usr/share/X11/xorg.conf.d/99-calibration.conf
> 
> 	Section "InputClass"      
> 	Identifier      "calibration"     
> 	MatchProduct    "ADS7846 Touchscreen"      
> 	Option  "Calibration"  "3900 240 3900 240"      
> 	EndSection
> 
> sudo reboot
> 
> 9) Gimmick :-)
> 
> 	sudo apt-get install mplayer
> 	wget http://download.blender.org/peach/bigbuckbunny_movies/BigBuckBunny_320x180.mp4
> 
> 	DON'T START THE GUI WITH STARTX, STAY IN THE CONSOLE!
> 	Enter in the console
> 	mplayer -vo fbdev2:/dev/fb1 -vf scale=320:-3 BigBuckBunny_320x180.mp4
> 
> Have a nice day

**I followed these directions and My screen is still all white. I've been stuck on this for 3 days.**

```
dmesg | grep spi
[    0.000000] Linux version 4.0.7-v7+ (pi@raspi2) (gcc version 4.8.3 20140106 (prerelease) (crosstool-NG linaro-1.13.1-4.8-2014.01 - Linaro GCC 2013.11) ) #1 SMP PREEMPT Sat Jul 11 20:44:05 CEST 2015
[    2.602162] bcm2708_spi 3f204000.spi: DMA channel 2 at address 0xf3007200 with irq 77
[    2.606758] bcm2708_spi 3f204000.spi: DMA channel 4 at address 0xf3007400 with irq 20
[    2.628669] ads7846 spi0.1: touchscreen, irq 497
[    2.631815] input: ADS7846 Touchscreen as /devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input0
[    2.637409] bcm2708_spi 3f204000.spi: SPI Controller at 0x3f204000 (irq 80)
[    2.639905] bcm2708_spi 3f204000.spi: SPI Controller running in dma mode
```

**This looks OK, however my /dev/fb1 never appeared**
```
pi@raspberrypi:~ $ ls /dev/fb*
/dev/fb0
```

**And I have a single error** 

```
[    2.631815] input: ADS7846 Touchscreen as /devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input0
[    2.637409] bcm2708_spi 3f204000.spi: SPI Controller at 0x3f204000 (irq 80)
[    2.639905] bcm2708_spi 3f204000.spi: SPI Controller running in dma mode
[    2.647491] systemd[1]: Started udev Coldplug all Devices.
[    2.651554] systemd[1]: systemd-modules-load.service: main process exited, code=exited, status=1/FAILURE
[    2.665024] systemd[1]: Failed to start Load Kernel Modules.
[    2.668028] systemd[1]: Unit systemd-modules-load.service entered failed state.
[    2.675272] systemd[1]: Started File System Check on Root Device.
[    2.828345] systemd[1]: Starting Apply Kernel Variables...
[    2.838029] systemd[1]: Mounting Configuration File System...
[    2.847129] systemd[1]: Mounting FUSE Control File System...
[    2.857699] systemd[1]: Starting Create Static Device Nodes in /dev...
[    2.866258] systemd[1]: Starting Syslog Socket.
[    2.870529] systemd[1]: Listening on Syslog Socket.
[    2.870810] systemd[1]: Starting Journal Service...
[    2.881850] systemd[1]: Started Journal Service.
```

**Found the source of the issue**

```
Feb 28 02:47:09 raspberrypi systemd-modules-load[807]: Failed to find module 'fbtft_device rotate=270 name=waveshar
Feb 28 02:47:09 raspberrypi systemd-modules-load[807]: Failed to find module 'fb_ili9340 width=320 height=240 buswi
Feb 28 02:47:09 raspberrypi systemd[1]: systemd-modules-load.service: main process exited, code=exited, status=1/FA
Feb 28 02:47:09 raspberrypi systemd[1]: Failed to start Load Kernel Modules.
```

**How do i fix this?**

**OK... So i corrected the error by commenting out the config lines for my screen**

```
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.

i2c-dev
spi-bcm2708
ads7846

#fbtft_device name=waveshare32b gpios=dc:22,reset:27 speed=48000000

#fb_ili9340 width=320 height=240 buswidth=8 init=-1,0xCB,0x39,0x2C,0x00,0x34,0x02,-1,0xCF,0x00,0XC1,0X30,-1,0xE8,0$

```

**I'll just wait for some pointers. It appears that I'm missing a module for my touch screen or the config line is wrong in /etc/modules**

ravenousld3341, 2017-03-02:
The commands listed above didn't seem to resolve anything

notro, 2017-03-03:
notro/rpi-firmware is deprecated, use ordinary raspian kernel.
You can probably use a Device Tree overlay from here: https://github.com/swkim01/waveshare-dtoverlays

ravenousld3341, 2017-03-03:
I'm going to reimage my sd card tonight and start over.
What process would you recommend?

notro, 2017-03-03:
I don't know anything about the Waveshare displays, except that many people have problems with them.

ravenousld3341, 2017-03-03:
Ahhh.
I see. Thanks for your time. I've engaged another user over on swkim01's github for further assistance.
Thanks again!

closed 2017-03-03 by ravenousld3341