notro/fbtft #445: raspberry pi lastest Raspbian 2017-03-02-raspbian-jessie 99-calibration.conf not work Touch revers

DRAgon734465502, 2017-03-06:
The lastest raspbian OS have updated to 2017-03-02-raspbian-jessie 

But the touch can not work well ，I use waveshare 3.5inch LCD with ads7846/xpt2046

The file 99-calibration.conf can not work well, seem the file have stopped to work but it can work well at old version Raspbian

the Touch reverse, touch X axis but go to Y axis

please Who can help me ， I am very urgent

swkim01, 2017-03-07:
That is because now x server uses libinput to handle input devices on X11 server instead of evdev.
[https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=172025&p=1123456](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=172025&p=1123456)

As to [coordination transformation matrix](https://wiki.ubuntu.com/X/InputCoordinateTransformation), you have to add the following code within the section of touchscreen in 99-callibration.conf or 40-libinput.conf.
in case of inverting x,
`    Option "TransformationMatrix" "-1 0 1 0 1 0 0 0 1"`
in case of inverting y,
`    Option "TransformationMatrix" "1 0 0 0 -1 1 0 0 1"`

You can confirm the properties of the input device by following commands.
```
$ xinput list
$ xinput list-props "ADS7846 Touchscreen"
```

DRAgon734465502, 2017-03-07:
The lastest raspbian OS have updated to 2017-03-02-raspbian-jessie
But I have some questiones at fbtft Device Tree （fbtft old driver can display ok but touch Inverted   ）
my LCD with touch chip xpt2046

1 The calibration file “/etc/X11/xorg.conf.d/99-calibration.conf”   is not working Now.
if you change any parameters of 99-calibration.conf ，Nothing will happen

2. The xinput-calibrate software APP will not work well .
If you calibrate ，you will get the following very big number
(MaxX and MaxY are so big ,the old number at old raspbian is {Option  "Calibration"   "3932 300 294 3801"} and work well）
----------------------------------------------------------------------------------------------
Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ADS7846 Touchscreen"
        Option  "MinX"  "159"
        Option  "MaxX"  "65512"
        Option  "MinY"  "102"
        Option  "MaxY"  "65637"
        Option  "SwapXY"        "0" # unless it was already set to 1
        Option  "InvertX"       "0"  # unless it was already set
        Option  "InvertY"       "0"  # unless it was already set
EndSection
---------------------------------------------------------------------------------------------

3. The touch parameters “xmin ymin xmax ymax” of  dtoverlay=ads7846 or fbtft Device Tree  can not be set 
and if you change the 4 parameters , they will change nothing and keep the default setting

I  have removed the file “/etc/X11/xorg.conf.d/99-calibration.conf”  and Add Device Tree  ads7846 to /boot/config.txt
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
dtoverlay=ads7846,cs=1,penirq=17,penirq_pull=2,speed=1000000,keep_vref_on=1,swapxy=1,pmax=255,xohms=60,xmin=200,xmax=3900,ymin=200,ymax=3900
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Then the touch go to X  Inverted and I  change "xmin=200,xmax=3900 "  to "xmin=3900,xmax=200"  but nothing  happened

DRAgon734465502, 2017-03-07:
Now the big question is how we can calibrate the touchscreen？？ 

Becasue of the question 2, I can not get the right calibration number
Becasue of the question 1 and 3 , if I can get the right calibration number , I also do know how to make them work

swkim01, 2017-03-07:
With reference to [above link](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=172025&p=1123456), the other solution to calibrate without changing 99-calibration.conf file is to reuse evdev simply.

1. Install evdev package.
`$ sudo apt-get install xserver-xorg-input-evdev`
2. Move 99-calibration.conf to /usr/share/X11/xorg.conf.d and make sure that /etc/X11/xorg.conf.d is empty.
3. /usr/share/X11/xorg.conf.d should contain something like:
10-quirks.conf 40-libinput.conf 45-evdev.conf 70-synaptics.conf 99-calibration.conf 99-fbturbo.conf
Just be sure that evdev.conf has a higher number than libinput.conf
this forces evdev to load after libinput.

DRAgon734465502, 2017-03-07:
@swkim01 
can you tell me my question 3
The touch parameters “xmin ymin xmax ymax” of dtoverlay=ads7846 or fbtft Device Tree can not be set
and if you change the 4 parameters , they will change nothing and keep the default setting

I have removed the file “/etc/X11/xorg.conf.d/99-calibration.conf” and Add Device Tree ads7846 to /boot/config.txt
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
dtoverlay=ads7846,cs=1,penirq=17,penirq_pull=2,speed=1000000,keep_vref_on=1,swapxy=1,pmax=255,xohms=60,xmin=200,xmax=3900,ymin=200,ymax=3900
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Then the touch go to X Inverted and I change "xmin=200,xmax=3900 " to "xmin=3900,xmax=200" 
But nothing happened

swkim01, 2017-03-07:
@DRAgon734465502 
I don't know what exactly the parameters of the touchscreen driver mean.
My guess is that they are just "ranges" of axes, not "limit values"...

DRAgon734465502, 2017-03-07:
@swkim01 
Sorry to disturb you again,   I have tried your methion

But also do not know how to calibrate

I have done what you writed 1,2,3

When run the calibration xinput APP, also get very big calibrate number "22186" "22186""48964""48827"

I write them to file  99-calibration.conf and reboot， change nothing

DRAgon734465502, 2017-03-07:
what do you mean “, the other solution to calibrate without changing 99-calibration.conf file is to reuse evdev simply.” ？ 

“ reuse evdev simply.” is meaning what？

swkim01, 2017-03-07:
@DRAgon734465502 
That means the X11 server use the touchscreen device by evdev module, not new libinput.
You can use it on x window with old settings as it was.
Please compare two output results of the following command with and without 45-evdev.conf.
`$ xinput list-props "ADS7846 Touchscreen"`

DRAgon734465502, 2017-03-07:
with 
[image]

DRAgon734465502, 2017-03-07:
I have compared two output results of the following command with and without 45-evdev.conf.
$ xinput list-props "ADS7846 Touchscreen"

they are same output 
1. under  /usr/share/X11/xorg.conf.d/ ,not have 45-evdev.conf.
   I  enter the command： $ xinput list-props "ADS7846 Touchscreen"

2. I make a new file 45-evdev.conf under /usr/share/X11/xorg.conf.d/
    $  sudo nano /usr/share/X11/xorg.conf.d/45-evdev.conf
     And add the following code to it
Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ADS7846 Touchscreen"
        Option  "Calibration"   "3932 300 294 3801"
        Option "SwapAxes" "0"
#       Option "TransformationMatrix" "0 -1.10 1.06 1.12 0 -0.06 0 0 1"
        Option "TransformationMatrix" "0 -1 1 1 0 0 0 0 1"
EndSection

Then enter the the command： $ xinput list-props "ADS7846 Touchscreen"

Get same results ：
[image]

I just want to be sure that
Now we can not calibrate by writing the 4 numbers <Option  "Calibration"   "3932 300 294 3801">:

And just can calibrate by set <Option "TransformationMatrix" "0 -1 1 1 0 0 0 0 1">  
                                        To  <Option "TransformationMatrix" "0 -1.10 1.06 1.12 0 -0.06 0 0 1">
???
If so, which is too hard to calibrate  ， I find that when I set 
<Option "TransformationMatrix" "0 -1 1 1 0 0 0 0 1">   the  touch is not very Accurate
So I change to set to <Option "TransformationMatrix" "0 -1.10 1.06 1.12 0 -0.06 0 0 1">
Then touch is very Accurate but I change the number by eye and Increase or decrease to get the best number  which is so difficult

swkim01, 2017-03-07:
It seems that my comment was not exact. You have to get different outputs.
The content of 45-evdev.conf is as follows or same as 10-evdev.conf.
```
# ...
Section "InputClass"
        Identifier "evdev touchscreen catchall"
        MatchIsTouchscreen "on"
        MatchDevicePath "/dev/input/event*"
        Driver "evdev"
EndSection
```
Then reboot or restart X window.
Now since x window uses evdev module for input device, you have to get the result like as :
```
$ xinput list-props "ADS7846 Touchscreen"
Device 'ADS7846 Touchscreen':
	Device Enabled (113):	1
	Coordinate Transformation Matrix (114):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (235):	0
	Device Accel Constant Deceleration (236):	1.000000
	Device Accel Adaptive Deceleration (237):	1.000000
	Device Accel Velocity Scaling (238):	10.000000
	Device Product ID (239):	0, 0
	Device Node (240):	"/dev/input/event3"
	Evdev Axis Inversion (241):	1, 0
	Evdev Axis Calibration (242):	<no items>
	Evdev Axes Swap (243):	0
	Axis Labels (244):	"Abs X" (232), "Abs Y" (233), "Abs Pressure" (234)
	Button Labels (245):	"Button Unknown" (231), "Button Unknown" (231), "Button Unknown" (231), "Button Wheel Up" (119), "Button Wheel Down" (120)
	Evdev Scrolling Distance (246):	0, 0, 0
	Evdev Middle Button Emulation (247):	0
	Evdev Middle Button Timeout (248):	50
	Evdev Third Button Emulation (249):	0
	Evdev Third Button Emulation Timeout (250):	1000
	Evdev Third Button Emulation Button (251):	3
	Evdev Third Button Emulation Threshold (252):	20
	Evdev Wheel Emulation (253):	0
	Evdev Wheel Emulation Axes (254):	0, 0, 4, 5
	Evdev Wheel Emulation Inertia (255):	10
	Evdev Wheel Emulation Timeout (256):	200
	Evdev Wheel Emulation Button (257):	4
	Evdev Drag Lock Buttons (258):	0
```

DRAgon734465502, 2017-03-07:
Yes
I get that result

GAMELASTER, 2017-03-14:
@DRAgon734465502 
Try to rotate your display, here you have some matrices:
Right rotate
`xinput set-prop 'ADS7846 Touchscreen' 'Coordinate Transformation Matrix' 0 1 0 -1 0 1 0 0 1`
Left rotate
`xinput set-prop 'ADS7846 Touchscreen' 'Coordinate Transformation Matrix' 0 1 0 -1 0 1 0 0 1`
Invert
`xinput set-prop 'ADS7846 Touchscreen' 'Coordinate Transformation Matrix' -1 0 1 0 -1 1 0 0 1`

GottZ, 2017-10-05:
i created a small visualizer for it: https://codepen.io/GottZ/pen/d73f2f844b52b91b7457febce2d1b18c
just throw the first 6 numbers into the input field at the bottom.
tell me if you really need the other three numbers. (i'd have to move away from how i'm currently doing it)
it has a live preview.
black is basically the display and red is the touch data.

GottZ, 2017-10-06:
new feature using arrow keys:
[image]

KurtJacobson, 2017-10-06:
@GottZ That is fantastic! Just what I needed to visualize what was happening. Thank you very much.

gamelaster, 2017-10-07:
https://wiki.archlinux.org/index.php/Calibrating_Touchscreen
There is way how to calibrate touch screen. Maybe it will be useful to make some calibration application

GottZ, 2017-10-07:
added some instructions etc. and made it public. well..
i guess the next step would be to implement xinput-calibrator for matrices

KurtJacobson, 2017-10-07:
Yes, it would be great to make something similar to xinput-calibrator, but for matrices. I was playing with this last night:  https://github.com/KurtJacobson/xtcal

It has a very similar interface to xinput-calibrator and seems promising.

gamelaster, 2017-10-08:
Good job guys 😎

sethml, 2018-01-02:
Making the touchscreen device default to the evdev driver rather than the libinput driver is pretty simple, and solves the problem. Make sure evdev driver is installed:
`$ sudo apt-get install xserver-xorg-input-evdev`

Then simply add a Driver directive - my `/usr/share/X11/xorg.conf.d/99-calibration.conf` looks like:
```
Section "InputClass"
  Identifier "calibration"
  MatchProduct "ADS7846 Touchscreen"
  Option "Calibration" "56 4019 3893 120"
  Option "SwapAxes" "1"
  Driver "evdev"
EndSection
```

Works well for my WaveShare 7" HDMI Touchscreen.

gamelaster, 2018-01-18:
https://github.com/gamelaster/evdev-calibration
I did a calibration app, it's not perfect, but it's works

TurboTronix, 2019-01-16:
Black on red best combination for color blind ppl :P

I'm going to see if I can manage to figure how to use these tools as I'm having issues with my touchscreen x axis, keeps going to far to the right into my other screen for some reason...

closed 2021-08-24 by github-actions[bot]