notro/fbtft #446: dtoverlay usage

Zhairgling, 2017-03-09:
Hi I'm using the p2090 ( https://www.adafruit.com/products/2090 ).

he is working and configured.

I'm using the manual methode for loading the module : 

```
sudo modprobe fbtft_device custom name=fb_ili9341  gpios=reset:25,dc:24,led:18 speed=16000000 rotate=0 bgr=1
```

But I need it to start at boot so I now that in past I should load it by  /etc/modules.

But now doesn't work.

Anybody can help me to write the good dtoverlay ?

thanks all

notro, 2017-03-09:
Strip the touch part and change the gpios: https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/rpi-display-overlay.dts

Zhairgling, 2017-03-10:
Strip the touch part ?

notro, 2017-03-10:
I thought you didn't use touch since you used fbtft_device.

Since it's capacitive, maybe this can be of use: https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/pitft28-capacitive-overlay.dts

Zhairgling, 2017-03-13:
@notro yes i'm using the touchscreen. this could help me but . I don't understand how use the .dts file, it isn't compiled ? in hexa ?

notro, 2017-03-13:
https://github.com/notro/fbtft/wiki/FBTFT-RPI-overlays

Zhairgling, 2017-03-13:
ok I will see. Thank you @notro

Zhairgling, 2017-05-17:
Hi @notro , 

I definitely need help with my screen. The existence of two different overlays none of which works in my configuration makes me confused. I tried to tinker with them without success. It seems that I am doing wrong.
I do not ask for a turnkey solution but if you can gag me it would be cool.

I use this screen https://www.adafruit.com/product/2090

It is handled by FT6206 for the touch, and an ili9341.

I set up with the two dts file with the good gpio, like the params of that modprobe:

sudo modprobe fbtft_device custom name=fb_ili9341  gpios=reset:25,dc:24,led:18 speed=16000000 rotate=0 bgr=1

And followed the procedure you gave me earlier.

I'm using a clean raspian minimal on a raspberry zero

I wish you could help me

notro, 2017-05-17:
Try this in /boot/config.txt
```
dtoverlay=pitft28-capacitive,speed=16000000
```
Src: https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/pitft28-capacitive-overlay.dts

D/C have to be wired to gpio 25.
According to this reset doesn't have to be connected: https://learn.adafruit.com/adafruit-2-dot-8-color-tft-touchscreen-breakout-v2/spi-wiring
Let LITE float high while testing since the overlay doesn't set the led gpio.

The touch controller won't work because the driver is missing in 4.9, but I see that it was added back yesterday and will be available in rpi-update in a few days.

Zhairgling, 2017-05-18:
Thank you @notro,

My screen now display my desktop :)
but like you said no touch for the moment.
I have by the way some error on the startx : 
```
 
pi@raspberrypi:~ $ sudo startx -- -layout TFT

X.Org X Server 1.18.4
Release Date: 2016-07-19
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
Current Operating System: Linux raspberrypi 4.9.24+ #993 Wed Apr 26 17:56:54 BST 2017 armv6l
Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1440 bcm2708_fb.fbheight=900 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:A8:0F:F6 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=fddb7552-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 11 November 2016  11:59:59AM
xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
Current version of pixman: 0.33.3
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Thu May 18 10:46:31 2017
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found.
The XKEYBOARD keymap compiler (xkbcomp) reports:
> Warning:          Type "ONE_LEVEL" has 1 levels, but <RALT> has 2 symbols
>                   Ignoring extra symbols
Errors from xkbcomp are not fatal to the X server
```

Zhairgling, 2017-06-01:
Hi @notro ,

I come to you again, I've solved easily the XKEYBOARD error.
I've done all days rpi-update but don't have yet the touch driver, 
Any advices ?

notro, 2017-06-01:
The module/driver has a different name now: edt-ft5x06, but the Device Tree is the same.
Verify it's presence:
```
$ modinfo edt-ft5x06
```

Zhairgling, 2017-06-01:
ok thank's dude, I will try tomorrow

Zhairgling, 2017-06-02:
Hi @notro ,

I've tried to add it in /boot/config.txt. But my screen isn't tactil yet.

However : 

'''
pi@raspberrypi:~ $ modinfo edt-ft5x06
filename:       /lib/modules/4.9.30-v7+/kernel/drivers/input/touchscreen/edt-ft5x06.ko
license:        GPL
description:    EDT FT5x06 I2C Touchscreen Driver
author:         Simon Budig <simon.budig@kernelconcepts.de>
srcversion:     69FD13E4508DBE143E8296F
alias:          i2c:ft6236
alias:          i2c:edt-ft5506
alias:          i2c:edt-ft5x06
alias:          of:N*T*Cfocaltech,ft6236C*
alias:          of:N*T*Cfocaltech,ft6236
alias:          of:N*T*Cedt,edt-ft5506C*
alias:          of:N*T*Cedt,edt-ft5506
alias:          of:N*T*Cedt,edt-ft5406C*
alias:          of:N*T*Cedt,edt-ft5406
alias:          of:N*T*Cedt,edt-ft5306C*
alias:          of:N*T*Cedt,edt-ft5306
alias:          of:N*T*Cedt,edt-ft5206C*
alias:          of:N*T*Cedt,edt-ft5206
depends:
intree:         Y
vermagic:       4.9.30-v7+ SMP mod_unload modversions ARMv7 p2v8
'''
did you have an idea ?

notro, 2017-06-02:
Use evtest for testing:
```
$ apt-get install evtest
$ evtest
input devices will be listed, and touch should be among them.
```

Zhairgling, 2017-06-08:
When I use evtest, my touchscreen isn't listed : 
```

pi@raspberrypi:~ $ evtest
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:      USB Keyboard
/dev/input/event1:      USB Keyboard
/dev/input/event2:      ASUS GLADIUS
/dev/input/event3:      ASUS GLADIUS
/dev/input/event4:      EP0110M09
Select the device event number [0-4]:
```

notro, 2017-06-08:
This means that the driver finds the touch controller at least.
Check if you get any interrupts on the gpio: ```cat /proc/interrupts```

Zhairgling, 2017-06-08:
he is registred too .

```
pi@raspberrypi:~ $ cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3
 16:          0          0          0          0  bcm2836-timer   0 Edge      ar                                                                                                                                    ch_timer
 17:       1797       2684       7357       4913  bcm2836-timer   1 Edge      ar                                                                                                                                    ch_timer
 23:         56          0          0          0  ARMCTRL-level   1 Edge      3f                                                                                                                                    00b880.mailbox
 24:          2          0          0          0  ARMCTRL-level   2 Edge      VC                                                                                                                                    HIQ doorbell
 46:          0          0          0          0  ARMCTRL-level  48 Edge      bc                                                                                                                                    m2708_fb dma
 48:          0          0          0          0  ARMCTRL-level  50 Edge      DM                                                                                                                                    A IRQ
 50:          0          0          0          0  ARMCTRL-level  52 Edge      DM                                                                                                                                    A IRQ
 51:       2005          0          0          0  ARMCTRL-level  53 Edge      DM                                                                                                                                    A IRQ
 54:          0          0          0          0  ARMCTRL-level  56 Edge      DM                                                                                                                                    A IRQ
 55:         38          0          0          0  ARMCTRL-level  57 Edge      DM                                                                                                                                    A IRQ
 56:         38          0          0          0  ARMCTRL-level  58 Edge      DM                                                                                                                                    A IRQ
 57:          0          0          0          0  ARMCTRL-level  59 Edge      DM                                                                                                                                    A IRQ, DMA IRQ
 59:          0          0          0          0  ARMCTRL-level  61 Edge      bc                                                                                                                                    m2835-auxirq
 62:     274766          0          0          0  ARMCTRL-level  64 Edge      dw                                                                                                                                    c_otg, dwc_otg_pcd, dwc_otg_hcd:usb1
 79:          0          0          0          0  ARMCTRL-level  81 Edge      3f                                                                                                                                    200000.gpio:bank0
 80:          0          0          0          0  ARMCTRL-level  82 Edge      3f                                                                                                                                    200000.gpio:bank1
 83:        905          0          0          0  ARMCTRL-level  85 Edge      3f                                                                                                                                    804000.i2c
 84:          0          0          0          0  ARMCTRL-level  86 Edge      3f                                                                                                                                    204000.spi
 86:        192          0          0          0  ARMCTRL-level  88 Edge      mm                                                                                                                                    c0
 87:       4880          0          0          0  ARMCTRL-level  89 Edge      ua                                                                                                                                    rt-pl011
 92:       8502          0          0          0  ARMCTRL-level  94 Edge      mm                                                                                                                                    c1
190:          0          0          0          0  pinctrl-bcm2835  24 Edge                                                                                                                                          ft6236
FIQ:              usb_fiq
IPI0:          0          0          0          0  CPU wakeup interrupts
IPI1:          0          0          0          0  Timer broadcast interrupts
IPI2:       2114       5940       2807       4318  Rescheduling interrupts
IPI3:         18         13         13         14  Function call interrupts
IPI4:          0          0          0          0  CPU stop interrupts
IPI5:        286        400         55         43  IRQ work interrupts
IPI6:          0          0          0          0  completion interrupts
Err:          0
```
We can see ```pi@raspberrypi:~ $ 190:          0          0          0          0  pinctrl-bcm2835  24 Edge                                                                                                                                          ft6236```

notro, 2017-06-08:
All zeroes, so there's has been no interrupt registered on that pin. Check wiring.

Zhairgling, 2017-06-09:
You are right. Thank you so much @notro

notro, 2017-06-10:
The overlay has several arguments: https://github.com/raspberrypi/linux/blob/rpi-4.9.y/arch/arm/boot/dts/overlays/README#L1031

martonborzak, 2017-09-02:
Hi,

@Zhairgling I am also trying to do the same setup as you. However cannot get the touch working. How did you wire up the display to the raspberry? 

Thanks,
Marton

Zhairgling, 2017-09-04:
hi,
Did you use evetest for testing the touchscreen ?

closed 2018-05-10 by notro