notro/fbtft #447: has the modprobe command changed?

moosepr, 2017-03-09:
Hi there

First off, top job on the driver, it works a treat, and opens up many many possibilities

i have been using ili9341 screens for a few months, and have got the settings fine tuned using retropie, have even made my own board to work it all (insert shameless plug) https://hackaday.io/project/20152-gameboy-zero-but-smaller

I had decided to have a play with pico-8, so i installed a fesh install of jessie lite, got wifi, ssh, and used the same config for fbtft and fbcp that i have done dozens of times. reboot and no screen!!! so i go back to basics and i cant for the life of me get the screen to work!! 

my normal magic command is
`sudo modprobe fbtft_device custom name=fb_ili9341  gpios=reset:25,dc:24 speed=80000000 bgr=1`

but every time i get

`modprobe: ERROR: could not insert 'fbtft_device': Invalid argument`

no matter what combination of commands and options i use, i cant seem to get it to work. Thought i was a bit of an expert with the ili9341 by now!! feel such a noob

notro, 2017-03-09:
Anything in the kernel log? ```$ dmesg```

moosepr, 2017-03-09:
ah good call!! should have been the first one to check.

for some reason it doesnt like me specifying the gpios

`[  172.988537] fbtft_device: `' invalid for parameter `gpios'
[  172.988569] fbtft_device: unknown parameter 'reset:25,dc:24' ignored
`

moosepr, 2017-03-09:
Righty!!! apologies for the silly questions, seems i made a boob on my fbtft.conf, seems i missed the = between the gpios command and the list of options. Thanks for pointing me in the right direction though

on a side note, i was looking at the datasheet for the ili9341 screens, and saw there was a tearing option buried within the commands.

im assuming this is not something we have direct access to using spi?

notro, 2017-03-09:
There's a TE (tearing effect) output pin on many controllers, but I have yet to see a display that exposes this pin.

moosepr, 2017-03-09:
ah that makes sense

thanks again for the drivers and the help!!! :)

closed 2017-03-09 by moosepr