notro/fbtft #448: SSD1322 OLED Support

BiatuAutMiahn, 2017-03-12:
I know this is a duplicate, but how would I go about getting this driver into the latest branch? Can i just drop it in?

notro, 2017-03-13:
What branch are you referring to?

BiatuAutMiahn, 2017-03-13:
#199

notro, 2017-03-13:
You can probably just drop it in. The FBTFT_REGISTER_DRIVER() macro needs additional Device Tree arguments though. Apart from that I don't know of anything that has changed.

BiatuAutMiahn, 2017-03-13:
ok cool, can you provide an example for the FBTFT_REGISTER_DRIVER()?

notro, 2017-03-13:
http://lxr.free-electrons.com/ident?i=FBTFT_REGISTER_DRIVER

BiatuAutMiahn, 2017-03-13:
ok, Thank you. Another question, do you think this driver will run under android?

notro, 2017-03-15:
I don't know Android, but fbtft seems to be present https://android.googlesource.com/kernel/common/+/android-4.9/drivers/staging/fbtft/

BiatuAutMiahn, 2017-03-15:
ok cool, thank you

closed 2018-05-10 by notro