notro/fbtft #451: Random pixels with Joy-it ST7735

rwmodeltrain, 2017-03-18:
Hi all,

I am having troubles with the Joy-it ST7735  (RB-TFT1.8-V2) together with my Raspberry B+. I've got two other 1.8" TFT screens, they are from sainsmart - they look exactly the same but seem to work different. They do work fine.

As you can see on the photo ([click](https://github.com/rwmodeltrain/departure-board/blob/master/wiki_images/TFT_borders.jpg?raw=true)), there is a border on two sides of the screen with what seems to be random pixels. The other difference I noticed was that I needed to use de bgr=1 option to have the colors displayed correct. 

`sudo mod probe fbtft_device name=sainsmart18 rotate=270 speed=26000000 bgr=1`

I think I need to do something else to get rid of the borders - but I don't know what. It seems there are different versions of this screen, mine had a screen protector with a green label. On some Arduino forums I've seen that a tweak has to be done to make it work on the Arduino. First I thought the TFT screen was broken, but I am not sure about that anymore after reading the Arduino topics.

So I would like to know whether it is possible to add a parameter to modprobe to make this work? I've searched a lot of forums but didn't find the answers. Can you help me out?

Thanks,
Richard

notro, 2017-03-22:
You can try ```name=adafruit18_green```.
This one offsets the window in GRAM that gets updated: http://lxr.free-electrons.com/source/drivers/staging/fbtft/fbtft_device.c#L1296

rwmodeltrain, 2017-03-22:
Tnx for your response! The result is better, though not ideal yet. Where I first had 2 'random' rows at the bottom, there is now still one row, and de single random row right now appears left.

So without really understanding what the code in the link you sent is used for, I think it would work if it would be:

```
1296 static void adafruit18_green_tab_set_addr_win(struct fbtft_par *par,
1297                                                 int xs, int ys, int xe, int ye)
1298 {
1299         write_reg(par, 0x2A, 0, xs + 1, 0, xe + 1); (instead of 2)
1300         write_reg(par, 0x2B, 0, ys + 2, 0, ye + 2); (instead of 1)
1301         write_reg(par, 0x2C);
1302 }
```

But.. no idea what to do to try this or make this work... is it possible and can you give me a hint/direction?

Richard

notro, 2017-03-22:
Ok, that makes sense.
You need to rebuild the kernel to make such a change: https://www.raspberrypi.org/documentation/linux/kernel/building.md
Local building on a B+ takes many hours, but it's possible.

rwmodeltrain, 2017-03-22:
Ok, I will give it a try! Thank you!

Richard

rwmodeltrain, 2017-03-23:
That did the job! Tnx.

closed 2017-03-23 by rwmodeltrain

teknoid, 2018-11-12:
Confirmed. Same issue on my Joy-It RB-TFT1.8-V2.
Maybe these "offset" values could be extracted into a module parameter in a future release?