notro/fbtft #453: Ili9163

moosepr, 2017-03-26:
Howdy

Sorry to bother you again, but I have been playing with some cheep 1.44 inch ili9163 screens and have stumbled upon an issue with the rotation.

Using 0 or 90 for the rotation, the screen works fine. Using 180 or 270, the chunk of the screen closest to the chip is junk. Looks to be about 32 pixels or so

https://imgur.com/gallery/YRp66

Oddly enough, I initially tried the screen as st7735, but the driver is trying to do 128x160, so its not ideal. I did try and force the driver to 128x128 resolution, and I got a similar 32 rows of junk closest to the driver chip.

Is this a known/common one?

notro, 2017-03-27:
The rotation problem is known. I have put panel specific rotation values in a controller driver and it can't be overridden...

st7735 is a MIPI DCS compatible display (set_addr_win is not set, uses the default or has 2a, 2b, 2c commands), so you can use another such driver that doesn't have .set_var().
This means you have to provide the initialization through the init=argument.
If you use the custom argument you can set width/height.
https://github.com/notro/fbtft/wiki/fbtft_device#custom-display

moosepr, 2017-03-27:
thanks for your reply

running on st7735r the screen is the right way round, but too tall

options fbtft_device name=fb_st7735r gpios=reset:25,dc:24 speed=40000000 bgr=1 rotate=0 custom=1 fps=60
http://i.imgur.com/UYkWUSa.jpg (see the bar is not centred)

so if i add some screen size in there

options fbtft_device name=fb_st7735r gpios=reset:25,dc:24 speed=40000000 height=128 bgr=1 rotate=0 custom=1 fps=60

http://i.imgur.com/IRkeIXP.jpg (the shape is better, but i have my hunk of garbage back)

do i just need to tweak some init arguments to get to start in the right place?

notro, 2017-03-27:
Register 0x36 controls mirror/flip and it might be what's causing you problems.
You can try the flexfb driver and have full control on the settings: https://github.com/notro/fbtft/wiki/flexfb#sainsmart-18-st7735r

moosepr, 2017-03-27:
top man!! i have dropped it into the fbtft.conf and it doesnt work (yet) i will keep playing :)

closed 2021-08-24 by github-actions[bot]