notro/fbtft #454: flexfb spi0.0: missing stop marker at end of init sequence

Stevelande, 2017-03-30:
Sorry for my bad English i am from Germany. 
I have a SPI Display at the RPI 1.

the Command at the Console works fine `sudo modprobe flexfb nobacklight regwidth=16 init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29,-3 width=480 height=320`

`sudo modprobe fbtft_device name=flexfb speed=16000000 gpios=reset:25,dc:24`

but not after restart

my /etc/modules-load.d/fbtft.conf
`:~ $ cat /etc/modules-load.d/fbtft.conf 
spi-bcm2835
flexfb
fbtft_device
`
my /etc/modprobe.d/flexfb.conf
`:~ $ cat /etc/modprobe.d/flexfb.conf
options flexfb nobacklight width=320 height=480 regwidth=16 
init=-1,0xb0,0x0,-1,0x11,-2,250,-1,0x3A,0x55,-1,0xC2,0x44,-1,0xC5,0x00,0x00,0x00,0x00,-1,0xE0,0x0F,0x1F,0x1C,0x0C,0x0F,0x08,0x48,0x98,0x37,0x0A,0x13,0x04,0x11,0x0D,0x00,-1,0xE1,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0xE2,0x0F,0x32,0x2E,0x0B,0x0D,0x05,0x47,0x75,0x37,0x06,0x10,0x03,0x24,0x20,0x00,-1,0x36,0x28,-1,0x11,-1,0x29
init=-3
options fbtft_device name=flexfb speed=16000000 gpios=reset:25,dc:24`

dmesg
`:~ $ dmesg
[    5.929120] spidev spi0.0: Deleting spi0.0
[    5.940878] flexfb spi0.0: missing stop marker at end of init sequence
[    5.941037] flexfb: probe of spi0.0 failed with error -22
`
i not understand why works at the console but not in file

can you help me?

notro, 2017-03-31:
> missing stop marker at end of init sequence

Not a good error message, but it means that init= contains more than 512 values.
Ref: http://lxr.free-electrons.com/ident?i=fbtft_init_display

I can't see anything wrong, you could try and pass the init=argument to fbtft_device instead.

Stevelande, 2017-03-31:
thanks for you anwser. I found the Solution here [](https://github.com/goodtft/LCD-show)

closed 2021-08-24 by github-actions[bot]