notro/fbtft #456: ILI9341 based display on AM335x

shrikant97, 2017-04-05:
Hello Notro,
I am trying to bring up the ILI9341 based LCD on AM335x based custom board.
The interface used is System-80 interface.
I compiled the FBTFT drivers and integrated into kernel.

But I am confused about the creating node in the DTSI file.
Kindly guide me for the same.

Best Regards,
Shrikant

notro, 2017-04-12:
The output is confusing, it seems you are adding two devices: itdb28.9 and fb_ili9341.0

tajudheenk, 2019-10-23:
Try using fbtft drivers from kernel v5.0. They worked for me. 
You will find it here
[https://github.com/torvalds/linux/tree/v5.0/drivers/staging/fbtft](https://github.com/torvalds/linux/tree/v5.0/drivers/staging/fbtft)

Here is my dts entry for ssd1351

```
&spi1 {
        status = "okay";
        pinctrl-names = "default";
        pinctrl-0 = <&pinctrl_spi1_default>;

        lcd@0{
                compatible = "solomon,ssd1351";
                status = "okay";
                spi-max-frequency = <5000000>;
                width = <128>;
                height = <128>;
                fps = <20>;
                reg = <0>;
                bgr;
                regwidth = <0x8>;
                buswidth = <8>;
                backlight = <2>;
                debug = <3>;
                reset-gpios = <&gpio 15 GPIO_ACTIVE_LOW>;
                dc-gpios = <&gpio 16 GPIO_ACTIVE_LOW>;
        };
};
```

closed 2021-08-24 by github-actions[bot]