notro/fbtft #457: Resistive touch input on latest Raspbian does not rotate

rricharz, 2017-04-06:
Whatever I do, I am unable to rotate the resistive touch input using the desktop with the lastest Raspbian Jessie. After calibration, the touch input works properly using the command line interface.

My setup is as follows:

**pi@gameboy:~ $ uname -a
Linux gameboy 4.4.50-v7+ #970 SMP Mon Feb 20 19:18:29 GMT 2017 armv7l GNU/Linux
pi@gameboy:~ $ dmesg | grep mpe
[    3.154383] stmpe-spi spi0.1: stmpe610 detected, chip id: 0x811
[    3.172747] input: stmpe-ts as /devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/stmpe-ts/input/input2
pi@gameboy:~ $ cat /boot/config.txt | grep pitft28
dtoverlay=pitft28-resistive,rotate=270,speed=48000000,fps=60
pi@gameboy:~ $**

What is the proper procedure to rotate the touch input device using the built in stmpe driver for the resistive touch screen (2.8 inch Adafruit) on the latest Raspbian deskop?

ravenousld3341, 2017-04-06:
In /boot/config.txt try adding
`dtoverlay=pitft28-resistive,rotate=270,speed=48000000,fps=60,**swapxy=1**`

rricharz, 2017-04-06:
Good idea. Unfortunately neither swapxy, SwapXY, swapaxes, SwapAxes does change anything.

rricharz, 2017-04-06:
I finally traced the problem down to the use of libinput instead of evdev in the newest version of Raspbian ( 4.4.50-v7+ #970 SMP Mon Feb 20).

The following fix switches back to evdev:
  cd /usr/share/X11/xorg.conf.d
  sudo mv  10-evdev.conf 45-evdev.conf

now you can run xinput_calibrator and put the result in /usr/share/X11/xorg.conf.d/99-calibration.conf
Make sure that there is no 99-calibration.conf in /etc/X11/xorg.conf.d

My 2.8" resistive touch screen is now properly calibrated both on the command line interface and on the desktop. Eventually it would be nice to get this to work with libinput.

ljdelight, 2017-10-19:
I'm on the latest `squeeze` (4.9.56-v7+) and I see a similar issue using pitft35. This gets it working under evdev:

```
cat > /boot/config.txt <<EOL
dtparam=spi=on
dtparam=i2c1=on
dtparam=i2c_arm=on
dtoverlay=pitft35-resistive,rotate=270,speed=42000000,fps=60
EOL

apt-get install xserver-xorg-input-evdev
mv /usr/share/X11/xorg.conf.d/10-evdev.conf /usr/share/X11/xorg.conf.d/45-evdev.conf
echo > /usr/share/X11/xorg.conf.d/99-fbturbo.conf
```

If the touchscreen screen is back, the then kernel sees the screen. The above should fix the inverted touch and rotation. If there's still an issue check that `evdev` is controlling by reading the output of
`FRAMEBUFFER=/dev/fb1 startx -- -verbose 6 -logverbose 6`

closed 2021-08-24 by github-actions[bot]