notro/fbtft #459: Disabling/re-enabling fbtft at will

theLuthier, 2017-04-20:
I'm working on [retro handheld project](http://bitfixgaming.boards.net/post/4168/thread) using rpi3 + retropie + pitft 2.8" + fbtft. Most of the time I expect to use the system in handheld mode via pitft + fbtft, which I have setup and it's working beautifully.

However I'd like to maintain the ability to plug in a composite cable from the rpi to a TV for playing on a bigger screen. Could you help me figure out how I could essentially disable fbtft while re-enabling composite out, and then back again? Ideally I want to press a button connected to GPIO to run a script for swapping display out modes without having to reboot.

Is this even possible? I've been playing around with the **tvservice -c** command which clearly sends an SDTV signal to my TV, but only a blank screen. I figure it's something to do with fbtft. Any help is appreciated! I've used fbtft on 2 projects and it's fantastic - Thanks for your work.

notro, 2017-04-23:
Those have different framebuffers so they operate independently: /dev/fb0 is HDMI/SDTV and /dev/fb1 is fbtft
I think this question is better asked in the pi forums.

theLuthier, 2017-04-24:
Ok I'll try there - Thanks for the reply @notro.

closed 2017-04-24 by theLuthier