notro/fbtft #462: Two displays two different framebuffers the same driver

voloviq, 2017-05-04:
Hi,
I'm curious whether it is possible to register two different framebuffers (fb0, fb1) using the same driver but with different chip select. I'm quite sure it is possible on SPI displays but on display which uses D0-D7, RD, WR, A0 lines it is possible. How to specify dts for that approach.

Thanks in advanced
BR
Voloviq

notro, 2017-05-04:
You have to use Device Tree,  fbtft_device can only add 1 device.

Here's an example for Raspberry Pi
```
/*
 * Device Tree overlay for the ITDB02-2.8 display
 *
 */

/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2708";

	fragment@0 {
		target-path = "/soc";
		__overlay__ {
			itdb02_28: itdb02_28 {
				compatible = "ilitek,ili9325";
				#address-cells = <1>;
				#size-cells = <1>;

				dc-gpios = <&gpio  3 0>; /* RS */
				cs-gpios = <&gpio 27 0>;
				wr-gpios = <&gpio  2 0>;
				rd-gpios = <&gpio 22 0>;
				db-gpios = <&gpio  9 0>, /* DB08 */
					   <&gpio 11 0>, /* DB09 */
					   <&gpio 18 0>, /* DB10 */
					   <&gpio 23 0>, /* DB11 */
					   <&gpio 24 0>, /* DB12 */
					   <&gpio 25 0>, /* DB13 */
					   <&gpio  8 0>, /* DB14 */
					   <&gpio  7 0>; /* DB15 */

				buswidth = <8>;
				rotate = <0>;
				reset-gpios = <&gpio 17 0>;
			};

			itdb02_28_2: itdb02_28_2 {
<second display>
			};
		};
	};

	__overrides__ {
		rotate =  <&itdb02_28>,"rotate:0";
	};
};
```

voloviq, 2017-05-05:
Hi Notro,
Thanks for the answer. Yes, of course, I used device tree but I think I do not precisely describe my issue. So my goal is to use both displays at the same time. Currently I made some improvement in my dts and now looks as follow

[   69.366929] fb_nt7108: module is from the staging directory, the quality is unknown, you have be.
[   69.384294] fbtft_of_value: width = 192
[   69.399760] fbtft_of_value: height = 64
[   69.403670] fbtft_of_value: buswidth = 8
[   69.408925] fbtft_of_value: bpp = 16
[   69.412546] fbtft_of_value: debug = 0
[   69.416239] fbtft_of_value: rotate = 0
[   69.420827] fbtft_of_value: fps = 10
[   69.424441] fbtft_of_value: txbuflen = 49152
[   69.580654] graphics fb0: fb_nt7108 frame buffer, 192x64, 24 KiB video memory, 4 KiB DMA buffer 0
root@chiliboard:/lib/modules/4.9.13-g2d7a080e68/kernel/drivers/staging/fbtft# insmod fb_st7565p.ko
[   75.476416] fb_st7565p: module is from the staging directory, the quality is unknown, you have b.
[   75.493577] fbtft_of_value: width = 128
[   75.507438] fbtft_of_value: height = 64
[   75.511344] fbtft_of_value: buswidth = 8
[   75.515301] fbtft_of_value: bpp = 16
[   75.522116] fbtft_of_value: debug = 0
[   75.525836] fbtft_of_value: rotate = 0
[   75.530454] fbtft_of_value: fps = 10
[   75.534072] fbtft_of_value: txbuflen = 32768
[   75.539450] fb_st7565p st7565p: gpio_request_one('reset-gpios'=51) failed with -16
[   75.548016] fb_st7565p: probe of st7565p failed with error -16

Unfortunately system indicate me that reset gpio is taken by fb_nt7108 so can't be taken be new. 

Now question is that it is possible to drive two displays on the same lines on the same time?

notro, 2017-05-05:
Ah, I see, can the displays share the databus? Unfortunately not.
Linux doesn't have a high speed parallel bus subsystem.
I did a [hack](https://github.com/notro/fbdbi) a couple of years ago adding a i80 bus, but dropped it since adding a bus will take a looong time getting accepted and not many use fbtft with a parallel bus.

voloviq, 2017-05-08:
Thanks Notro
I think this issue can be closed.

closed 2017-05-09 by notro