notro/fbtft #467: ssd1963 and ft5205 device tree overlay

tiibro, 2017-06-16:
Hi Folks,

I am adding a 3.5" screen with a capacitive touch panel to a Compute Module 3.  You can read about my project here: [https://www.raspberrypi.org/forums/viewtopic.php?t=185948](https://www.raspberrypi.org/forums/viewtopic.php?t=185948).

I can get my screen working with the following commands:

`sudo modprobe --first-time flexfb width=320 height=240 regwidth=16 buswidth=16 init=-1,0x01,-2,10,-1,0xE2,0x1D,0x02,0x04,-1,0xE0,0x01,-2,5,-1,0xE0,0x03,-2,5,-1,0X01,-2,10,-1,0XB0,0X00,0X00,0X01,0X3F,0X00,0XEF,0X00,-1,0XF0,0X03,-1,0XE6,0X01,0X06,0X24,-1,0XB4,0X01,0X97,0X00,0X46,0X00,0X00,0X00,0X00,-1,0XB6,0X01,0X6,0X00,0X0D,0X00,0X00,0X00,-1,0X2A,0X00,0X00,0X01,0X3F,-1,0X2B,0X00,0X00,0X00,0XEF,0X00,0XEF,-1,0XB8,0X0F,0X01,0XBA,0X01,-1,0X29,-3`

and

`sudo modprobe --first-time fbtft_device name=flexpfb gpios=reset:23,dc:5,wr:6,db00:7,db01:8,db02:9,db03:10,db04:11,db05:12,db06:13,db07:14,db08:15,db09:16,db10:17,db11:18,db12:19,db13:20,db14:21,db15:22 rotate=0`

then setting fbcp running using 

`/usr/local/bin/fbcp`

I'd like to use a device tree overlay to set these parameters on boot.  Here is the dts file I've started:
```
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2810"; /* CM3 */

	fragment@0 {
		target = <&i2c1>;
		__overlay__ {
			status = "okay";
		};
	};
	
	fragment@1 {
		target = <&gpio>;
		__overlay__ {
			ssd1963_gpio_pins: ssd1963_gpio_pins {
				brcm,pins = <5 6 7 8 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23>;
				brcm,function = <1>; /* out */
			};

			ft5206_gpio_pins: ft5206_gpio_pins {
				brcm,pins = <>;
				brcm,function = <1>; /* out */
			};
		};
	};

	fragment@2 {
		target-path = "/soc";
		__overlay__ {
			ssd1963_gpio: ssd1963_gpio {
				compatible = "ssd1963-gpio";
				pinctrl-names = "default";
				pinctrl-0 = <&i80_gpio_pins>;
				#address-cells = <1>;
				#size-cells = <1>;

				width = <320>;
				height = <240>;
				rotate = <0>;
				fps = <20>;
				regwidth = <16>;
				buswidth = <16>;
				
				reset-gpios = = <&gpio 23 0>;
				dc-gpios = <&gpio 5 0>;
				wr-gpios = <&gpio 6 0>;
				data-gpios = <&gpio 7 0>, /* db0 */
				<&gpio 8 0>, /* db1 */
				<&gpio 9 0>, /* db2 */
				<&gpio 10 0>, /* db3 */
				<&gpio 11 0>, /* db4 */
				<&gpio 12 0>, /* db5 */
				<&gpio 13 0>, /* db6 */
				<&gpio 14 0>, /* db7 */
				<&gpio 15 0>, /* db8 */
				<&gpio 16 0>, /* db9 */
				<&gpio 17 0>, /* db10 */
				<&gpio 18 0>, /* db11 */
				<&gpio 19 0>, /* db12 */
				<&gpio 20 0>, /* db13 */
				<&gpio 21 0>, /* db14 */
				<&gpio 22 0>; /* db15 */

				init = <0x1000001 /* soft_reset */
				0x10000e2 0x1d 0x02 0x04 /* set_pll_mn */
				0x10000e0 0x01		 /* set_pll */
				0x10000e0 0x03		 /* set_pll */
				0x1000001		 /* soft_reset */
				0x10000b0 0x00 0x00 0x01 0x3f 0x00 0xef 0x00 /* set_lcd_mode */
				0x10000f0 0x03 /* set_pixel_data_interface */
				0x10000e6 0x01 0x06 0x24 /* set_lshift_freq */
				0x10000b4 0x01 0x97 0x00 0x46 0x00 0x00 0x00 0x00 /* set_hori_period */
				0x10000b6 0x01 0x06 0x00 0x0D 0x00 0x00 0x00 /* set_vert_period */
				0x100002a 0x00 0x00 0x01 0x3f /* set_column_address */
				0x100002b 0x00 0x00 0x00 0xef 0x00 0xef /* set_page_address */
				0x10000b8 0x0f 0x01 0xba 0x01 /* set_gpio_conf */
				0x1000029>; /* set_display_on */
			};
		};
	};

	fragment@3 {
                target = <&i2c1>;
                __overlay__ {
                        /* needed to avoid dtc warning */
                        #address-cells = <1>;
                        #size-cells = <0>;
			
                        ft5206: ft5206@38 {
                                compatible = "focaltech,ft5x06";
                                reg = <0x38>;

				/* options need to match driver */
                                /* interrupt-parent = <&gpio>; */
                                /* interrupts = <24 2>; */
                                /* touchscreen-size-x = <320>; */
                                /* touchscreen-size-y = <240>; */
                        };
                };
        };
};
```

I haven't tried to compile and load because I know there are issues.  My questions are:

1. For the TFT I am using the flexfb and fbtft_device drivers, how do I reference these from the dts?  I have looked at other examples, but I don't think I've found any parallel (flexpfb) examples.  I know that `compatible` properly links the software (driver?) to the hardware, but I am struggling (sorry).
2. I wonder if I've properly moved the arguements from the `modprobe` statements and added these to the dts file.
3. I've left space in the dts for the touchscreen control (via FT5206); currently I am struggling to compile a driver (or find a suitable pre-compiled driver).  Any advice would be most welcome.  I have asked here: [https://github.com/notro/rpi-source/issues/29](https://github.com/notro/rpi-source/issues/29) for help, but maybe that's not the most appropriate place.

Again, any help with my questions/problems listed above would be most welcome.  Thanks in advance for the help!

notro, 2017-06-16:
1. flexfb doesn't have DT support.
You can use any of the fbtft drivers that are MIPI compatible: set_addr_win() uses MIPI_DCS_SET_*, or 0x2a,2b,2c.
However the problem is that you get set_var() as well, which flexfb doesn't set. This means that register MIPI_DCS_SET_ADDRESS_MODE,0x36 is set as well, and the value might not suit your need. It controls flip/mirror/bgr.

2. Turn on debug output and you'll see if the driver has picked up the right values.

tiibro, 2017-06-18:
>  flexfb doesn't have DT support.

I am sorry.  I missed this in my research so I apologise for asking.

As dt support isn't available for the flexfb, is the next best option to add the terminal commands into the /etc/rc.local files?  Additionally, I use crontab to start other scripts and could add a shell script for the screen, but timing-wise this seems to execute about the same as the rc.local file.  Any advice on which is the "better" option?

Thanks.

notro, 2017-06-19:
If you use systemd (Raspian): https://github.com/notro/fbtft/wiki#make-it-permanent-debian-jessie-arch-linux-systemd

closed 2021-08-24 by github-actions[bot]