notro/fbtft #468: nanopi-m1 sun8i st7789s with kernel 4.11.2 it shows only white screen

mani009, 2017-06-23:
Hello,

i try to use the tft screen st7789s.

i compiled the kernel for the nano-pi-m1 with the driver st7789s:

```fb_st7789s.c

/*
 * FB driver for the st7789s LCD display controller
 *
 * This display uses 9-bit SPI: Data/Command bit + 8 data bits
 * For platforms that doesn't support 9-bit, the driver is capable
 * of emulating this using 8-bit transfer.
 * This is done by transferring eight 9-bit words in 9 bytes.
 *
 * Copyright (C) 2013 Christian Vogelgsang
 * Based on adafruit22fb.c by Noralf Tronnes
 *
 * This program is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 2 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program; if not, write to the Free Software
 * Foundation, Inc., 675 Mass Ave, Cambridge, MA 02139, USA.
 */

#include <linux/module.h>
#include <linux/kernel.h>
#include <linux/init.h>
#include <linux/delay.h>

#include "fbtft.h"

#define DRVNAME		"fb_st7789s"
#define WIDTH		240
#define HEIGHT		320
#define TXBUFLEN	(4 * PAGE_SIZE)
#define DEFAULT_GAMMA	"D0 03 0B 14 17 1D 3F 44 4E 0A 13 13 1E 20\n" \
			"D0 03 0B 12 14 1C 3F 44 4E 0B 17 15 1E 21"

static int init_display(struct fbtft_par *par)
{
	fbtft_par_dbg(DEBUG_INIT_DISPLAY, par, "%s()\n", __func__);

	par->fbtftops.reset(par);

	/* startup sequence for MI0283QT-9A */
	write_reg(par, 0x01); /* software reset */
	mdelay(5);
	write_reg(par, 0x28); /* display off */

	/* --------------------------------------------------------- */
	write_reg(par, 0xCF, 0x00, 0x83, 0x30);
	write_reg(par, 0xED, 0x64, 0x03, 0x12, 0x81);
	write_reg(par, 0xE8, 0x85, 0x01, 0x79);
	write_reg(par, 0xCB, 0x39, 0X2C, 0x00, 0x34, 0x02);
	write_reg(par, 0xF7, 0x20);
	write_reg(par, 0xEA, 0x00, 0x00);
	/* ------------power control-------------------------------- */
	write_reg(par, 0xC0, 0x26);
	write_reg(par, 0xC1, 0x11);
	/* ------------VCOM --------- */
[... 305 lines cut ...]
[  145.856711] fb_st7789s spi0.0: fbtft_write_spi(len=4): 00 00 00 ef
[  145.856946] fb_st7789s spi0.0: fbtft_write_reg8_bus8: 2b 00 00 01 3f
[  145.857008] fb_st7789s spi0.0: fbtft_write_spi(len=1): 2b
[  145.857230] fb_st7789s spi0.0: fbtft_write_spi(len=4): 00 00 01 3f
[  145.857503] fb_st7789s spi0.0: fbtft_write_reg8_bus8: 2c
[  145.857567] fb_st7789s spi0.0: fbtft_write_spi(len=1): 2c
[  145.857784] fb_st7789s spi0.0: fbtft_write_vmem16_bus8(offset=0, len=153600)
[  145.858234] fb_st7789s spi0.0: fbtft_write_spi(len=16384): 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...
[  145.875947] fb_st7789s spi0.0: fbtft_write_spi(len=16384): 63 bb 63 bb 63 bb 6b db 6b db 6b db 6b fc 73 fc 73 fc 73 fc 73 fc 74 1c 7c 1c 7c 1c 7c 3d 7c 3d ...
[  145.887913] fb_st7789s spi0.0: fbtft_write_spi(len=16384): ac fe ac fe b4 fe ac fe b4 fe b4 fe b4 fe b4 fe b4 fe b4 fd b4 fe b4 fd b4 fe b4 fd b4 dd b4 dd ...
[  145.904252] fb_st7789s spi0.0: fbtft_write_spi(len=16384): bc 7c bc 5c bc 5c c4 5b bc 5b bc 3b bc 3b bc 3b bc 3b bc 1a bb fa bb fa bb da b3 b9 b3 b9 b3 99 ...
[  145.916771] fb_st7789s spi0.0: fbtft_write_spi(len=16384): 92 d6 92 b5 92 b5 8a b5 8a b4 8a 94 8a 93 8a 73 8a 73 82 72 82 52 82 52 82 52 7a 31 7a 31 7a 31 ...
[  145.928731] fb_st7789s spi0.0: fbtft_write_spi(len=16384): 49 cd 49 cd 49 cc 49 cc 49 cc 41 ac 41 ac 41 ab 41 ab 41 ab 39 8b 39 aa 39 aa 39 8a 39 8a 39 8a ...
[  145.942792] fb_st7789s spi0.0: fbtft_write_spi(len=16384): 21 48 21 48 21 47 21 27 21 47 21 27 21 27 21 47 19 27 21 27 19 27 19 27 19 27 19 27 19 26 19 27 ...
[  145.958902] fb_st7789s spi0.0: fbtft_write_spi(len=16384): 10 e5 10 e5 10 e5 10 e5 10 e5 10 e5 10 e5 10 e5 10 e5 10 e4 10 e4 10 e4 10 c4 10 e4 10 c4 10 c4 ...
[  145.975133] fb_st7789s spi0.0: fbtft_write_spi(len=16384): 29 8a 29 aa 29 aa 29 ab 29 ab 29 ab 29 ab 29 ac 29 cc 29 cc 29 cc 29 cc 29 cd 29 ed 29 cd 29 ed ...
[  145.988204] fb_st7789s spi0.0: fbtft_write_spi(len=6144): b5 d9 ff df ff ff ff ff f7 bf a5 38 32 2f 31 ef 31 ef 31 ef 32 0f 32 10 3a 10 3a 10 32 10 3a 10 ...
[  145.994428] fb_st7789s spi0.0: Display update: 1108 kB/s, fps=1
[  147.246137] fb_st7789s spi0.0: fbtft_update_display(start_line=0, end_line=319)
[  147.246198] set_addr_win(xs=0, ys=0, xe=239, ye=319)
```

Can you give me a little advice why the display only show white screen ?

thanks a lot.

notro, 2017-06-23:
A white display means that it hasn't been initialized properly.
Usually 3 causes for this:
1. Wrong init sequence
2. Wrong wiring/gpio
3. Speed to high

In your case, try to lower the speed.

mani009, 2017-06-23:
Hello notro,

thanks for your time and reply.

I changed the module Settings to:
sudo modprobe fbtft_device name=st7789s debug=7 speed=100000 gpios=reset:203,dc:1,cs:67,led:0

but the result is the same for 1 sec black screen and then a white screen.

I believe the wiring must be okay,the pins are connect by pole plug as descripted by wiki.

[http://wiki.friendlyarm.com/wiki/index.php/Matrix_-_2%278_SPI_Key_TFT](url)

i find the pagesize int the macro from the driver is strange 4 * 4096.

notro, 2017-06-23:
> but the result is the same for 1 sec black screen and then a white screen.

By black do you mean black and backlit or just not backlit?

> i find the pagesize int the macro from the driver is strange 4 * 4096.

That's the size of the transfer buffer. You can change it with the txbuflen argument.
txbuflen=-2 will disable buffering if your spi controller supports 16 bits per word, but I haven't tried it since the Raspberry Pi doesn't support it.

mani009, 2017-06-24:
By black do you mean black and backlit or just not backlit?

I mean its backlit.

When i rmmod i have an error in the dmesg output:

```
[  163.687589] Unable to handle kernel paging request at virtual address f15ab000
[  163.694836] pgd = c0004000
[  163.697665] [f15ab000] *pgd=6e4cb811, *pte=00000000, *ppte=00000000
[  163.703992] Internal error: Oops: 7 [#1] SMP ARM
[  163.708622] Modules linked in: fb_st7789s(C) bluetooth rtl8192cu rtl_usb rtl8192c_common rtlwifi g_ether usb_f_rndis u_ether g_mass_storage usb_f_mass_storage libcomposite ipv6 [last unloaded: fbtft_device]
[  163.727091] CPU: 1 PID: 68 Comm: kworker/1:1 Tainted: G         C      4.11.2 #6
[  163.734494] Hardware name: sun8i
[  163.737770] Workqueue: events fb_deferred_io_work
[  163.742500] task: ef330000 task.stack: ef358000
[  163.747067] PC is at fbtft_write_vmem16_bus8+0xa8/0x140
[  163.752317] LR is at fbtft_write_vmem16_bus8+0x4c/0x140
[  163.757567] pc : [<c0599b80>]    lr : [<c0599b24>]    psr: 20030013
               sp : ef359eb0  ip : 00004000  fp : ef0be180
[  163.769057] r10: 00002000  r9 : eebefb30  r8 : ee6d0010
[  163.774299] r7 : 00000000  r6 : f15ab000  r5 : 00002000  r4 : 00012c00
[  163.780841] r3 : ee6d0010  r2 : f15ab000  r1 : ee6d000e  r0 : f15af000
[  163.787386] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
[  163.794536] Control: 10c5387d  Table: 6e4f406a  DAC: 00000051
[  163.800297] Process kworker/1:1 (pid: 68, stack limit = 0xef358210)
[  163.806579] Stack: (0xef359eb0 to 0xef35a000)
[  163.810964] 9ea0:                                     00025800 00000000 00025800 eebefb30
[  163.819171] 9ec0: 00000000 00000001 1c18fb3a 00000026 ef7c0340 c0596fb4 0000013f ef7c0740
[  163.827378] 9ee0: ef330000 c9d9f380 ee5a0fc0 ee5a0fc0 eebefb30 eebef800 ef359f4c ee4bf3d4
[  163.835585] 9f00: ee4bf3e8 ee4bf3d4 eebef800 ee4bf3d0 00000000 ef7c0340 ef0be180 c03fceac
[  163.843790] 9f20: ef0be180 eebefac8 ef7c3300 00000000 ef7c0340 c0131d64 ef7c0358 c0b02d00
[  163.851996] 9f40: ef7c0340 ef0be198 00000008 ef7c0358 c0b02d00 ef358000 ef7c0340 c0131fb0
[  163.860202] 9f60: 00000000 ef0bc740 ef0be180 ef169200 00000000 ef0bc740 ef0be180 c0131f78
[  163.868407] 9f80: ef16921c ef08dee8 00000000 c013751c ef0bc740 c01373f4 00000000 00000000
[  163.876610] 9fa0: 00000000 00000000 00000000 c0107678 00000000 00000000 00000000 00000000
[  163.884811] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  163.893015] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000 ffffffff ffffffff
[  163.901253] [<c0599b80>] (fbtft_write_vmem16_bus8) from [<c0596fb4>] (fbtft_update_display+0xcc/0x38c)
[  163.910601] [<c0596fb4>] (fbtft_update_display) from [<c03fceac>] (fb_deferred_io_work+0x8c/0xd4)
[  163.919520] [<c03fceac>] (fb_deferred_io_work) from [<c0131d64>] (process_one_work+0x138/0x34c)
[  163.928260] [<c0131d64>] (process_one_work) from [<c0131fb0>] (worker_thread+0x38/0x510)
[  163.936387] [<c0131fb0>] (worker_thread) from [<c013751c>] (kthread+0x128/0x158)
[  163.943823] [<c013751c>] (kthread) from [<c0107678>] (ret_from_fork+0x14/0x3c)
[  163.951077] Code: 11a02006 12481002 1086000c 0a000010 (e0d230b2) 
[  163.957366] ---[ end trace 8289539aefa00ad3 ]---
```

When i set the txbuflen flag i get this error:

```
[  350.166824] fb_st7789s spi0.0: SPI transfer failed: -110
[  350.172370] spi_master spi0: failed to transfer one message from queue
[  350.179080] fb_st7789s spi0.0: fbtft_update_display: write_vmem failed to update display buffer
[  350.187947] fb_st7789s spi0.0: Display update: 682 kB/s, fps=0
```

notro, 2017-06-24:
If the display is black and backlit, this means that it's initialized and displays the initial zeroed framebuffer.
This is strange, that the next display update should mess up the initialization. For this to happen, the transfer has to be garbled so the init registers gets overwritten.
Or maybe not, command zero is a no-op on this controller.

> When i rmmod i have an error in the dmesg output:

I guess someone is writing to the framebuffer as the module is removed. Indication that the fbtft module refcounting is wrong.

> When i set the txbuflen flag i get this error:

-110 is timeout, so the driver/controller can't handle a 150kb buffer.
Try setting txbuflen=64. This should force PIO in case the spi controller driver uses DMA and this is broken.

mani009, 2017-06-24:
If i have understood correctly the driver must be init not with 150Kib memory mode but with dma mode ??

When i set the txbuflen to 64 the driver shows:
```
[   39.727869] graphics fb1: fb_st7789s frame buffer, 240x320, 150 KiB video memory, 0 KiB buffer memory, fps=20, spi0.0 at 25 MHz
[   39.727939] fb_st7789s spi0.0: fbtft_backlight_update_status: polarity=1, power=0, fb_blank=0
```

notro, 2017-06-24:
> If i have understood correctly the driver must be init not with 150Kib memory mode but with dma mode ??

I don't understand what you say here, but the framebuffer is 150kB, so a full display update needs to transfer 150kB. fbtft by default uses a transfer buffer to split up the transfer. txbuflen is the size of this buffer and is by default 4096.

If the spi controller driver can do dma it does so. Usually on transfers below a certain threshold, it's faster to  just let the cpu fill the fifo instead of using the dma engine. On the Raspberry Pi this is 117 bytes.

Maybe the best approach is to bisect and find when it broke.
Try and build 3.4 yourself and see if it works, then 4.0 and so on...

mani009, 2017-07-03:
Hello,

the kernel 3.4 lichee from friendlyarm is modified.When i use this kernel. The driver output shows it has DMA mode:

[ 1963.869366] graphics fb8: fb_st7789s frame buffer, 240x320, 150 KiB video memory, 16 KiB DMA buffer memory, fps=30, spi0.2 at 25 MHz

Did you mean that ?

notro, 2017-07-12:
fbtft can't tell spi to use dma. It's the spi controller driver that makes that call.

If you can't get help in a forum somewhere to find out why it doesn't work, I suggest you try something like this ([bisecting](https://en.wikipedia.org/wiki/Bisection_(software_engineering))):
Linux 3.4 has been tested and work.
Next try the middle version between 4.12 and 3.4: roughly 4.0
If that works try 4.6, if it doesn't try 3.11.
Continue cutting the version nmuber in half until you find one version that works and the next that doesn't work.

Then look at the changes to the spi controller driver done between those versions to see if something sticks out. If it doesn't, you have to git bisect between those versions.

closed 2021-08-24 by github-actions[bot]