notro/fbtft #469: Raspberry Pi Zero with Waveshare 3.5 (A)

potato, 2017-06-30:
Hi,

I was wondering, is it possible to use the waveshare 3.5" (A) display with a pi zero?
I tried multiple things, including the dtoverlay from [swkim01/waveshare-dtoverlays
](https://github.com/swkim01/waveshare-dtoverlays/) (which worked flawlessly with the same display on a raspberry pi 2), and some methods from [this issue](https://github.com/notro/fbtft/issues/215). The result was always the same, blank white screen. Am I missing something, or it is just not possible to use the display with a pi zero?

Thanks,
potato

ravenousld3341, 2017-06-30:
You can use it on a PI Zero so long as you supply enough power. There's projects that use a 3.5in screen and a Pi Zero.

Some of these screens show up faulty. My original one did, the edge connector for the screen's board wasn't soldered properly. All i did was heat up the solder and watched it fall into place.

potato, 2017-06-30:
Thanks for the reply,
the screen is not faulty, I tried the exact same screen with my pi 2, and it worked. I used with the pi zero an 1.5A external power source, I think it should be enough for it.

notro, 2017-07-12:
I don't know of anything that should prevent it from working.
Have you checked that the driver is loaded and probed?
```
$ dmesg | grep graphics
[   39.727869] graphics fb1: fb_st7789s frame buffer, 240x320, 150 KiB video memory, 0 KiB buffer memory, fps=20, spi0.0 at 25 MHz

```

potato, 2017-07-12:
right now with the configuration from the last try (with waveshare-dtoverlays) I got this:
```
[   11.269036] graphics fb1: fb_ili9486 frame buffer, 480x320, 300 KiB video memory, 32 KiB DMA buffer memory, fps=33, spi0.0 at 16 MHz
```

the relevant lines from my config.txt:
```
dtparam=spi=on
dtparam=i2c_arm=on
dtoverlay=waveshare35a
```

and in my cmdline.txt I have this right after `rootwait`:
```
fbcon=map:1 fb con=font:ProFont6x11 logo.nologo
```

saper-2, 2017-08-05:
Hi,
Let me add my 3 cents 😄  :
1. "fb con" - there should not be a space..... Remove all con remap from cmdline, the LCD when initialized should turn black (even if no console is mapped to it).
2. Use full raspbian system image (not noobs or other crippled version)
3. Measure voltages on GPIO connector: 5V and 3,3V should be within 5% error margin: 5V +/-5% = 4,75...5.25V , 3,3V +/- ~5% = 3.15...3,40V.

I agree with **ravenousld3341** - I think the problem lies with Pi0 3,3V bus which can supply ca 50mA...  While display need a bit more and white leds draw a one hell of current too (and they are usually connected directly to 3,3V on those displays) - So Pi2 can handle this much current being drawn from it witout issue, but Pi0 have a much smaller DC/DC converter for 3,3V ....

potato, 2017-08-18:
1. it was just a paste typo :)
2. using the full raspbian image
3. voltages are fine, within the error margin

I see now the point behind the current-problem. I'll try to power the display from a separate supply, maybe that will solve the problem.

I'll keep this issue updated after the experiment.

saper-2, 2017-08-18:
I know this is a bit stretched, but check soldering of pinheader on your Pi0 - maybe there is a bad solder on the MOSI/SCK line (or too much solder flux residues - those can be a real PITA)... 

If voltages are within error (especially 3,3V), then current drawn from Pi0 shouldn't be an issue - the larger current being drawn from dc/dc converter the more voltage will drop. 
*Personally, I'd see a problem if 3,3V line would drop to less than 3,25V - but that's me, I'm just a really pedantic 😄 about power lines (and their stability) in my circuits (I put on my PCBs a lot of decoupling capacitors around power pins of chips 😁  - so I actually never had any real problems with power in my designs).*

Other than those, I don't have other ideas... 
This is original waveshare display or clone (and where u get it) maybe I'll order one myself and test it 😃

potato, 2017-09-19:
it's an original rev4.0 waveshare display, and actually it was a soldering issue, more precisely the lack of my soldering skills :D after rechecking (and in some cases resoldering) the pins everything works fine.
thanks for all the help!

closed 2017-09-19 by potato