notro/fbtft #473: ILI9488 kind of works

MohammadMubin, 2017-07-16:
hi guys i tried to attach a 3.5" mcu friend tft display which works great on ardunio but not so great on the pi

sudo modprobe fbtft_device custom name=flexpfb speed=48000000 fps=60 gpios=reset:20,dc:1,wr:3,cs:10,db00:0,db01:14,db02:2,db03:21,db04:18,db05:19,db06:8,db07:9
 
sudo modprobe flexfb width=320 height=480 buswidth=8 init=-1,0x11,-2,200,-1,0x36,-1,0x3A,0x05,-1,0x36,0x48,-1,0x13,-1,0x29,-1,0x2C,-3

i edited the X11 config files to match it with the frame buffer and wrote startx 
i also tried displaying images but i get the same thing 
[image]

Any help would be appreciated

notro, 2017-07-28:
I would try lowering the speed

MohammadMubin, 2017-07-30:
Thanks For your Response,but how low? like an exact value would be good

notro, 2017-07-30:
Most controllers are rated at max 10MHz, but many can do 32MHz. It depends a lot on the wiring.

flafyrex, 2017-08-06:
Don't worry! I am working for support on the ILI9488! I am now fighting with compiling the kernel! When I will be done with the driver, I will post a fork(is this what you are calling it?) of the FBTFT Driver with ILI9488 support.

flafyrex, 2017-08-06:
And yup. That looks that the speed is too fast.
Lower it down or remove it completely!

flafyrex, 2017-08-06:
And the screen is the same(mine is bigger - 3.95 TFT Display). Mine works, but I need to compile the kernel with GPIO Optimizations.

MohammadMubin, 2017-08-06:
I haven't tried lowering the speed because I am on vacation and forgot to bring a power cable  (:P) and I am not on a raspberry pi but I am on a orange pi (Da DUM)

MohammadMubin, 2017-08-07:
Ok I tested and I saw no difference it was the exactly the same as the picture above (after running startx as fb8)

MohammadMubin, 2017-08-07:
@flafyrex 

Still the same no difference  but SPI ili9341 works completely fine but this is parallel display which could be the problem

talnacialex, 2017-09-16:
@notro I am now working on a patch for the ILI9488 MCUFriend driver. Would you accept it? It has overlays that work(unfortunately, there is no touch support). And I am using the Arduino UNO version hooked up to the raspberry pi with lots of wires. Works fast without GPIOLIB. Works better. Nice job!

notro, 2017-09-17:
I'm no longer a fbtft maintainer. Thomas Petazzoni is listed as maintainer, but he really does nothing.
Greg Kroah-Hartman is the staging maintainer, but I don't know how fbtft is handled these days.
https://github.com/notro/fbtft/wiki/Development

talnacialex, 2017-09-17:
Hi, Notro! Tell me if you can get me in contact with them, because I want to give them a Patch File that contains the FBTFT Driver for Arduino UNO Mcufriend.com 3.95 LCD using ILI9488 and a overlay that works. I am using this display on the raspberry pi. This uses part of the pinout of the HY-???(i forgot the name). I am using with recompiled kernel and without GPIOLIB and works very nice. Also, Good job of this driver!
Thank you.

With regards:
Talnaci Alexandru

talnacialex, 2017-09-17:
And this is how the LCD looks like: 
[image] front
[image] back

notro, 2017-09-17:
You should find all the info you need on the wiki page I gave.

MohammadMubin, 2017-09-18:
@talnacialex  Now try to get working  on a Orange Pi since its cheaper :P but i will try on the Raspberry Pi too once i buy one

talnacialex, 2017-09-22:
I don't have a Orange Pi, nor a Banana Pi. I only have this Raspberry Pi and This Screen. And, @notro, I followed the wiki. This is how I made it working.

talnacialex, 2017-12-06:
Ok everybody! Here I attach the patch file. I can't test it on the Raspberry Pi, because right now it is used in a personal project. So.... if there are people that want it working(i think so) with rotation and everything, use my patch. For beginners, run `patch -p1 < ili9488.patch` from the root of the kernel sources. It should automatically patch the files that make this LCD work with the kernel! Sorry that I didn't gave the patch fast enough... I lost the VM I compiled the kernel. I remade the patch from scratch. Hope it works :) 
[ili9488.patch in a zip file(ili9488.zip)](https://github.com/notro/fbtft/files/1536119/ili9488.zip)

bogdanalin92, 2018-02-20:
i can't install the patch it says that it cannot find a file to patch. Can you explain step by step what can i do?

talnacialex, 2018-02-20:
I said that I did not had time to recompile the kernel again with the patch. I made a patch file out of my hand. You can read the patch file to see what file should be modified. Now I am busy with other personal projects and I can't start working on the FBTFT kernel right away. I will try in my future time(if possible) to recompile it.

closed 2021-08-24 by github-actions[bot]