notro/fbtft #474: RA8875  ER-TFTM070-5 Rpi 3  with newest rasbian

sonic2wb, 2017-07-28:
So I wired up the display but theres no RS (aka d/c) pin. unless i am missing something. 
When i modprobe the driver the screen lights up and u can see the frame buffer fill the screen with black lines. dmesg so the display is detected.  then i run the startx command and the desktop appears on the display but only the right half of the display works. I can see the mouse and can click on the wifi icon and date and time. u can even kinda see the start menu trying to be drawn behind this White block.  

if u run it in console that white box is now black and the console text scrolls up behind it.  its driving me crazy!! I thought at first the lcd was bad but I hooked up my arduino mega and all the test apps run just fine. and the display shows just fine.

any ideas? 
[image]

notro, 2017-07-28:
I don't know how you power the display, but if you power it from the Pi you will probably run into strange issues due to undervoltage.
I found this: https://github.com/sumotoy/RA8875/wiki/EastRising-and-Buydisplay-SPI-configuration-and-wiring

sonic2wb, 2017-07-29:
Thanks for the reply. i am still getting that white block. in front of the left side of the screen. it seems it is something in my display i even did the framebuffer copy from the wiki and it works except for that dang white box here is a better picture showing the mouse pointer .  only other thing i know to do is try the fix in issue 96 if i can figure out how to compile drivers on the pi......
[image]

sonic2wb, 2017-07-29:
i ment issue 176

sonic2wb, 2017-07-29:
built a new driver with the fix from issue 176 i still have the box on the left side. it seems to be something in the int of the display.   If i hook display to my arduino all the code and all of the display works just fine. I gave screen its own power source etc..  Guess ill just have shelve the display for now..

sonic2wb, 2017-07-29:
The only other thing i can think of is this display doesnt have a D/C pin? unless i am just missing it. i have the reset pin on gpio 25.    oh and another note is the screen saver works on the frambuffer the screen goes all black.

sonic2wb, 2017-07-29:
It works!  the patch in #176 did the trick. took a bit of time and research but i was able to compile a new ra_8875.ko  and install it into my rasbian install. also did some testing my display works great at 24khtz 

[image]

sonic2wb, 2017-07-29:
I will post a link to instructions and to the modified fb_ra8875.c file later today. I hope its possible to add the patch to the kernel source so on the next kerel update i wont have to rebuild the module.

closed 2017-07-29 by sonic2wb

notro, 2017-07-30:
I'm no longer a maintainer of fbtft.
https://github.com/notro/fbtft/wiki/Development