notro/fbtft #475: ILI9225 won't display screen.

Holydohnut, 2017-08-02:
Hi all. I got sent an ILI9225 in the post by mistake, but I'm trying to get it working anyway. A very helpful chap on the sudomod forums advised me how to wire it up, but the driver just doesn't seem to be working. If I specify the ILI9225 in the modprobe line, the back light doesn't come on. If I specify the hx8430bn which I believe has the same resolution, the back light comes on, but running the con2fbmap line does nothing. Does anyone have any ideas?

notro, 2017-08-21:
You have to use fb_ili9320 or fb_ili9325 (they differ in rotation and init) to get the right spi protocol and register setting for display updates.
Use fbtft_device or Device Tree to match it to your needs:
```
modprobe fbtft_device custom name=fb_ili9325 speed=16000000 buswidth=16 gpios=reset:xx init=-1,reg,val,-1,reg,val,....,-3
```
https://github.com/notro/fbtft/wiki/fbtft_device
http://elixir.free-electrons.com/linux/latest/source/drivers/staging/fbtft/fbtft_device.c

Holydohnut, 2017-08-21:
Hi Notro. Thanks for replying. Unfortunately neither of those seem to work. I think because they are both 320x240 and the ILI9225 is only 176x220?

notro, 2017-08-21:
Use ```modprobe fbtft_device custom ... width=176 height=220```

raomin, 2018-03-05:
Hi!
I also have an ILI9225, and I can't get it to initialise properly on a raspberry zero with fbtft_device.
It works fine on the same board with https://github.com/nopnop2002/ili9225spi_rpi
So, I kept the same wiring and ported the init from ili9225.c with fb_ili9325.
But I only get a white screen...

`modprobe fbtft_device custom name=fb_ili9325 speed=16000000 buswidth=16 gpios=reset:3,dc:2,cs:8 speed=32000000 init=-1,0x01,0x031C,-1,0x02,0x0100,-1,0x03,0x1030,-1,0x08,0x0808,-1,0x0C,0x0000,-1,0x0F,0x0801,-1,0x20,0x0000,-1,0x21,0x0000,-2,50,-1,0x10,0x0A00,-1,0x11,0x1038,-2,50,-1,0x12,0x1121,-1,0x13,0x0066,-1,0x14,0x5F60,-1,0x30,0x0000,-1,0x31,0x00DB,-1,0x32,0x0000,-1,0x33,0x0000,-1,0x34,0x00DB,-1,0x35,0x0000,-1,0x36,0x00AF,-1,0x37,0x0000,-1,0x38,0x00DB,-1,0x39,0x0000,-1,0x50,0x0400,-1,0x51,0x060B,-1,0x52,0x0C0A,-1,0x53,0x0105,-1,0x54,0x0A0C,-1,0x55,0x0B06,-1,0x56,0x0004,-1,0x57,0x0501,-1,0x58,0x0E00,-1,0x59,0x000E,-2,50,-1,0x07,0x1017,-3 width=176 height=220`

Here is what I get in dmesg: 
```
[   71.664212] fbtft: module is from the staging directory, the quality is unknown, you have been warned.
[   71.670744] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[   71.672534] spidev spi0.0: spidev spi0.0 125000kHz 8 bits mode=0x00
[   71.672559] spidev spi0.1: spidev spi0.1 125000kHz 8 bits mode=0x00
[   71.672627] bcm2708_fb soc:fb: soc:fb id=-1 pdata? no
[   71.672683] spidev spi0.0: Deleting spi0.0
[   71.686748] fbtft_device: GPIOS used by 'fb_ili9325':
[   71.686765] fbtft_device: 'reset' = GPIO3
[   71.686771] fbtft_device: 'dc' = GPIO2
[   71.686777] fbtft_device: 'cs' = GPIO8
[   71.686795] spidev spi0.1: spidev spi0.1 125000kHz 8 bits mode=0x00
[   71.686805] spi spi0.0: fb_ili9325 spi0.0 32000kHz 8 bits mode=0x00
[   71.811048] fb_ili9325: module is from the staging directory, the quality is unknown, you have been warned.
[   72.105813] fb_ili9325 spi0.0: transfer size of 77440 too big for dma-transfer
[   72.133973] graphics fb1: fb_ili9325 frame buffer, 176x220, 75 KiB video memory, 4 KiB buffer memory, fps=20, spi0.0 at 32 MHz

```
I'm not sure about the speed or the buswidth, nor where to specify the SDA port (MOSI GPIO10)

What did I miss?

@Holydohnut Did you manage to get it working?

raomin, 2018-03-05:
I managed to get a black screen with buswidth=8 !
`modprobe fbtft_device custom name=fb_ili9325 speed=16000000 buswidth=8 gpios=reset:3,dc:2,cs:8 speed=32000000 init=-1,0x01,0x031C,-1,0x02,0x0100,-1,0x03,0x1030,-1,0x08,0x0808,-1,0x0C,0x0000,-1,0x0F,0x0801,-1,0x20,0x0000,-1,0x21,0x0000,-2,50,-1,0x10,0x0A00,-1,0x11,0x1038,-2,50,-1,0x12,0x1121,-1,0x13,0x0066,-1,0x14,0x5F60,-1,0x30,0x0000,-1,0x31,0x00DB,-1,0x32,0x0000,-1,0x33,0x0000,-1,0x34,0x00DB,-1,0x35,0x0000,-1,0x36,0x00AF,-1,0x37,0x0000,-1,0x38,0x00DB,-1,0x39,0x0000,-1,0x50,0x0400,-1,0x51,0x060B,-1,0x52,0x0C0A,-1,0x53,0x0105,-1,0x54,0x0A0C,-1,0x55,0x0B06,-1,0x56,0x0004,-1,0x57,0x0501,-1,0x58,0x0E00,-1,0x59,0x000E,-2,50,-1,0x07,0x1017,-3 width=176 height=220`

But nothing when I try to `con2fbmap 1 1` or to ` fbi -d /dev/fb1 -T 1 -noverbose -a Mystery-100x100.jpg`

raomin, 2018-03-06:
Actually, I don't know if it's normal but a few lines remains white after loading the module:
[image]

raomin, 2018-03-08:
@notro Hi, any hope you could help on this one?
I ended up using a ili9341, which works fine on the same setup, but I really hope I could get the ili9225 working.

koosthoe, 2018-04-27:
any luck getting the ili9225 to work with Rpi Zero? im having issues and support online seems slim to none

mstroh76, 2018-07-15:
If you first start demo of https://github.com/nopnop2002/ili9225spi_rpi  and afterwards load the Module fbtft_device. It's partial working. 
Problems: Colors wrong, picture is moving,

CRImier, 2019-12-03:
[Made it work](https://crimier.wordpress.com/2019/12/03/ili9225-and-fbtft/), using flexfb => a proper driver will be needed once the flexfb removal changes propagate into Raspbian and the like. It's a shame flexfb will no longer be there, it's a godsend for unsupported displays.

closed 2021-08-24 by github-actions[bot]