notro/fbtft #477: Support register params with DC=0 (e.g. SSD1306, SSD1225)

StefanBruens, 2017-08-21:
Unfortunately SSD devices use two different concepts for register params:

e.g SSD1351:
- register number has DC=0, all N register params have DC=1

e.g. SSD1306, SSD1225:
- registers and all following register params have DC=0

Currently fbtft_write_reg8_bus9 and fbtft_write_reg8_bus8 assume register params use DC=1.

notro, 2017-08-28:
The work around for that is to pass regnr and value separately:
```
	/* Set Display Clock Divide Ratio/ Oscillator Frequency */
	write_reg(par, 0xD5);
	write_reg(par, 0x80);
```

fbtft_device:
```
modprobe fbtft_device ... init=-1,0xd5,-1,0x80,...
```

StefanBruens, 2017-08-28:
Unfortunately the multiple write_reg fails in combination with 9 bit emulation. Each write_reg results in an independent 5 byte transfer (should be 9, but you corrected that in tinydrm), and the padding is interpreted as value.

The init sequence might actually work, will evaluate this.

Would you accept a patch to tinydrm with an additional "quirk" transfer mode, e.g mipi_dbi_typec1_command_no_dc ?

notro, 2017-08-30:
> Would you accept a patch to tinydrm with an additional "quirk" transfer mode, e.g mipi_dbi_typec1_command_no_dc ?

No, because then it's not MIPI DBI anymore.
Better add a new driver for the controller in question with the necessary protocol functions.

closed 2021-08-24 by github-actions[bot]