notro/fbtft #478: Correct padding for 9 bit emulation

StefanBruens, 2017-08-21:
fbtft_write_spi_emulate_9 pads write up to a multiple of 4 words, although the lowest common multiple of 8 (buswidth) and 9 (device wordsize) is 72 bits, i.e. 8 device words.

Also, the padding/NOP commands may be different from 0x00, e.g.:
SSD1325: 0x00 undefined, NOP = 0xE3
SSD1306: 0x00 - Lower column start address=0, NOP = 0xE3
SSD1305: same as SSD1306
SSD1351: 0x00 undefined, NOP = 0xE3 or NOP = 0xD1

For SSD1306/1305, the 0x00 command is unproblematic as long as it followed by the correct address command, but this is not given in general.

notro, 2017-08-28:
Have you seen displays with those controllers that use 9-bit?

The 9-bit emulation was written for MIPI DBI compatible controllers.
fbtft isn't the best it could be because I had only 2 displays to start with, which I built a library around. And this library has turned out to be quite inflexible.

tinydrm is the fbtft successor: https://github.com/notro/tinydrm/wiki

StefanBruens, 2017-08-28:
The padding is correct in the new in-kernel tinydrm code, thanks for upstreaming this.

The 1306 is quite common for small 128x64 monochrome OLED displays, the SSD1351 starts to appear in 128x128 full color OLEDs.

For the 0xE3 NOP maybe an display dependent alternative to MIPI_DCS_NOP would do.

notro, 2017-08-30:
I mean, have you seen displays where you can actually use the 9-bit mode?
AFAIK all these Adafruit displays use D/C with SPI.

StefanBruens, 2017-08-31:
I have one of these: http://xdec.de/msp430-oled-display-ssd1306-128x64/
which I have set to 3-Wire mode

There is also this one: http://www.dx.com/p/waveshare-0-96-oled-b-ssd1306-display-screen-module-deep-blue-340467
and: http://www.hotmcu.com/ssd1306-096-128%C3%9764-oled-display-%E2%80%93-i2cspi-interface-p-144.html

For the SSD1351, there is http://www.newhavendisplay.com/nhd15128128ugc3-p-5602.html, which has BS0/1 (interface select) on Pins 19/20.

notro, 2017-09-02:
Any particular reason why you choose spi over i2c for the ssd1306? 
i2c is even less wires and there's a fbdev version: [drivers/video/fbdev/ssd1307fb.c](http://elixir.free-electrons.com/linux/latest/source/drivers/video/fbdev/ssd1307fb.c)

StefanBruens, 2017-09-06:
I use i2c for reading out sensors, and any transfer >= 50 bytes is killing bus access latency in my case. As far as I can see, the i2c fbdev driver always does full frame updates, i.e. 1kByte, thus the bus would be busy for 20/80 ms (@400kBit/s resp 100kBit/s).

closed 2021-08-24 by github-actions[bot]