notro/fbtft #479: ILI9341 LCD

mopplayer, 2017-08-30:


notro, 2017-08-30:
I haven't looked at BBB for a very long time, but you seem to have problems with the pin muxing, so maybe the spi signals doesn't reach the pins?

> [ 43.031979] pinctrl-single 44e10800.pinmux: bad data for mux lcd_ctrl_pins
> [ 43.039103] pinctrl-single 44e10800.pinmux: no pins entries for lcd_ctrl_pins

mopplayer, 2017-08-30:
@notro Hi
How could I to check the signals?
I already replaced wires with short length.

mopplayer, 2017-08-30:
static int pcs_parse_one_pinctrl_entry(struct pcs_device *pcs,
						struct device_node *np,
						struct pinctrl_map **map,
						unsigned *num_maps,
						const char **pgnames)
{
	struct pcs_func_vals *vals;
	const __be32 *mux;
	int size, rows, *pins, index = 0, found = 0, res = -ENOMEM;
	struct pcs_function *function;

	mux = of_get_property(np, PCS_MUX_PINS_NAME, &size);
	if ((!mux) || (size < sizeof(*mux) * 2)) {
		dev_err(pcs->dev, "bad data for mux %s\n",
			np->name);
		return -EINVAL;
	}

	size /= sizeof(*mux);	/* Number of elements in array */
	rows = size / 2;

	vals = devm_kzalloc(pcs->dev, sizeof(*vals) * rows, GFP_KERNEL);
	if (!vals)
		return -ENOMEM;

	pins = devm_kzalloc(pcs->dev, sizeof(*pins) * rows, GFP_KERNEL);
	if (!pins)
		goto free_vals;

	while (index < size) {
		unsigned offset, val;
		int pin;

		offset = be32_to_cpup(mux + index++);
		val = be32_to_cpup(mux + index++);
		vals[found].reg = pcs->base + offset;
		vals[found].val = val;

		pin = pcs_get_pin_by_offset(pcs, offset);
		if (pin < 0) {
			dev_err(pcs->dev,
				"could not add functions for %s %ux\n",
				np->name, offset);
			break;
		}
		pins[found++] = pin;
	}

	pgnames[0] = np->name;
	function = pcs_add_function(pcs, np, np->name, vals, found, pgnames, 1);
	if (!function)
		goto free_pins;

	res = pcs_add_pingroup(pcs, np, np->name, pins, found);
	if (res < 0)
		goto free_function;

	(*map)->type = PIN_MAP_TYPE_MUX_GROUP;
	(*map)->data.mux.group = np->name;
	(*map)->data.mux.function = np->name;

	if (PCS_HAS_PINCONF) {
		res = pcs_parse_pinconf(pcs, np, function, map);
		if (res)
			goto free_pingroups;
		*num_maps = 2;
	} else {
		*num_maps = 1;
	}
	return 0;

free_pingroups:
	pcs_free_pingroups(pcs);
	*num_maps = 1;
free_function:
	pcs_remove_function(pcs, function);

free_pins:
	devm_kfree(pcs->dev, pins);

free_vals:
	devm_kfree(pcs->dev, vals);

	return res;
}
@notro

mopplayer, 2017-08-30:
@notro 
root@beaglebone:/sys/kernel/debug/pinctrl/44e10800.pinmux# cat pingroups 
registered pin groups:
group: pinmux_uart0_pins
pin 92 (44e10970.0)
pin 93 (44e10974.0)

group: cpsw_default
pin 66 (44e10908.0)
pin 67 (44e1090c.0)
pin 68 (44e10910.0)
pin 69 (44e10914.0)
pin 70 (44e10918.0)
pin 71 (44e1091c.0)
pin 72 (44e10920.0)
pin 73 (44e10924.0)
pin 74 (44e10928.0)
pin 75 (44e1092c.0)
pin 76 (44e10930.0)
pin 77 (44e10934.0)
pin 78 (44e10938.0)
pin 79 (44e1093c.0)
pin 80 (44e10940.0)

group: cpsw_sleep
pin 66 (44e10908.0)
pin 67 (44e1090c.0)
pin 68 (44e10910.0)
pin 69 (44e10914.0)
pin 70 (44e10918.0)
pin 71 (44e1091c.0)
pin 72 (44e10920.0)
pin 73 (44e10924.0)
pin 74 (44e10928.0)
pin 75 (44e1092c.0)
pin 76 (44e10930.0)
pin 77 (44e10934.0)
pin 78 (44e10938.0)
pin 79 (44e1093c.0)
pin 80 (44e10940.0)

group: davinci_mdio_default
pin 82 (44e10948.0)
pin 83 (44e1094c.0)

group: davinci_mdio_sleep
pin 82 (44e10948.0)
pin 83 (44e1094c.0)

group: pinmux_mmc1_pins
pin 88 (44e10960.0)

group: pinmux_emmc_pins
pin 32 (44e10880.0)
pin 33 (44e10884.0)
pin 0 (44e10800.0)
pin 1 (44e10804.0)
pin 2 (44e10808.0)
pin 3 (44e1080c.0)
pin 4 (44e10810.0)
pin 5 (44e10814.0)
pin 6 (44e10818.0)
pin 7 (44e1081c.0)

group: user_leds_default
pin 21 (44e10854.0)
pin 22 (44e10858.0)
pin 23 (44e1085c.0)
pin 24 (44e10860.0)

group: user_leds_sleep
pin 21 (44e10854.0)
pin 22 (44e10858.0)
pin 23 (44e1085c.0)
pin 24 (44e10860.0)

group: pinmux_i2c0_pins
pin 98 (44e10988.0)
pin 99 (44e1098c.0)

group: pinmux_i2c2_pins
pin 94 (44e10978.0)
pin 95 (44e1097c.0)

group: pinmux_bb_uart1_pins
pin 97 (44e10984.0)
pin 96 (44e10980.0)

group: pinmux_pwm_bl_pins
pin 18 (44e10848.0)

group: pinmux_pwm_bl_pins
pin 18 (44e10848.0)

group: pinmux_pwm_bl_pins
pin 18 (44e10848.0)

group: pinmux_pwm_bl_pins
pin 18 (44e10848.0)

group: pinmux_pwm_bl_pins
pin 18 (44e10848.0)

group: pinmux_pwm_bl_pins
pin 18 (44e10848.0)

group: pinmux_pwm_bl_pins
pin 18 (44e10848.0)

group: pinmux_pwm_bl_pins
pin 18 (44e10848.0)

group: nxp_hdmi_bonelt_pins
pin 108 (44e109b0.0)
pin 40 (44e108a0.0)
pin 41 (44e108a4.0)
pin 42 (44e108a8.0)
pin 43 (44e108ac.0)
pin 44 (44e108b0.0)
pin 45 (44e108b4.0)
pin 46 (44e108b8.0)
pin 47 (44e108bc.0)
pin 48 (44e108c0.0)
pin 49 (44e108c4.0)
pin 50 (44e108c8.0)
pin 51 (44e108cc.0)
pin 52 (44e108d0.0)
pin 53 (44e108d4.0)
pin 54 (44e108d8.0)
pin 55 (44e108dc.0)
pin 56 (44e108e0.0)
pin 57 (44e108e4.0)
pin 58 (44e108e8.0)
pin 59 (44e108ec.0)

group: nxp_hdmi_bonelt_off_pins
pin 108 (44e109b0.0)

group: spi0_pins_s0
pin 84 (44e10950.0)
pin 85 (44e10954.0)
pin 86 (44e10958.0)
pin 87 (44e1095c.0)

group: mcasp0_pins
pin 107 (44e109ac.0)
pin 103 (44e1099c.0)
pin 101 (44e10994.0)
pin 100 (44e10990.0)
pin 27 (44e1086c.0)

mopplayer, 2017-08-30:
Hi, @notro 
It should be modified for Kernel 4.4.x
`			reset-gpios = <&gpio2 16 0>;
			dc-gpios = <&gpio2 28 0>;`
to
`			reset-gpios = <&gpio1 16 0>;
			dc-gpios = <&gpio1 28 0>;`
And it cloud display now, but errors still showed in dmesg
`[ 43.031979] pinctrl-single 44e10800.pinmux: bad data for mux lcd_ctrl_pins
[ 43.039103] pinctrl-single 44e10800.pinmux: no pins entries for lcd_ctrl_pins`

closed 2021-08-24 by github-actions[bot]