notro/fbtft #480: adafruit22a issues

twentytwoo, 2017-09-15:
Running `sudo modprobe fbtft_device name=adafruit22a rotate=270`

Causes the display to turn black momentarily and then white again - the display seems to work erratically, some days working fine, and others nothing at all...

I can control backlight brightness, I've added `fbcon=map:10` to `/boot/cmdline.txt` also.

dmesg:
```
[  108.868547] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[  108.874440] spidev spi0.0: spidev spi0.0 125000kHz 8 bits mode=0x00
[  108.874459] spidev spi0.1: spidev spi0.1 125000kHz 8 bits mode=0x00
[  108.874499] bcm2708_fb soc:fb: soc:fb id=-1 pdata? no
[  108.874532] spidev spi0.0: Deleting spi0.0
[  108.875235] fbtft_device: GPIOS used by 'adafruit22a':
[  108.875244] fbtft_device: 'reset' = GPIO25
[  108.875248] fbtft_device: 'dc' = GPIO24
[  108.875251] fbtft_device: 'led' = GPIO18
[  108.875264] spidev spi0.1: spidev spi0.1 125000kHz 8 bits mode=0x00
[  108.875273] spi spi0.0: fb_ili9340 spi0.0 32000kHz 8 bits mode=0x00
[  108.908561] fb_ili9340: module is from the staging directory, the quality is unknown, you have been warned.
[  108.910742] fb_ili9340 spi0.0: fbtft_request_gpios: 'reset' = GPIO25
[  108.910767] fb_ili9340 spi0.0: fbtft_request_gpios: 'dc' = GPIO24
[  108.910786] fb_ili9340 spi0.0: fbtft_request_gpios: 'led' = GPIO18
[  109.288865] graphics fb1: fb_ili9340 frame buffer, 240x320, 150 KiB video memory, 4 KiB DMA buffer memory, fps=20, spi0.0 at 32 MHz
```

lsmod:
```
Module                  Size  Used by
fb_ili9340              3722  2 
fbtft_device           37141  0 
fbtft                  32647  2 fbtft_device,fb_ili9340
syscopyarea             3161  1 fbtft
sysfillrect             3716  1 fbtft
sysimgblt               2422  1 fbtft
fb_sys_fops             1677  1 fbtft
spidev                  7034  0 
brcmfmac              267114  0 
brcmutil                9911  1 brcmfmac
spi_bcm2835             7424  0 
bcm2835_gpiomem         3791  0 
uio_pdrv_genirq         3718  0 
fixed                   3029  0 
uio                    10166  1 uio_pdrv_genirq
snd_bcm2835            23131  0 
snd_pcm                97825  1 snd_bcm2835
snd_timer              22706  1 snd_pcm
snd                    68784  3 snd_timer,snd_bcm2835,snd_pcm
mac80211              650707  0 
cfg80211              525998  2 mac80211,brcmfmac
rfkill                 21373  2 cfg80211
ipv6                  384391  26 
```

Any advice is greatly appreciated...

twentytwoo, 2017-09-15:
Probably related to this issue: https://github.com/adafruit/Adafruit_Python_ILI9341/issues/1
>when I run the examples the screen turns black for a second and then goes back to white.

notro, 2017-09-17:
The first thing to try with problems like this is to lower the speed, which you have tried. The next is checking wiring. The ultimate test is to look at the signals with a scope or logic analyzer.

twentytwoo, 2017-09-18:
Looked at the signals and noticed MOSI looked a bit weak, checked the pin soldering and saw it was pretty poor, re-soldered and it works reliably now. Also explains why sometimes it'd work, I guess the pins just touched the contacts more... My bad.

closed 2017-09-18 by twentytwoo