notro/fbtft #482: Compatibility

DavideLuongo91, 2017-10-17:
Hi! I want to have some informations:

- can I use fbtft on Raspberry pi Zero W?
- do you know mcufriend 3.0" is it compatible?
- can I use fbtft with a 7" SSD1963?

Thank you really much

notro, 2017-10-21:
> can I use fbtft on Raspberry pi Zero W?

fbtft works with all Raspberry Pi's. 

> do you know mcufriend 3.0" is it compatible?

Don't know

> can I use fbtft with a 7" SSD1963?

Yes, but it will be very slow and you might experience tearing.
https://github.com/notro/fbtft/issues/58
https://github.com/notro/fbtft/issues/138
https://github.com/notro/fbtft/issues/467

A better solution would be to find a DPI display: https://www.raspberrypi.org/documentation/hardware/raspberrypi/dpi/README.md

DavideLuongo91, 2017-10-23:
But Why if I use last release of my RPi Zero it doesn't works? What version of Raspbian should I use? Thank you

andykelk, 2017-12-06:
> do you know mcufriend 3.0" is it compatible?

I have one of these. I've managed to get a console displayed on it. Although it's rotated into portrait mode. I've also used a utility to copy fb0 to fb1 so I can get opengl output on there but it's quite slow. I'm still trying to work out what chipset it has and any other information to get it configured properly.

closed 2021-08-24 by github-actions[bot]