notro/fbtft #486: My display doesn't work automatically after reboot

rezaee, 2017-10-31:
Hi
I have an 2.2" ili9341 LCD, I can make it work by this commands:

```
sudo modprobe fbtft_device custom name=fb_ili9341 gpios=reset:1,dc:201,led:6 speed=16000000 rotate=90 bgr=1

FRAMEBUFFER=/dev/fb8 startx
```

But when I try to make it permanent:

```
/etc/modprobe.d/fbtft.conf
options fbtft_device custom name=fb_ili9341 gpios=reset:1,dc:201,led:6 speed=16000000 rotate=90 bgr=1

/etc/modules-load.d/fbtft.conf

spi-bcm2835
fbtft_device

 /usr/share/X11/xorg.conf.d/99-fbdev.conf
Section "Device"
  Identifier "myfb"
  Driver "fbdev"
  Option "fbdev" "/dev/fb8"
EndSection

```

It doesn't work after reboot (I have a blank black screen that looks like when I run "options fbtft_device custom name=fb_ili9341 gpios=reset:1,dc:201,led:6 speed=16000000 rotate=90 bgr=1" , I mean LCD is on but blank screen)

notro, 2017-11-01:
So your problem is X not using /dev/fb8, but it's using some other display?
You can start by looking in /var/log/Xorg.0.log.
This isn't fbtft specific, so I suggest you seek help elsewhere. I know very little about the userspace side of graphics.

rezaee, 2017-11-01:
I did explain what I did more specifically here: https://forum.armbian.com/topic/5561-how-to-configure-scriptbincustomfex-for-spi-lcd/

If you have time take a look it, maybe gives some idea to solve problem...

rezaee, 2017-11-01:
This is contetnt of log file:

```
X.Org X Server 1.18.4
Release Date: 2016-07-19
[    46.642] X Protocol Version 11, Revision 0
[    46.642] Build Operating System: Linux 4.4.0-97-generic armv7l Ubuntu
[    46.642] Current Operating System: Linux nanopim1 3.4.113-sun8i #16 SMP PREEMPT Tue Jun 13 14:15:57 CEST 2017 armv7l
[    46.643] Kernel command line: root=UUID=10b3b795-f372-4ea9-b78a-93ae9355c20c rootwait rootfstype=ext4 console=tty1 console=ttyS0,115200 hdmi.audio=EDID:0 disp.screen0_output_mode=1920x1080p60 panic=1$
[    46.643] Build Date: 13 October 2017  01:59:44PM
[    46.643] xorg-server 2:1.18.4-0ubuntu0.7 (For technical support please see http://www.ubuntu.com/support)
[    46.643] Current version of pixman: 0.33.6
[    46.644]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    46.644] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    46.644] (==) Log file: "/var/log/Xorg.0.log", Time: Wed Nov  1 10:29:43 2017
[    46.650] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    46.650] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    46.665] (==) No Layout section.  Using the first Screen section.
[    46.665] (==) No screen section available. Using defaults.
[    46.665] (**) |-->Screen "Default Screen Section" (0)
[    46.666] (**) |   |-->Monitor "<default monitor>"
[    46.666] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[    46.666] (**) |   |-->Device "myfb"
[    46.666] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    46.667] (**) Option "BlankTime" "0"
[    46.667] (**) Option "StandbyTime" "0"
[    46.667] (**) Option "SuspendTime" "0"
[    46.667] (**) Option "OffTime" "0"
[    46.667] (==) Automatically adding devices
[    46.667] (==) Automatically enabling devices
[    46.667] (==) Automatically adding GPU devices
[    46.667] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    46.678] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    46.678]    Entry deleted from font path.
[    46.678] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    46.678]    Entry deleted from font path.
[    46.678] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    46.678]    Entry deleted from font path.
[    46.678] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    46.678]    Entry deleted from font path.
[    46.678] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    46.678]    Entry deleted from font path.
[    46.678] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    46.678]    Entry deleted from font path.
[    46.678] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[    46.678] (==) ModulePath set to "/usr/lib/arm-linux-gnueabihf/xorg/extra-modules,/usr/lib/xorg/extra-modules,/usr/lib/xorg/modules"
[    46.678] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[    46.678] (II) Loader magic: 0xb6f85f70
[    46.679] (II) Module ABI versions:
[    46.679]    X.Org ANSI C Emulation: 0.4
[    46.679]    X.Org Video Driver: 20.0
[    46.679]    X.Org XInput driver : 22.1
[    46.679]    X.Org Server Extension : 9.0
[    46.682] (--) using VT number 2
[    46.682] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    46.683] (II) no primary bus or device found
[    46.683] (II) LoadModule: "glx"
[    46.689] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    46.756] (II) Module glx: vendor="X.Org Foundation"
[    46.756]    compiled for 1.18.4, module version = 1.0.0
[    46.756]    ABI class: X.Org Server Extension, version 9.0
[    46.756] (==) AIGLX enabled
[... 80 lines cut ...]
[    48.449] (II) No input driver specified, ignoring this device.
[    48.449] (II) This device may have been added with another device file.
[    48.451] (II) config/udev: Adding input device sunxi-gpiokey (/dev/input/event1)
[    48.451] (**) sunxi-gpiokey: Applying InputClass "evdev keyboard catchall"
[    48.451] (II) Using input driver 'evdev' for 'sunxi-gpiokey'
[    48.451] (**) sunxi-gpiokey: always reports core events
[    48.452] (**) evdev: sunxi-gpiokey: Device: "/dev/input/event1"
[    48.452] (--) evdev: sunxi-gpiokey: Vendor 0x1 Product 0x1
[    48.452] (--) evdev: sunxi-gpiokey: Found keys
[    48.452] (II) evdev: sunxi-gpiokey: Configuring as keyboard
[    48.452] (**) Option "config_info" "udev:/sys/devices/virtual/input/input1/event1"
[    48.452] (II) XINPUT: Adding extended input device "sunxi-gpiokey" (type: KEYBOARD, id 7)
[    48.452] (**) Option "xkb_rules" "evdev"
[    48.452] (**) Option "xkb_model" "pc105"
[    48.452] (**) Option "xkb_layout" "us"
[    48.455] (II) config/udev: Adding input device sunxi-ths (/dev/input/event2)
[    48.456] (II) No input driver specified, ignoring this device.
[    48.456] (II) This device may have been added with another device file.
[    70.907] (II) evdev: sunxi-gpiokey: Close
[    70.908] (II) UnloadModule: "evdev"
[    70.908] (II) evdev: vmouse: Close
[    70.908] (II) UnloadModule: "evdev"
[    71.003] (II) Server terminated successfully (0). Closing log file.

```

rezaee, 2017-11-01:
I found this tutorial in armbian documentation, but as I know my LCD uses "fb8" framebuffer. So I am not sure how can I configure this settings? maybe it runs after this changes?

```
script.bin/fex file
 
The settings in the [disp_init] section of the script.bin/fex file define the display output enabled at boot.

An example configuration for HDMI:
 
[disp_init]
disp_init_enable = 1
disp_mode = 0
screen0_output_type = 3
screen0_output_mode = 4
fb0_framebuffer_num = 2
fb0_format = 10
fb0_pixel_sequence = 0
fb0_scaler_mode_enable = 0
 
    disp_mode selects single-screen output or different dual screen modes. Generally this is 0, which means use screen0 with fb0 (one screen).
    screen0_output_type = 3 means HDMI output.
    screen0_output_mode selects the video/monitor mode to use (resolution and refresh rate). See the table in the Fex guide.
    fb0_framebuffer_num selects the number of buffers for fb0, generally you need 2 or more for video acceleration or Mali (3D), 3 is better.
    fb0_format and fb0_pixel_sequence determine the pixel format in the framebuffer. The above example (values of 10 and 0) selects the most common variant of 32bpp truecolor (ARGB).
    fb0_scaler_mode_enable selects whether the scaler should be enabled. Enabling it does not really scale pixels, it configures the scaler to scale pixels 1-to-1 which can fix screen refresh-related problems at 1080p resolution. See the section below.
 
Similar parameter are defined for screen1 (which is usually disabled in practice).

```

closed 2021-08-24 by github-actions[bot]