notro/fbtft #490: ILI9488 4 Wire SPI interface driver is not displaying correct image

titusece, 2017-12-06:
Hello,

[image]

I am trying to interface the ILI9488 (320x480) LCD display into ARM board which boots Linux OS.
And using FBTFT driver for that.
https://github.com/notro/fbtft
I have created a driver for ILI9488 based on the above driver base and also I have added ILI9488 init code from DS.
Now I can see some thing on my LCD display but not correct image.

What could be the problem ?
Can you please help me out on this ?

https://www.dropbox.com/s/d4lpuefnl98j473/problem-warp-display.webm?dl=0

Attached driver.

[fb_ili9488_ER-TFT035-6.zip](https://github.com/jaretburkett/ILI9488/files/1535738/fb_ili9488_ER-TFT035-6.zip)

LCD:
http://www.buydisplay.com/download/manual/ER-TFT035-6_Datasheet.pdf

Thanks!

notro, 2017-12-10:
AFAICT ILI9488 doesn't support RGB565 in SPI mode, only RGB666 which fbtft doesn't support:
- https://github.com/notro/fbtft/issues/487
- https://github.com/wdim0/esp8266_with_KeDei_lcd_module/blob/master/README.md
- https://id.scribd.com/document/334355551/ILI9488-Preliminary-DS-V090

titusece, 2017-12-12:
Hello,
Thanks for your response.
Yes, I referred the #487 post and your code is solved the problem.
Thank you very much!

I just added the below code in **drivers/staging/fbtft/fbtft-bus.c** file and removed this 'fbtft_write_vmem16_bus8' and replaced with 'write_vmem16_bus8' in **drivers/staging/fbtft/fbtft-core.c**

`/* 16bpp converted to 18bpp stored in 24-bit over 8-bit databus */
int write_vmem16_bus8(struct fbtft_par *par, size_t offset, size_t len)
{
	u16 *vmem16;
	u8 *txbuf = par->txbuf.buf;
	size_t remain;
	size_t to_copy;
	size_t tx_array_size;
	int i;
	int ret = 0;

	fbtft_par_dbg(DEBUG_WRITE_VMEM, par, "%s(offset=%zu, len=%zu)\n",
		__func__, offset, len);

	/* remaining number of pixels to send */
	remain = len / 2;
	vmem16 = (u16 *)(par->info->screen_buffer + offset);

	if (par->gpio.dc != -1)
		gpio_set_value(par->gpio.dc, 1);

	/* number of pixels that fits in the transmit buffer */
	tx_array_size = par->txbuf.len / 3;

	while (remain) {
		/* number of pixels to copy in one iteration of the loop */
		to_copy = min(tx_array_size, remain);
		dev_dbg(par->info->device, "    to_copy=%zu, remain=%zu\n",
						to_copy, remain - to_copy);

		for (i = 0; i < to_copy; i++) {
			u16 pixel = vmem16[i];
			u16 b = pixel & 0x1f;
			u16 g = (pixel & (0x3f << 5)) >> 5;
			u16 r = (pixel & (0x1f << 11)) >> 11;

			u8 r8 = (r & 0x1F) << 3;
			u8 g8 = (g & 0x3F) << 2;
			u8 b8 = (b & 0x1F) << 3;

			txbuf[i * 3 + 0] = r8;
			txbuf[i * 3 + 1] = g8;
			txbuf[i * 3 + 2] = b8;
		}

		vmem16 = vmem16 + to_copy;
		ret = par->fbtftops.write(par, par->txbuf.buf, to_copy * 3);
		if (ret < 0)
			return ret;
		remain -= to_copy;
	}

	return ret;
}`

closed 2017-12-12 by titusece

awetechnique, 2018-03-15:
[image]

i use your driver for my LCD display also, however, i encountered the problem of top of the screen missing some lines and bottom of the screen has some lines is not covered as shown above. My LCD TM035NDH07 resolution is 272x480. Do you encountered similar problem also?

[Final specification TM035NDH07-00 V2 1.pdf](https://github.com/notro/fbtft/files/1814029/Final.specification.TM035NDH07-00.V2.1.pdf)

Thanks!

kumarmohan7, 2018-04-17:
@titusece 
How to Build & Load fb_ili9488_ER-TFT035-6.c file in raspberry pi driver...
Please Help to interface ili9488 ER-TFT035-6 TFT to Raspberry Pi...........
If you have fb_ili9488.ko driver please share...
Thanks...