notro/fbtft #491: Using Display with parallel interface

Quaker03, 2017-12-07:
Hi,
I am trying to use my display with parallel interface on raspberry pi 2.
The driver is st7789 which uses the exactly same commands than ili9341. So I try to use this one from fbtft.
fbcp is installed this way:
`sudo apt-get install cmake`
`git clone https://github.com/tasanakorn/rpi-fbcp`
`cd rpi-fbcp/`
`mkdir build`
`cd build/`
`cmake ..`
`make`
`sudo install fbcp /usr/local/bin/fbcp`

What I am doing now is:
`sudo modprobe fbtft dma`
`sudo modprobe fbtft_device custom name=ili9341 gpios=reset:11,dc:15,cs:14,wr:18,db00:2,db01:3,db02:4,db03:17,db04:27,db05:22,db06:10,db07:9`
`fbcp &`

`tail /var/log/messages` says:
Primary display is 1920x1200
Secondary display is 240x320 16bps

`dmesg`:
fbtft: module is from the staging directory, the quality is unknown, you have been warned.
fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
bcm2708_fb soc:fb: soc:fb id=-1 pdata? no
fbtft_device: GPIOS used by 'fb_ili9341':
fbtft_device: 'reset' = GPIO11
fbtft_device: 'dc' ? GPIO15
....
bcm2708_fb soc:fb: soc:fb id=-1 pdata? no
platform fb_ili9341.0: fb_ili9341 id=0 pdata? yes
fb_ili9341: module is from the staging directory, the quality is unknown, you have been warned.
graphics fb1: fb_ili9341 frame buffer, 240x320, 150 KiB video memory, 16 KiB DMA buffer memory, fps=20

But I don't get any screen und the cs-pin gets never toggled.
Can anyone help me please?

notro, 2017-12-10:
cs is pulled low in [fbtft_init_display()](http://elixir.free-electrons.com/linux/latest/ident/fbtft_init_display) and never touched after that.
This is the function that bit-bangs the bus: [fbtft_write_gpio8_wr()](http://elixir.free-electrons.com/linux/latest/ident/fbtft_write_gpio8_wr)

You can use the debug parameter to get more info in the kernel log: https://github.com/notro/fbtft/wiki/Debug

Quaker03, 2017-12-14:
Thanks it helped a lot.
I needed to create a new module for the st7789v driver but now it is working.
Is it normal that I only can get 3fps via parallel interface or am I doing anything wrong?

All I did is loading linux source -> loading fbtft -> write new module fb_st7789v.c and add it in Kconfig and Makefile -> make the new kernel -> install modules -> install kernel -> reboot -> modprobe command like in my first post with fb_st7789v -> display gets black -> install fbcp and start it -> display shows screen. But only with 3fps. Can I get it faster anyway? I have read about any adafruid parallel display that gets over 60fps...

notro, 2017-12-16:
> I have read about any adafruid parallel display that gets over 60fps...

That has to be a hardware bus like the DPI/RGB bus which requires a different type of display. 

fbtft does gpio bit banging which is slow and this bus type is called DBI in MIPI jargon.
The gpio subsystem has support for setting many gpios at once, which in theory should speed things up, but I did a hack where I added support for this to the bcm2835 gpio driver, but it didn't amount to much so I didn't pursue it further.
The Raspberry Pi has a hardware bus that support DMA which can drive displays like you have, but there is no subsystem for i8080 type busses in the Linux kernel, so it's alot of work to get this done. And now that we have full DPI and DSI support in the vc4 driver, I don't think anyone will take on that task.

eorojas, 2018-02-09:
First, let me apologize. though have of experience talking to devices I am new to the occult issues of Linux drivers.  I am in the same situation with 4.4 a kernel.  So should I be using the st7789v or the fb_ili9341 driver?  I'm just getting going so have not tried either.  Thanks?

eorojas, 2018-02-12:
The spec I [have](http://www.newhavendisplay.com/specs/NHD-2.4-240320CF-CSXN-F.pdf). p4. Indicates that for an 8-bit interface I should be talking to DB8-DB15 to deliver 8-bit data, but your initialization uses DB0-DB7.  Am  I missing something?

eorojas, 2018-02-26:
@notro, so I've managed to build and install the driver using the Linux command line.  However, the driver is not actually running.  My build does not have modprobe installed, is there a configuration I can use to tell the kernel that the device is present?  I can add modprobe to the kernel build, but we are trying to minimize the size.  BTW, this is on a UP board running Linux 4.4, not a Rasberry.  thanks for your thoughts

notro, 2018-02-27:
Since you ask I assume there's no Device Tree. You can put the module arguments for fbtft_device on the kernel commandline: fbtft_device.name= etc. At least it worked that way some years back.

eorojas, 2018-02-27:
I may not have been clear.  I do initialize the driver from the Linux command line.
cat /proc/cmdline 
systemd.show_status=auto earlyprintk=ttyS0,115200n1 console=ttyS0,115200n1 reboot=efi fbtft_device.custom=1 fbtft_device.verbose=3 fbtft_device.name=ili9341 fbtft_device.gpios=reset:26,cs:20,dc:12,wr:1,rd:15,db08:9,db09:25,db10:11,db11:8,db12:7,db13:5,db14:6,db15:16 fbtft_device.buswidth=8 fbtft_device.debug=7   reset_devices sugarcube.bootid=29abf103ae7f7978c45bcac2edf7caed0cae5168 sugarcube.factoryver=1.1-3-g6b1324f-20180206-1949-prod

However, what I am seeing is the fbtft_device_init is called and completes, registers the driver, etc., but fb_ili9341:init_display is not called.  The fb_ili9341 driver appears in sys/devices/platform and bus/.  /dev/fb0 is VGA video and no other fbX appears.  My sense is there is something I'm not doing to connect to tell the kernel that I have a device I want it to control.  any thoughts on this?

eorojas, 2018-02-27:
Thanks for your replay, I clarified my situation on github, but here is
what I say there.

I may not have been clear. I do initialize the driver from the Linux
command line.
cat /proc/cmdline
systemd.show_status=auto earlyprintk=ttyS0,115200n1 console=ttyS0,115200n1
reboot=efi fbtft_device.custom=1 fbtft_device.verbose=3
fbtft_device.name=ili9341
fbtft_device.gpios=reset:26,cs:20,dc:12,wr:1,rd:15,db08:9,db09:25,db10:11,db11:8,db12:7,db13:5,db14:6,db15:16
fbtft_device.buswidth=8 fbtft_device.debug=7 reset_devices
sugarcube.bootid=29abf103ae7f7978c45bcac2edf7caed0cae5168
sugarcube.factoryver=1.1-3-g6b1324f-20180206-1949-prod

However, what I am seeing is the fbtft_device_init is called and completes,
registers the driver, etc., but fb_ili9341:init_display is not called. The
fb_ili9341 driver appears in sys/devices/platform and bus/. /dev/fb0 is VGA
video and no other fbX appears. My sense is there is something I'm not
doing to connect to tell the kernel that I have a device I want it to
control. any thoughts on this?

-- 
----------------
Emilio (eo) Rojas, co-active life-performance coach, and agile advocate
I'd like to support you and your team to accomplish your dreams.
https://www.linkedin.com/in/emilrojas/
CPCC, CSM
---

You have no idea how hard I've looked
for a gift to bring You.
Nothing seemed right.
What's the point of bringing gold to
the gold mine, or water to the ocean.
Everything I came up with was like
taking spices to the Orient.
It's no good giving my heart and my
soul because you already have these.
So I've brought you a mirror.
Look at yourself and remember me.

~ Rumi

notro, 2018-02-27:
> fbtft_device.name=ili9341

I believe this should be:
fbtft_device.name=fb_ili9341

eorojas, 2018-02-27:
that's better!  thanks. I have an error, from dd.c probe of fb_ili9341.0 failed with error -12, but that is something I can work with.

closed 2021-08-24 by github-actions[bot]